esphome/feature-requests #2977: FQDN support for UDP addresses list

skykingjwc, 2024-12-06:
**Describe the problem you have/What new integration you would like**
Would like to be able to use FQDN names in the UDP address list in addition to IP addresses.
This would allow users to use devicename.local addresses leveraging the advantages of mDNS and DNS and eliminate the burden of tracking IP addresses manually.

Addition of an FQDN option in lieu of an IP address.  For example:

`  - id: udp_external
    update_interval: 60s
    addresses:
      - remotedevice.local    #<----- FQDN instead of an IP address
 `

**Please describe your use case for this integration and alternatives you've tried:**

Problem is the current limitation of IP addresses only requires manual tracking of IP address to device mappings and precludes the possibility of using mDNS or DNS for address resolution.

**Additional context**

nagyrobi, 2024-12-07:
That imposes a lot of overhead to work properly. If the name resolution would happen once when the component starts, you wouldn't be happier anyway because if the IP changes behind the name, it won't work until you reboot the node. If the name would be resolved every time a new packet is sent, that would slow down seriously the whole loop - you don't want that...

skykingjwc, 2024-12-09:
> That imposes a lot of overhead to work properly. If the name resolution would happen once when the component starts, you wouldn't be happier anyway because if the IP changes behind the name, it [...]

Other components such as https://esphome.io/components/http_request.html#http-request-post-action use a FQDN.  Name caching is a widely accepted method of addressing performance issues.  No one is suggesting that the lookup be done during each iteration of a loop.

clydebarrow, 2024-12-12:
The http_request component doesn't do dns lookups directly, it relies on the platform libraries (ESP-IDF or Arduino) to handle that. To manage this properly ESPHome really needs a DNS component, that would handle lookups and caching with appropriate refreshing. Such a component could then be used by any other component that needs to resolve hostnames to addresses.