esphome/feature-requests #2980: Suport for CrowPanel E-Paper displays

girlpunk, 2024-12-07:
**Describe the problem you have/What new integration you would like**
Support for CrowPanel E-Paper (E-Ink) displays, utilising the same APIs and graphics libraries as the WaveShare ones

**Please describe your use case for this integration and alternatives you've tried:**
I have attempted using the exisitng WaveShare components, however as the drivers for these panels are different (the SSD1683), they are not compatible.

**Additional context**
Example use case: https://www.elecrow.com/wiki/CrowPanel_ESP32_E-paper_5.79-inch_HMI_Display.html
Driver datasheet: https://www.elecrow.com/download/product/DIS08792E/SSD1683_Datasheet.PDF

brillb, 2025-01-12:
I would also like these, as they are manufactured in a fully ready-to-go case, which simply accepts USB-C power input and then ready to go.  Similar to the waveshare cloud devices, but in more form factors/sizes and button controls.

droberin, 2025-02-15:
Hi.
I also tried Waveshare drivers on one of these 4.2 inches e-paper screen device:

```yaml
esphome:
  name: fishfoodscreen
  friendly_name: FishFoodScreen

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: "SomeSecretPasswordo=="

ota:
  - platform: esphome
    password: "someOTAPasswordo=="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

spi:
  clk_pin: GPIO12
  mosi_pin: GPIO11

display:
  - platform: waveshare_epaper
    model: 4.20in
    cs_pin: GPIO45
    dc_pin: GPIO46
    busy_pin: GPIO48
    reset_pin: GPIO47
    rotation: 270
    update_interval: 30s
    id: my_epaper_display
    lambda: |-
      it.fill(COLOR_OFF);
      it.printf(50, 120, id(font_large), TextAlign::CENTER, "%s", id(datetime_text).state.c_str());
      it.update();

# Font for text rendering
font:
  - file: "fonts/Arial.ttf"
    id: font_large
    size: 32

# Home Assistant datetime helper
text_sensor:
  - platform: homeassistant
    entity_id: input_datetime.plecomida
    id: datetime_text
    on_value:
      then:
        - component.update: my_epaper_display
```

It doesn't show any error but also screen never gets updated whatsoever. It was a good test...

janwest2, 2025-03-05:
I have the same issue, any news?

SeanTheITGuy, 2025-03-07:
If anyone wants to tackle this for esphome, I've recently added support to circuitpython for the Crowpanel 4.2", so the appropriate settings and init strings are available.

WebSpider, 2025-03-11:
Linking PR: https://github.com/adafruit/circuitpython/pull/10107

Dave-Casey, 2025-03-27:
I'm not experienced in C or custom components in ESPHome but I was feeling brave the other afternoon but got stuck, I don't know how to even include external components in the build.  I added this to my YAML but it doesn't get added to the build.  Ideas?
`external_components:
  - source: "/config/esphome/"
    components: [ssd1683]`

semvis123, 2025-05-17:
I've made a simple working version for the crowpanel 4.2 epaper display, its a bit rough, and doesn't implement LUT, but it works good enough for me :)
See:
https://github.com/semvis123/esphome-crowpanel-4.2-epaper

Dave-Casey, 2025-05-20:
This is exciting @semvis123 , aside from copying the font files into my directory, is it as simple as updating the yaml file or do I need to do other steps to get this working in my environment?

semvis123, 2025-05-20:
@Dave-Casey thanks for your interest, it should be as easy as adjusting the yaml file.

0. Add the crowpanel epaper component as an external component.
1. Define pins and display, see my yaml file.
2. Define the display lambda in the yaml.
3. Copy the font and icons. (And choose the codepoints you want to use)

For steps 1-3 you can look at my yaml file as a reference. 

See: https://esphome.io/components/external_components.html

Dave-Casey, 2025-05-21:
I followed this and adapted my yaml to match yours with the exception of the unique instance stuff.  I created a components folder and copies over your folder contents.  I now get this error: 

`File "/config/esphome/components/crowpanel_epaper/display.py", line 47, in <module>
    cv.Required(CONF_DC_PIN): cv.output_pin,
                              ^^^^^^^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'output_pin'
Failed config

display.crowpanel_epaper: [source /config/esphome/display-1.yaml:69]
  
  Platform not found: 'display.crowpanel_epaper'.
  platform: crowpanel_epaper`

Interesting enough output_pin does not appear in the display.py file, it's using cv.output_pin_schema.
I apologize, I'm in a bit over my head on this one.

semvis123, 2025-05-21:
@Dave-Casey I've fixed my repo, noticed that the folder structure was not completely correct for the external components.
You might need to delete the .esphome folder to purge the cache.

Also I've added a minimal example: 
https://github.com/semvis123/esphome-crowpanel-4.2-epaper/blob/minimal-example/esphome.yaml

Dave-Casey, 2025-05-21:
Thanks @semvis123 

"Hello world!"

Just wanted to confirm that your code is working on my device now. I had to slightly adjust the YAML example to include the espnow_receiver component as an external component (hosted locally). Here’s a sanitized snippet of what my working config looks like:

```
external_components:
  - source: github://semvis123/esphome-crowpanel-4.2-epaper
    components: [crowpanel_epaper]
  - source:
      type: local
      path: components/espnow_receiver
    components: [espnow_receiver]
```

Now I need to figure out how to adapt this to my own projects — let me know as you add improvements!

semvis123, 2025-05-21:
@Dave-Casey Great! Just fyi, the espnow_receiver is some connection logic that I use to connect with my other esp32 that has the sensors. You should be able to remove it safely if you don't have such a setup.

semvis123, 2025-05-27:
@janwest2 
No worries, it should be fixed again, there was a slight mistake in a recent pull request. But it's already fixed. To get the laters version just delete the .esphome folder and it should redownload.

See:
https://github.com/semvis123/esphome-crowpanel-4.2-epaper/pull/4

semvis123, 2025-05-29:
@janwest2 Ah my bad I forgot to update the example code, you now don't need to define the color anymore, and it uses the `Color::ON` and `Color::OFF`. I've updated it, so the minimal-example branch should work again. But you might have to do the same change in your code.

james-bell, 2026-10-02:
Just thought I'd chime in with a working Hello World yaml for the Crowpanel HMI E-Paper 4.2".  Bit more than the standard hello world but I was just happy it worked.

```
esphome:
  name: crowpanel-42-e-ink
  friendly_name: Crowpanel 4.2 e-ink
  on_boot:
    then:
      - logger.log: "ESPHome boot complete. Enabling display power and triggering refresh."
      - switch.turn_on: epd_power
      - delay: 1s
      - lambda: id(epaper_display).update();

esp32:
  variant: ESP32S3
  flash_size: 8MB
  framework:
    type: esp-idf
    
# Enable PSRAM (recommended)
psram:
  mode: octal
  speed: 80MHz

external_components:
  - source: github://ESPBoards/esphome-lvgl-crowpanel-epaper-5.79-4.2
    
logger:

api:
  encryption:
    key: "whatever yours is"

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: Crowpanel 4.2 e Fallback Hotspot
    password: "tricky password"

captive_portal:

# Keep the display powered
switch:
  - platform: gpio
    pin: GPIO7
    id: epd_power
    name: "E-Paper Power"
    restore_mode: ALWAYS_ON
    internal: true

binary_sensor:
  - platform: status
    id: wifi_connected_status
    name: "WiFi Connection Status"

time:
  - platform: sntp
    id: esptime
    timezone: "CET-1CEST,M3.5.0,M10.5.0/3"
    servers: pool.ntp.org                    

display:
  - platform: crowpanel_epaper
    id: epaper_display
    model: "4.20in-v1.2"   # or "4.20in"
    clk_pin: GPIO12
    mosi_pin: GPIO11
    cs_pin: GPIO45
    dc_pin: GPIO46
    reset_pin: GPIO47
    busy_pin: GPIO48
    full_update_every: 9999
    auto_clear_enabled: false
    update_interval: never
    #update_interval: 10s
    #full_update_every: 10
    #rotation: 180

font:
  - file: "fonts/Roboto-Regular.ttf"
    id: large_font
    size: 36

lvgl:
  displays: epaper_display
  buffer_size: 25%
  on_draw_end:
    - component.update: epaper_display
  pages:
    - id: main_page
      bg_color: 0xFFFFFF
      widgets:
        - label:
            align: CENTER
            text: "Hello, E-Paper!"
            text_color: 0x000000
            text_font: "large_font"
```