esphome/feature-requests #2981: Add a restart button/command to the ESPHome dashboard/device entries, and for Wi-Fi devices display the RSSI

vumaddibly, 2024-12-07:
**Describe the problem you have/What new integration you would like**

I would like to see a restart button or command added to each ESPHome dashboard entry,  and for Wi-Fi-connected devices, I would also like to see the RSSI displayed.

**Please describe your use case for this integration and alternatives you've tried:**

When working on the household electrics I often find I have to turn off and reboot an AP. Having a mesh Wi-Fi system, my ESPHome devices will usually connect to one of the remaining nodes but with an inferior signal. When the AP is restored the devices that have switched AP will not automatically reconnect to the best AP as [FR #731](https://github.com/esphome/feature-requests/issues/731) is still pending. This request would provide a simple workaround and would avoid every user having to expose a per-device restart button, create multiple RSSI sensors and build a new dashboard that mirrors the dashboard provided by the ESPHome Add-on.

**Additional context**

randybb, 2024-12-07:
ESPHome Dashboard is not actively contacting devices unless you click on the log or install button. For such avtions/data you have HA or anything else connected via mqtt.
Even you would implement it, it would be probably rejected.

vumaddibly, 2024-12-07:
@randybb, thanks for the reply. Would it still be possible for a restart button/command to be added to the ESPHome dashboard? If an RSSI sensor existed in a device's configuration could that be displayed within its dashboard entry?
````markdown
sensor:
  # WiFi Signal Sensor
  - platform: wifi_signal
    name: "RSSI"
    update_interval: 60s

randybb, 2024-12-07:
Dashboard is not getting these data from HA and it will be not actively using API for that, so don't think so anybody would ever do it. The idea is to have the UI cleanest as possible, if you need something extra, you have plenty of other options how to achieve it.
In HA it is super simple, there is no need to duplicate the same thing...
[image]

vumaddibly, 2024-12-07:
That looks very neat indeed! Would you be happy to share the YAML?

randybb, 2024-12-07:
Unfortunatelly, I made it long time ago and it is not super optimized, do every line is defined for each device - quite annoying. It is using [multiple-entity-row](https://github.com/benct/lovelace-multiple-entity-row)

```yaml
show_header_toggle: false
title: MCU Status
type: entities
entities:
  - entities:
      - entity: sensor.basement_uptime
        name: false
      - entity: sensor.basement_wifi_signal
        name: false
      - entity: button.basement_restart
        icon: mdi:restart
        name: false
    entity: binary_sensor.basement_status
    icon: mdi:devices
    name: Basement
    secondary_info:
      entity: sensor.basement_version
      name: false
    type: custom:multiple-entity-row
  - entities:
      - entity: sensor.basement_2_uptime
        name: false
      - entity: sensor.basement_2_wifi_signal
        name: false
      - entity: button.basement_2_restart
        icon: mdi:restart
        name: false
    entity: binary_sensor.basement_2_status
    icon: mdi:devices
    name: Basement 2
    secondary_info:
      entity: sensor.basement_2_version
      name: false
    type: custom:multiple-entity-row
```

For tracking current SDK version (yea, IDF 5.1.5 is in the current beta :) ) I am using a different card with https://github.com/custom-cards/flex-table-card - this one is fully dynamic. Not sure about buttons, but it might be possible.

[image]
```yaml
type: custom:flex-table-card
title: SDK Version
entities:
  include:
    - sensor.*_device_info
columns:
  - name: Name
    data: friendly_name
  - name: Value
    data: state
    modify: /ESP-IDF[^\|]*|SDK[^\|]*\|Core[^\|]*/.exec(x)
sort_by: friendly_name+
grid_options:
  columns: full

```

vumaddibly, 2024-12-07:
Thanks, it's very much appreciated. It's always good to have a head start :+1:

donburch888, 2025-01-12:
In my ESPHome builder tab in HA, I have a file `_common_wifi.yaml` which I include into all my devices yaml to ensure all devices report Wi-fi signal, Uptime, ESPHome version (because my Arlec PC191HA power points don't automatically update to the new firmware when it is installed) ... and a few other bits of wi-fi info because, well, why not. 

```
###########################################################
#
# Start with the Wi-fi connection
#
###########################################################

wifi:
  ssid:     $wifi_ssid
  password: $wifi_password
  manual_ip:
    static_ip: 192.168.1.${deviceIP}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename Fallback Hotspot"
    password: !secret wifi_ap_password

ota:
  platform: esphome
  password: !secret esphome_ota_password

captive_portal:

# this displays the device's status at http:IP_Address
web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: !secret esphome_api_encryption

###########################################################
#
#   Add some common sensors - wi-fi signal strength, 
#       uptime, ESPHome version & compile date/time
#
###########################################################

sensor:
  - platform: wifi_signal
    name: $devicename Wifi signal
    update_interval: ${update_interval_network}     # how often to report wifi signal strength

  # human readable uptime sensor output to the text sensor 
  - platform: uptime
    name: $devicename Uptime
    id: uptime_sensor
    update_interval: ${update_interval_network}     # how often to report
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

text_sensor:
  - platform: version
    name: $devicename ESPHome Version
    hide_timestamp: False

  - platform: wifi_info
    ip_address:
      name: $devicename Wifi IP Address
    ssid:
      name: $devicename Wifi connected to SSID
    mac_address:
      name: $devicename Wifi MAC Address

  - platform: template
    name: $devicename Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start

time:
  - platform: sntp
    id: sntp_time
    update_interval: "120h"         # I guess this is to poll the SNTP server ?
```

Then in my lovelace dashboard I have a panel to summarise network data:
[image]
```
type: entities
entities:
  - entity: switch.hs110_101
    secondary_info: last-updated
  - entity: switch.hs110_102
    secondary_info: last-updated
  - entity: switch.pc191ha_106
    secondary_info: last-updated
  - entity: sensor.pc191ha_111_wifi_signal
    secondary_info: last-updated
  - entity: sensor.pc191ha_112_wifi_signal
    secondary_info: last-updated
  - entity: sensor.pc191ha_113_wifi_signal
    secondary_info: last-updated
  - entity: sensor.pc191ha_114_wifi_signal
    secondary_info: last-updated
  - entity: sensor.lifx_mini_140_rssi
    secondary_info: last-updated
  - entity: sensor.lifx_color_141_rssi
    secondary_info: last-updated
  - entity: sensor.lifx_color_142_rssi
    secondary_info: last-updated
  - entity: cover.blindegwt_160
    secondary_info: last-updated
  - entity: sensor.greenhouse_wifi_signal
    secondary_info: last-updated
title: Wi-Fi device connections
state_color: true
show_header_toggle: false
```
Note the last-updated as secondary_info because HA generally assumes the last reported value, when in fact the value has not changed or **_is no longer being reported_**