esphome/feature-requests #2982: USB hub function on RP2040 Pico W

ruteclrp, 2024-12-08:
**Describe the problem you have/What new integration you would like**
I need an ESPHOME device to connect to a modbus slave via USB. This requires the board to have USB hub functionality. The Raspberry Pico W can do this, but requires the code to set dr_mode=host. 
this is not yet implemented by in esphome for RP2040 boards. 

**Please describe your use case for this integration and alternatives you've tried:**

ability to set dp_mode to host for the USB connection on the board

**Additional context**

nagyrobi, 2024-12-08:
@clydebarrow

clydebarrow, 2024-12-08:
It needs a lot more than that. The USB device will need a (software) driver on the Pico, and that will depend on what the device is. Why a Pico? There is code for an ESP32-S3 that will operate in USB host mode driving serial ports, but again it depends on what the device is.

ruteclrp, 2024-12-08:
Clyde Stubbs skrev den 2024-12-08 10:02:

> It needs a lot more than that. The USB device will need a (software) 
> driver on the Pico, and that will depend on what the device is. Why a 
> Pico? There is code for an ESP32-S3 that will operate in USB host mode 
> driving serial ports, but again it depends on what the device is.
> 
> [2].

Thanks Clyde,

firstly, the USB connected slave is a rather dumb one which works out of 
the box with USB connections on Raspberry 3 and 4 ( which I have tried 
successfully).

Secondly, the stupid modbus implementation on the slave requires some 
unconventional functions to read certain holding registers. In fact, it 
requires dummy writing with fc 0x06 to read the registers, and such 
cannot easily be accomplished on the standard modbus implementaion in 
HA.

What I am trying to ascertain is if the custom function for modbus in 
ESPHOME can get me the desired result - but that requires an ESP device 
to have USB host (hub) functionality, hence the Pico.

Now I however realises that the host function need to be activated by 
code, which is missing for the RP2040 boards in ESPHOME.

Can you confirm that activating the host function on an ESP32-S3 board 
is available in ESPHOME? and whether it is a parameter to be set in the 
.yaml configuration file for that board.

Thank you for you help here.

/Robert

Links:
------
[1] 
https://github.com/esphome/feature-requests/issues/2982#issuecomment-2525548266
[2] 
https://github.com/

randybb, 2024-12-08:
Why are you making it hard? Just take two gpios and connect them ta rs485 transceiver.

ruteclrp, 2024-12-08:
Hi Roman,

yes, I could tear down the slave unit to uncover the RS485 bus, but cas 
is that it is an enclosed unit provided by the heat pump maker with only 
one USB-B port available. This is why I would rather just do everything 
using the available port on the unit.

Roman Priesol skrev den 2024-12-08 11:14:

> Why are you making it hard? Just take two gpios and connect them ta 
> rs485 transceiver.
> 
> [2].

Links:
------
[1] 
https://github.com/esphome/feature-requests/issues/2982#issuecomment-2525574357
[2] 
https://github.com/

clydebarrow, 2024-12-08:
RS485 wasn't mentioned in the original request so I assume it's not involved.

> Can you confirm that activating the host function on an ESP32-S3 board
is available in ESPHOME? 

There is currently no USB host support merged into ESPHome for any board, and it's not as simple as "activating the host function". As I said before the device needs a a driver to talk to it over USB. It seems that there is a Linux driver since it works on a Pi, but a Pico is not a Pi and doesn't run Linux.

I do have code I wrote that works as an external component and enables an ESP32-S3 to drive a USB serial adapter but without knowing what device you are using I can't tell you if it will work. I am using it to talk modbus via a 4 port USB-RS485 VCP vendor specific adapter. It also works with USB CDC (ACM) devices.

randybb, 2024-12-08:
modbus is a protocol working on top of the rs485 physical layer, but it is still uart. So if uart over usb host works, and you have such adapter, then we got a solution :)

ruteclrp, 2024-12-08:
Hi Clyde,

thanks again. The slave is a vendor developed black box. It is something 
developed some time back in 2014 and runs a serial connection via a 
"Virtual Com port" on the USB. The Comp port settings are:

9600 baud

8 bit word

1 stop bit

No parity

No hardware flow control

So it seems rather basic and your mentioned component might work out of 
the box.

/Robert

Clyde Stubbs skrev den 2024-12-08 11:39:

> RS485 wasn't mentioned in the original request so I assume it's not 
> involved.
> 
>> Can you confirm that activating the host function on an ESP32-S3 board
>> is available in ESPHOME?
> 
> There is currently no USB host support merged into ESPHome for any 
> board, and it's not as simple as "activating the host function". As I 
> said before the device needs a a driver to talk to it over USB. It 
> seems that there is a Linux driver since it works on a Pi, but a Pico 
> is not a Pi and doesn't run Linux.
> 
> I do have code I wrote that works as an external component and enables 
> an ESP32-S3 to drive a USB serial adapter but without knowing what 
> device you are using I can't tell you if it will work. I am using it to 
> talk modbus via a 4 port USB-RS485 VCP vendor specific adapter. It also 
> works with USB CDC (ACM) devices.
> 
> [2].

Links:
------
[1] 
https://github.com/esphome/feature-requests/issues/2982#issuecomment-2525582440
[2] 
https://github.com/

clydebarrow, 2024-12-08:
> modbus is a protocol working on top of the rs485 physical layer, but it is still uart.

Modbus is a request/reply protocol that is implemented on a wide variety of transports - RS485 is common, but it is also used over RS232, UDP, USB and other transports that have nothing to do with RS485.

ruteclrp, 2024-12-09:
Hi Clyde. Let's not spend more time on the rs485 discussion as this is entirely related to getting the usb host function working to enable the VCP connection to the slave. 
you mentioned that you have developed a "custom component" which can be used with esphome? is that something you could share?
also, presume it need to be called via some lamda scripting in the .yaml file to be included in the firmware upload to the device?

Thanks again
Robert

clydebarrow, 2024-12-09:
What's the VID and PID of the USB device?

ruteclrp, 2024-12-10:
Hi ClydeI did a debug with the USBview application from Microsoft, and that did not even recognize the device. it did trigger the hub and ports to show, but still stated nothing was connected. Question is if it is possible for such device to function without pid and vid?Or, is it because Windows needs special driver here?It does work out of the box with a Rasbperry Pi 4 attached. Sent from my iPhone

nagyrobi, 2024-12-10:
In Windows, it shoud pop up in device manager as an unrecognized device if it has no drivers installed. Right-click on that and cycling through the properties window dropdows you should see listed the PID/VID.

clydebarrow, 2024-12-10:
> It does work out of the box with a Rasbperry Pi 4 attached

So on Linux the `lsusb` command will list attached devices with their VID/PID.

E.g.

```
➜  ~ lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 1a40:0101
Bus 001 Device 003: ID 067b:2303
```
The first two of those are hubs, device 003 is a PL2303 USB-serial adapter.

ruteclrp, 2024-12-10:
Hi Clyde

Sorry for the image - I could not copy from the terminal window.

The RPI4 is on an Argon box with SSD attached via the USB bridge. 
Suspect the Bus1, Device4 is the one, but will try tonight to unplug the 
VCP connection to wee which one disappears.

Clyde Stubbs skrev den 2024-12-10 10:52:

>> It does work out of the box with a Rasbperry Pi 4 attached
> 
> So on Linux the lsusb command will list attached devices with their 
> VID/PID.
> 
> E.g.
> 
> ➜  ~ lsusb
> Bus 001 Device 001: ID 1d6b:0002
> Bus 001 Device 002: ID 1a40:0101
> Bus 001 Device 003: ID 067b:2303
> 
> The first two of those are hubs, device 003 is a PL2303 USB-serial 
> adapter.
> 
> [2].

Links:
------
[1] 
https://github.com/esphome/feature-requests/issues/2982#issuecomment-2531048083
[2] 
https://github.com/
--=_d1a513c92bfd005e5ca2bc41824b3dc8
Content-Type: multipart/related;
 boundary="=_1f7f652e55229025fdf52e47ecc21572"

--=_1f7f652e55229025fdf52e47ecc21572
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 12pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hi Clyde</p>
<p><img ***@***.***" /></p>
<p><br /></p>
<p>Sorry for the image - I could not copy from the terminal window.</p>
<p>The RPI4 is on an Argon box with SSD attached via the USB bridge. Suspec=
t the Bus1, Device4 is the one, but will try tonight to unplug the VCP conn=
ection to wee which one disappears.</p>
<p><br /></p>
<p><br /></p>
<p id=3D"reply-intro">Clyde Stubbs skrev den 2024-12-10 10:52:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<p dir=3D"auto">It does work out of the box with a Rasbperry Pi 4 attached<=
/p>
</blockquote>
<p dir=3D"auto">So on Linux the <code class=3D"v1notranslate">lsusb</code> =
command will list attached devices with their VID/PID.</p>
<p dir=3D"auto">E.g.</p>
<pre class=3D"v1notranslate"><code class=3D"v1notranslate">=E2=9E=9C  ~ lsu=
sb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 1a40:0101
[... 180 lines cut ...]
N8uO+3YZUaCbNEyyIgneixbUq3v6O0Akwb3w4Up9c+34ktHcLhPCTuE+2duWBLdu8bL3a45EdtQk
y/Vv1HaamGfqjz4cb26SFLhxM1cHKSikSEGhveUU+fg78C6ErMZJpCSrv1YkwRm9UCTBvUDiSn1z
7cCWjK7WiH/PKcPVLn2i+pR6HASWqNjekZklTAx1EYFYU2NA+88KyTIlsg2SBfbTyzganMHvQ730
TqotCRDu57PIfrOuEuThzPVv1HZe+FpaebnXy1BNGkRKsiIJHtgFTAORBHdHqtmS4JxLnSO6T9be
QIWRrP0xpzQ8N7o4JK7ZvFr5gJeawJXLzyJZrkS2YRdD8QJo72xzvLFWQt74ox/O0cZBPN6YhPzg
5BJmJnrh1Drj+qNhknUWJJLgHOj9O7tIgpMC7rttko15isn5WUz022/wvkRFSNFS2fWw21+aBSmK
2H4ckqUJ42IO2uX4ZzurWPho02PhuNbNJiTJ2uNAg0P9591eSyoexe/IrW1g7+Sn/vquV6TX9bK7
MpLlSmTbF8I1CflM+jP+/HlNE4k1TA88dpUx8sQ5Yml49XhxXtxdxsVuGm83/lR3AT0ROcnqtRBJ
8ABn+Hd2kQQvIF96jjipgxpjohLO9gs4aIsjPnCN8JLgPJIN1ZP8HgpJso3nOb3baygiuKwl1NQ1
SCizKpFt1fXSoWyLGNUhbyjmasKqNd88L8wiloZXjxf3dodSqoiCZEUSfMEmXcHtnv6dXSTBqzhG
Kwn+a5GsFQenRFLvVFdb7FPyOvUHH1FAD5INksj2/iCYhKWi8mqNZG3dqZE4aERCXqtC38gcphLD
6GypflS4PV2za3gkWycZLZLgDPz9OrtIgvdqY9eLfRSy69g7v67gaYy+ThFKEtyXdBgua8Qk1Ei2
mZLgZs7TTZfOUVeuRDbpZVNO9rGLhHwIkrXla6OIg/DxUnV67HkKa7MDlJddIVVbtbTBrZGsSIIb
U6gVmkLZXSaS4BqRFuslmXtmkFmmbTTFXZI434CSJHilr/w6I9nmSoKbxNd6DOdqubU31Ro8cSWy
PUnW8tvFDuJvP7I/VVz/Rm3nWcHBOazNa9snbplkRRKcG0MiCe6FFFfqm2tXWw6PZG9zC9dNSYJb
+4cv9zM0MjtAa88IpmYSGO40dhtYJMuVyLbSBWc7KVosPNffUdltROuWZ1tzWNislXn1w5nr36jt
3OKya3CSsJlAb1sJ3zPTeH/A7eeGXcMjWdfiRBI8wAsiCe4JEFPqG1w7rpqpc7Sr/b+bVDW3f0Vw
4ivUQgu3vebiUq3yOq2iH52jbaAXxQLlgfUdDTyJbD815xopdy7OXP9GbeeDXyh/RE2yIgleO/X3
7o8iCe7LVS6SzGeHBSxkHdvAOXZc0rFVKJItXGFJ9gYlwTvG3uDNb8+gb2stXeBodwPp41Fk6FRB
lWTpbwyJ7CeDr5CYfIb+9kdoqxxs0M78e9/tEYgzx7+a36K0c5MOLx7RgYp1bP7pkA5nfnBDj2SZ
7xczQUAQEAQeNAJCsg/a/dJ4QUAQaDYCQrLNRljeLwgIAg8agf8HmJlvPCdJkPUAAAAASUVORK5C
YII=
--=_1f7f652e55229025fdf52e47ecc21572--

--=_d1a513c92bfd005e5ca2bc41824b3dc8--

nagyrobi, 2024-12-10:
@ruteclrp no image can be seen, you should probably use the Github web UI to communicate within the issue, not your e-mail client, which also apparently spams your posts with random html code.

ruteclrp, 2024-12-10:
Sorry guys, my webmail client.

Here it is with slave disconnected:
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 002: ID 174c:1156
Bus 002 Device 001: ID 1d6b:0003

Here it is with slave connected:
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 004: ID 0483:5740
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 002: ID 174c:1156
Bus 002 Device 001: ID 1d6b:0003

So the slave is:

ID 0483:5740

/Robert

nagyrobi, 2024-12-10:
See https://the-sz.com/products/usbid/index.php?v=0x0483&p=0x5740&n=

It's just a Virtual COM Port by STMicroelectronics

ruteclrp, 2024-12-10:
Yes, I know, it is a VCP on the modbus master display controller of a heat pump. It acts as a slave via the VCP to allow a modbus master to read its registers. That works with a direct connection to the RPI4 running my HA. However, as the modbus implementation on the controller is unconventional, I am trying to have an ESPHOME device to do the modbus master function - using the custom function procedure to access the unconventional holding registers. 

/Robert

clydebarrow, 2024-12-10:
Well, you might just be in luck - the STM32 VCP appears to be a vanilla CDC-ACM port, which is supported. Try this:

```
external_components:
  - source: github://clydebarrow/esphome@power-meter
    components: [modbus, usb_uart, usb_host]
    refresh: 1h

usb_uart:
  - type: STM32_VCP
    channels:
      id: usb_uart_id
      baud_rate: 9600
      buffer_size: 1024
      debug: true
      dummy_receiver: true
```

You can use `usb_uart_id` as an id wherever you need a uart component, e.g. in the modbus config. With this you should see log events when you plug the device in to the S3 USB port.

You will ideally need an S3 board that has two USB connectors, one for UART0 and one for the USB interface - use UART0 in the logger config and flash via the UART port leaving the USB port free for your device. Alternatively flash and log via WiFi.

ruteclrp, 2024-12-11:
Thanks Clyde, much appreciated.
I am now trying to set up my .yaml file for the device, and wonder if I need to specify the GPIOs for the usb_uart?
I gather it would default to the s3 board's hardware uart for Arduino framework (USB_CDC on pin 19/20)?
I would also need to set the logger to uart0?

sorry for being a bit slow here, by I am really a newbie to ESPs and lower level fiddling with micro processors.

ruteclrp, 2024-12-11:
Also, tried to compile and got this error:
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/dvi-lv12.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 452, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/data/external_components/a4d2ac70/esphome/components/usb_host/__init__.py", line 50, in to_code
    add_idf_sdkconfig_option("CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE", 1024)
  File "/esphome/esphome/components/esp32/__init__.py", line 149, in add_idf_sdkconfig_option
    raise ValueError("Not an esp-idf project")
ValueError: Not an esp-idf project

nagyrobi, 2024-12-11:
See https://esphome.io/components/esp32.html#esp-idf-framework

ruteclrp, 2024-12-11:
Thanks. Learning all the time.

now, it fails when compiling the usb_uart component with:

src/esphome/components/usb_uart/USBUartComponent.cpp: In member function 'void esphome::usb_uart::USBUartComponent::start_output(esphome::usb_uart::USBUartChannel*)':
src/esphome/components/usb_uart/USBUartComponent.cpp:229:66: error: missing template arguments before '(' token
     uart::UARTDebug::log_hex(uart::UART_DIRECTION_TX, std::vector(data, data + len), ',');  // NOLINT()
                                                                  ^
*** [.pioenvs/dvi-lv12/src/esphome/components/usb_uart/USBUartComponent.o] Error 1

Suspect this is related to my earlier question on setting gpio for this uart?

clydebarrow, 2024-12-11:
>wonder if I need to specify the GPIOs for the usb_uart? 

You can't specify them, they are fixed by the hardware. And don't use Arduino. Like I said above you will preferably use a board with two USB connectors, and select UART0 for logging.

ruteclrp, 2024-12-11:
Thanks Clyde,
much better with esp-idf. Though it fails on compiling the special usb_uart component. Seems something missing in my config.

clydebarrow, 2024-12-11:
No, that was my fault, I made a change but obviously didn't properly test it. In the external component source line, replace `@power-meter` with `@58d8721775c7db9fc1a789d2fb125827f335cae5` - that rolls back to a previous commit.

clydebarrow, 2024-12-11:
I just updated the repository, so the original should now work, if the external component is refreshed.

ruteclrp, 2024-12-14:
Hi Clyde. Now got my hands on some ESP32-S3 boards. Just rather confused. I can build a simple installation with the Arduino framework which nicely connects to my wifi. When I change the framework setting to esp-idf in the .yaml file and install it nothing happens. It will not connect. Also, I cannot read the log via USB when on the idf framework. What is it that I do not understand here???

clydebarrow, 2024-12-14:
Best place for general support for ESPHome is on Discord: https://discord.com/channels/429907082951524364/429907082955718657

ruteclrp, 2024-12-14:
Thumbs up. Found the missing pieces to get the board running under the idf framework - now I am in business :-)
But, powering the board via the first USB port (the one also used to flash) and connecting the modbus slave to the other still gives me the no response message when firing the modbus request. Is this because I need to specify something about which uart shall be routed to which USB port?
The log says:
[10:48:37][C][logger:185]: Logger:
[10:48:37][C][logger:186]:   Level: DEBUG
[10:48:37][C][logger:188]:   Log Baud Rate: 115200
[10:48:37][C][logger:189]:   Hardware UART: UART0
[10:48:37][C][modbus:165]: Modbus:
[10:48:37][C][modbus:167]:   Send Wait Time: 250 ms
[10:48:37][C][modbus:168]:   CRC Disabled: NO
[10:48:37][C][usb_host:267]: USBClient
[10:48:37][C][usb_host:268]:   Vendor id 0483
[10:48:37][C][usb_host:269]:   Product id 5740
[10:48:37][C][usb_uart:169]:   UART Channel 0
[10:48:37][C][usb_uart:170]:     Baud Rate: 9600 baud
[10:48:37][C][usb_uart:171]:     Data Bits: 8
[10:48:37][C][usb_uart:172]:     Parity: NONE
[10:48:37][C][usb_uart:173]:     Stop bits: 0
[10:48:37][C][usb_uart:175]:     Debug: true
[10:48:37][C][usb_uart:177]:     Dummy receiver: true

both seems to use uart0

I would expect that the USB hub would use USBCDC

If not, I guess setting the logger to baud_rate: 0 would disable the logger and allow the coexistence on uart0?

Just not sure what you had espected to happen with the hub component I included in the build

/Robert

clydebarrow, 2024-12-14:
Do you see anything in the log when you plug in the USB device? Is the board providing 5V to the USB port?

ruteclrp, 2024-12-14:
The log says:

[13:56:41][C][logger:185]: Logger:
[13:56:41][C][logger:186]:   Level: VERBOSE
[13:56:41][C][logger:188]:   Log Baud Rate: 115200
[13:56:41][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG

[13:56:41][C][modbus:165]: Modbus:
[13:56:41][C][modbus:167]:   Send Wait Time: 250 ms
[13:56:41][C][modbus:168]:   CRC Disabled: NO
[13:56:41][C][usb_host:267]: USBClient
[13:56:41][C][usb_host:268]:   Vendor id 0483
[13:56:41][C][usb_host:269]:   Product id 5740
[13:56:41][C][usb_uart:169]:   UART Channel 0
[13:56:41][C][usb_uart:170]:     Baud Rate: 9600 baud
[13:56:41][C][usb_uart:171]:     Data Bits: 8
[13:56:41][C][usb_uart:172]:     Parity: NONE
[13:56:41][C][usb_uart:173]:     Stop bits: 0
[13:56:41][C][usb_uart:175]:     Debug: true
[13:56:41][C][usb_uart:177]:     Dummy receiver: true
[13:56:41][C][modbus_controller:349]: ModbusController:
[13:56:41][C][modbus_controller:350]:   Address: 0x10
[13:56:41][C][modbus_controller:351]:   Max Command Retries: 4
[13:56:41][C][modbus_controller:352]:   Offline Skip Updates: 0
[13:56:41][C][modbus_controller:354]: sensormap
[13:56:41][C][modbus_controller:356]:  Sensor type=4 start=0x7 offset=0x0 count=1 size=2
[13:56:41][C][modbus_controller:360]: ranges
[13:56:41][C][modbus_controller:362]:   Range type=4 start=0x7 count=1 skip_updates=0
[13:56:41][C][modbus_controller:365]: server registers
[13:56:41][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'LV12 Outside temp'
[13:56:41][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'temperature'
[13:56:41][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[13:56:41][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: '°C'
[13:56:41][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1

[13:56:56][V][modbus_controller:232]: Updating modbus component
[13:56:56][V][modbus_controller:198]: Range : 7 Size: 1 (4) skip: 0
[13:56:56][V][modbus_controller:043]: Sending next modbus command to device 16 register 0x07 count 1
[13:56:56][V][usb_uart:126]: Channel not initialised - write ignored
[13:56:56][V][modbus:223]: Modbus write: 10.04.00.07.00.01.83.4A (8)
[13:56:56][V][modbus_controller:568]: Command sent 4 0x7 1 send_count: 1
[13:56:56][V][modbus:042]: Stop waiting for response from 16
[13:56:57][V][modbus_controller:043]: Sending next modbus command to device 16 register 0x07 count 1
[13:56:57][V][usb_uart:126]: Channel not initialised - write ignored
[13:56:57][V][modbus:223]: Modbus write: 10.04.00.07.00.01.83.4A (8)
[13:56:57][V][modbus_controller:568]: Command sent 4 0x7 1 send_count: 2
[13:56:57][V][modbus:042]: Stop waiting for response from 16

[13:57:29][V][esp-idf:000]: E (53276) HUB: Root port reset failed

[13:57:29][W][component:237]: Component usb_host took a long time for an operation (313 ms).
[13:57:29][W][component:238]: Components should block for at most 30 ms.

Since the last attempt I removed the uart assignment for the logger to uart0. Now the logger defaults to USB_SERIAL_JTAG

Still, the USB host component seems not to work as intended.

You asked if the board provided 5V to the uab port. How would I check that easiest?

ruteclrp, 2024-12-14:
Found an anomaly.

The stop bits were not set, and should default to 1. However the validation check of the configuration file showed that stop bits were 0. 

The spec for the VCP is:
Com port settings:
9600 baud
8 bit word
1 stopbit
No parity
No hardware flow control

I have now set the stop bits to 1 in the conf, and recompiles.

The log now says:
[14:25:19][C][usb_host:267]: USBClient
[14:25:19][C][usb_host:268]:   Vendor id 0483
[14:25:19][C][usb_host:269]:   Product id 5740
[14:25:19][C][usb_uart:169]:   UART Channel 0
[14:25:19][C][usb_uart:170]:     Baud Rate: 9600 baud
[14:25:19][C][usb_uart:171]:     Data Bits: 8
[14:25:19][C][usb_uart:172]:     Parity: NONE
[14:25:19][C][usb_uart:173]:     Stop bits: 0
[14:25:19][C][usb_uart:175]:     Debug: true
[14:25:19][C][usb_uart:177]:     Dummy receiver: true

stop bits seems wrong here!

But, the log continues:

[14:25:36][D][usb_host:027]: Event flags 2X
[14:25:37][W][component:237]: Component usb_host took a long time for an operation (310 ms).
[14:25:37][W][component:238]: Components should block for at most 30 ms.
[14:25:37][W][component:237]: Component usb_host took a long time for an operation (59 ms).
[14:25:37][W][component:238]: Components should block for at most 30 ms.
[14:25:37][D][usb_host:032]: New device 1
[14:25:37][D][usb_host:068]: Open device 1
[14:25:37][D][usb_host:074]: Get descriptor device 1
[14:25:37][D][usb_host:080]: Device descriptor: vid 483 pid 5740
[14:25:37][D][usb_host:089]: Device connected: Manuf: STMicroelectronics; Prod: STM32 Virtual COM Port  ; Serial: 48D874673036
[14:25:44][V][modbus_controller:232]: Updating modbus component
[14:25:44][V][modbus_controller:198]: Range : 7 Size: 1 (4) skip: 0
[14:25:44][V][modbus_controller:043]: Sending next modbus command to device 16 register 0x07 count 1
[14:25:44][V][usb_uart:126]: Channel not initialised - write ignored
[14:25:44][V][modbus:223]: Modbus write: 10.04.00.07.00.01.83.4A (8)
[14:25:44][V][modbus_controller:568]: Command sent 4 0x7 1 send_count: 1
[14:25:44][V][modbus:042]: Stop waiting for response from 16
[14:25:45][V][modbus_controller:043]: Sending next modbus command to device 16 register 0x07 count 1
[14:25:45][V][usb_uart:126]: Channel not initialised - write ignored
[14:25:45][V][modbus:223]: Modbus write: 10.04.00.07.00.01.83.4A (8)

So, now the slave is recognised.

But, why is the channel not initialised?

ruteclrp, 2024-12-14:
Just found your other support thread with https://github.com/esphome/feature-requests/issues/2944#top

My board is the same - also have that soldering point on the back. Suspect that the issue is the missing 5 V to the USB port.

Still, the stop bits issue persist and suspect it is set in your component rather than with parameter in config? can see that when I state the "stop_bits: 1" in config it is ignored and the validation returns "stop_bits: '1'"

Anyway, will solder the usb-otg connection on the board tomorrow and see how it goes.

ruteclrp, 2024-12-15:
Some observations.

1. soldering the usb-ota point on the back side of the board did not generate any noticeable change. It may have caused 5 V to the USB port, but my slave is externally powered anyway, thus this change was not needed to establish the connection.
2. I observed the same pattern that building the firmware with the esp-idf framework could not be flashed via cable as the board did not properly boot. Starting out with a minimal Arduino based firmware first and then replace with the esp-idf firmware wia ota was also for me the way to go. Did not however make the conclusion that it was the flashing method that caused it.
3. The issue I have is not that the slave unit is not seen. When I unplug/plug the slave USB it is clearly recognised in the log.
4. If is set the logger to hardware_uart: uart0, then it does not work. Leaving it out of the configuration defaults the port to: hardware_uart: USB_SERIAL_JTAG
5. The issue that remain is that the modbus channel need initialising.
6. Whether this is related to the stop_bits issue is not clear to me
It seems that my issue and Felipecrs experience are almost identical.

Thanks for your great assistance.

Robert

ruteclrp, 2024-12-15:
OK, had a look at you code and looks like your default stop_bits are set to 1, even when the validation says 0?

Maybe it is all because there is not a mbc_master_create_serial function being invoked?

clydebarrow, 2024-12-15:
> Maybe it is all because there is not a mbc_master_create_serial function being invoked?

No, that's part of the ESP-IDF modbus implementation and is not used by ESPHome. Stop bits aren't even used by a virtual CDC_ACM device so that's just cosmetic.

The "channel not initialised" error is odd, since the device is clearly being connected, but there should be more log entries related to it. Attach your YAML, and turn on verbose logging for usb_uart - that will dump the device descriptor.

ruteclrp, 2024-12-15:
[dvi-lv12.yaml.txt](https://github.com/user-attachments/files/18141869/dvi-lv12.yaml.txt)

Here is the YAML file for the board. 

When you say "turn on verbose logging for usb_uart" do you mean, verbose level under logger, and debug under usb_uart? 
or, is there some extra setting under usb_uart I need to set?

ruteclrp, 2024-12-15:
[logs_dvi-lv12_logs(1).txt](https://github.com/user-attachments/files/18141884/logs_dvi-lv12_logs.1.txt)

OK, here is the log with the descriptor after I plugged the slave in.

ruteclrp, 2024-12-15:
So, the log states that the usb_uart sits on uart0. Should it not have been on the USB_CDC pin pair for the S3 board?

[image]

clydebarrow, 2024-12-15:
> the log states that the usb_uart sits on uart0.

No it doesn't, and it can't anyway - UART0 is a completely different peripheral, built into the ESP, "UART Channel 0" is what your USB device is meant to implement.

> do you mean, verbose level under logger

yes. Though you already have that, but you have not set the logger uart to UART0. Uncomment the `hardware_uart` line and remove the `baud_rate` line. You will not see all the logs over WiFi, you need to capture via UART0. Once it's working that's not so important, but you're not there yet.

And here is the problem:

```
[21:01:09][E][component:164]: Component usb_host set Error flag: No CDC-ACM device found
```

The STM is evidently not implementing the CDC-ACM spec. The verbose logging will dump the descriptors which will shed some light on it once you get the logging working properly.

ruteclrp, 2024-12-16:
Clive, you have to help me here with something which may be obvious, just not to me. 
I have been trying to get the log output to my connected laptop via the usb (com). This is the one which defaults to the JTAG if I do not specify uart0 in the logger section. 
However, how do I get the uart0 routed to my laptop?

clydebarrow, 2024-12-16:
You either need a board with two USB connectors, one for the built-in USB and one for UART0 via an on-board interface chip, or you need a separate USB-serial adapter to wire to your board's UART pins.

clydebarrow, 2024-12-16:
I just updated the external component so that the USB config dump _will_ work over WiFi, so that might make things easier.

ruteclrp, 2024-12-16:
> You either need a board with two USB connectors, one for the built-in USB and one for UART0 via an on-board interface chip, or you need a separate USB-serial adapter to wire to your board's UART pins.

My board is a 2 port board. However, connecting to either of those ports when the board is running does not allow access the to log. Somehow uart0 is not routed the the usb-uart bridge. Not specifying the hardware_uart routes the logger to JTAG which is available.
Was thinking that I needed a manual usb-serial adapter to connect to the uart0 pins - which seems is required for this board.

Anyway, will try to dump the log via wifi now you have made the update to the component.

Reverting.

ruteclrp, 2024-12-16:
[logs_dvi-lv12_logs(2).txt](https://github.com/user-attachments/files/18155068/logs_dvi-lv12_logs.2.txt)

Here is the extended log of the usb connection. I repeated plugging in the device a few times. I hope this makes sense to you.

clydebarrow, 2024-12-16:
I just pushed a commit that should make it more flexible about identifying devices that resemble CDC-ACM but don't actually match the spec, which is what your device looks like.

ruteclrp, 2024-12-17:
Hi Clyde

Happy to report that we are in business. Had to remove the dummy receiver but expect that is as intended. 
Thanks for your great support and patience with my ignorance.

clydebarrow, 2024-12-17:
Awesome!