esphome/feature-requests #2987: [ST7701S] Add power saving mode

brambo123, 2024-12-14:
**Describe the problem you have/What new integration you would like**

I would like to see an option to put the ST7701S display driver in 'Sleep in' mode.
In this mode, the DC/DC converter, internal oscillator and panel driver circuit are stopped.

Currently there is already a pause/resume function from LVGL including a function to prevent burn-in of the LCD.
But if the screen is off anyway, wouldn't it be better to just turn it off?

**Please describe your use case for this integration and alternatives you've tried:**

I would like to use a couple of Guidance ESP32-S3-4848S040 screens as a thermostat.
Low power consumption means less heat and therefore better temperature measurement.

**Additional context**

I tested a very simple code:
```
const uint8_t SLEEP_IN = 0x10;
const uint8_t SLEEP_OUT = 0x11;
void ST7701S::set_sleep(bool sleep) {
  if (sleep) {
    this->write_command_(SLEEP_IN);
  } else {
    this->write_command_(SLEEP_OUT);
  }
}
```
(And removed `this->spi_teardown();` from `ST7701S::write_init_sequence_()`)
Simply called at the same time as turning on/off the backlight and it works perfectly fine.
But just throwing in some code is not enough for a neat integration.

USB power consumption:
Backlight off, LVGL pause with snow: +- 90mA
Backlight off, ST7701S 'Sleep in' mode, LVGL pause (without snow, as screen is powered down anyway): +- 60mA

clydebarrow, 2024-12-16:
The `spi_teardown()` is there because some boards have the SPI pins reused for display driver pins, so have to be released after the chip is setup and before the RGB driver is configured.

brambo123, 2026-10-05:
> The `spi_teardown()` is there because some boards have the SPI pins reused for display driver pins, so have to be released after the chip is setup and before the RGB driver is configured.

The new mipi drivers seems not to have spi_teardown anymore.
Perhaps it can be added after all?

clydebarrow, 2026-10-08:
Yes, it could be added. Putting the display into sleep mode obviously requires the SPI channel to be working so the two would be incompatible