esphome/feature-requests #2989: ESPHome can't play mp3 from SD

KMTsvetanov, 2024-12-15:
### The problem

I have a ESP32-S3 and I added a SD to it.

Tried thise setup:

The rest of the yaml file is working!

### Which version of ESPHome has the issue?

2024.11.3 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  name: esphome-web-a60990
  friendly_name: ESPHome Web a60990

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

i2c:
  id: i2c_id_1
  sda: GPIO6       # I2C SDA pin
  scl: GPIO5       # I2C SCL pin
  scan: false

# Define the PN532 NFC/RFID reader (I2C connection)
pn532_i2c:
  id: pn532_i2c_id_1
  i2c_id: i2c_id_1      # Reference the above I2C bus (default is i2c_0)
  update_interval: 1s    # Poll for tags every 1 second

  on_tag:
    then:
      - lambda: |-
          id(tag_event_state).publish_state("Tag Scanned");  
      # - homeassistant.tag_scanned: !lambda 'return x;'  # Send the tag ID to Home Assistant    
      - switch.turn_on: vibration_motor     # Turn on the motor when tag is scanned
      - delay: 1s                           # Wait for 1 second
      - switch.turn_off: vibration_motor    # Turn off motor after 1 second    
      # - lambda: |-
      #     std::string tag_id = x;
      #     std::string path = "/001/";  // Folder name based on tag
      #     // Play the first MP3 in the selected playlist
      #     id(my_audio_player).play_mp3(path + "001.mp3");
      #     # Play the MP3 file from the SD card
      # - media_player.play
      # - media_player.play_media:
      #     id: my_audio_player
      #     media_url: "file:///sd/001.mp3"  # Corrected file path      
      - media_player.play_media:
          id: my_audio_player
          media_url: 'file:///sdcard/001.mp3'          
          # media_id: "/sdcard/001.mp3"
          # media_type: "music"        
  on_tag_removed:
    then:      
      - lambda: |-
          id(tag_event_state).publish_state("Tag Removed");
      - switch.turn_off: vibration_motor    # Optional feedback: Turn off motor
      # - homeassistant.tag_scanned: !lambda 'return x;'  # Send the tag ID to Home Assistant when removed   
      - media_player.pause  #stop   

# Define a global text sensor to show tag scan events in Home Assistant
text_sensor:
  - platform: template
    id: tag_event_state
    name: "Tag Scan Event"  # This is the name displayed in Home Assistant
    lambda: |-
      return id(tag_event_state).state;
    update_interval: never  # No need for an update interval

output:
  # "Vibration Motor"
  - platform: gpio
    pin: GPIO4       # Choose the GPIO pin connected to the gate/base of the MOSFET/transistor
    id: motor_control_id_1
    inverted: false    # If the motor runs when the GPIO pin is low, keep this as true

switch:
  # "Vibration Motor"
  - platform: output
    id: vibration_motor
    name: "Vibration Motor"
    output: motor_control_id_1    

# Define the I²S audio interface
i2s_audio:
  id: my_audio_interface
  i2s_lrclk_pin: GPIO17  # LCK or Word Select (WS / LRCK)
  i2s_bclk_pin: GPIO16   # BCK or Bit Clock (BCK)
  # i2s_mclk_pin: GPIO0    # SCK or Master Clock (MCLK), optional, only if needed by your DAC

# Wi-Fi Signal Strength Sensor
sensor:
  - platform: wifi_signal
    name: "Wi-Fi Signal Strength"
    update_interval: 60s  # Update every 60 seconds (you can adjust this as needed)    

# SPI configuration for ESP32-S3
spi:
  id: my_spi_bus     # ID for the custom SPI bus
  clk_pin: GPIO36     # SCK (Clock)
  mosi_pin: GPIO35    # MOSI (Master Out Slave In)
  miso_pin: GPIO37    # MISO (Master In Slave Out)
  interface: hardware # Hardware SPI

# SPI device for SD card
spi_device:
  id: sd_spi_device       # ID for the SPI device (SD card)
  cs_pin: GPIO38         # Chip Select pin for SD card (CS)
  spi_id: my_spi_bus     # Use the previously defined SPI bus
  spi_mode: mode3        # Use SPI mode 3 (default, works for most SD cards)
  bit_order: msb_first   # Default bit order (MSB first)

# Single media_player configuration
media_player:
  - platform: i2s_audio
    id: my_audio_player
    name: "ESPHome I2S Media Player"
    i2s_audio_id: my_audio_interface  # Reference the I²S audio interface
    dac_type: external                # Use an external DAC
    mode: stereo                      # Set to stereo/mono mode
    i2s_dout_pin: GPIO15              # DIN or Data Out (DOUT)
```

### Anything in the logs that might be useful for us?

[image]

_No response_

KMTsvetanov, 2024-12-17:
Did I understand correctly that ESPHome does not have a SD functionality?
I can play music that is passed from Home Assistant using WIFI and played with an external DAC, but not if the music is stored in a local SD connected to my ESP32?

I think that that can be a really simple functionality. Strange...
Can you help me understand this? Thank you!

ssieb, 2024-12-17:
esphome does not have any SD card handling.

KMTsvetanov, 2024-12-17:
Is there any other option that I can do?

ssieb, 2024-12-17:
It's something will likely get added eventually.  There's a PR for embedding files in the flash if you don't need to change them often.
https://github.com/esphome/esphome/pull/7320

KMTsvetanov, 2024-12-17:
Thank you. I will be following the process

KayOhtie, 2025-02-19:
While this works for small audio files, like soundFX for the Voice PE, it doesn't work as well if, say, you wanted to build something with an SD card to play longer audio files without having to rely on wifi signal quality for streaming. For instance, my Voice PE is unreliable at streaming long audio where I need it to. Having an SD card with a FLAC file of the long audio (hour+) to play back from would eliminate that issue entirely.