esphome/feature-requests #2995: Support for Sensirion SEN6X sensors

tuct, 2024-12-22:
**Describe the problem you have/What new integration you would like**

No official support for the new SEN6X sensors from Sensirion

**Please describe your use case for this integration and alternatives you've tried:**

Ability to use the new Sensors

**Additional context**

The Sensors are currently available via evaluation kits but will be released soon (Q2/2025)

I managed to "port" the existing SEN5X component here:
https://github.com/tuct/esphome_external_components/tree/main
But it only supports the SEN66 sensor and just works, not fully tested for now

nikolai2111, 2024-12-24:
This would be very helpful and perhaps Sensirion will be generous enough to maintain this part in the future.

Scarpetta3, 2025-01-31:
I am very interested too.

Regards
Matteo

tuct, 2025-03-15:
I am using by implementation for about a month plus now, also with the released version of the sensor!
Hope we get an official support soon!

I the meantime, I published my AirQuality Monitor based on the SEN66 and esphome here:
https://www.printables.com/model/1203658-aqmood-a-slightly-emotional-air-quality-monitor
and
here:
https://github.com/tuct/aqMood

iBobik, 2025-03-18:
Datasheet: https://sensirion.com/media/documents/FAFC548D/6731FFFA/Sensirion_Datasheet_SEN6x.pdf

chris-in-texas, 2025-04-26:
Just an FYI that eval units are now shipping as well or have been for a bit too.  You can purchase them direct or from your vendors.

tuct, 2025-04-26:
If you are in Europe, have some spare units and sell them here:
https://www.ebay.at/itm/388221079874?mkcid=16&mkevt=1&mkrid=5221-175123-2357-0&ssspo=o3rdo8zir_c&sssrc=2047675&ssuid=o3rdo8zir_c&widget_ver=artemis&media=COPY

tuct, 2025-04-26:
No eval units ,released Product!

chris-in-texas, 2025-04-26:
> No eval units ,released Product!

That's great news, I hope to see them over here then as well at some point in quantity!  They have been "coming soon" for about 18 months now.

cracrama, 2025-05-01:
Additional support for this sensor would be great. Especially that SEN 66 has atmospheric pressure sensor according Datasheet.

daradib, 2025-05-10:
> they did de-rated the life from 10+ years to 6+ years now between the Sen55 and Sen66

According to the data sheet, the lifetime is still 10+ years, excluding the (not yet released) formaldehyde sensor on the SEN68 which is indeed limited to 6+ years.

> atmospheric pressure sensor

I don't think there's a sensor, but a pressure value or altitude value can be provided to the SEN66 for CO2 pressure compensation, similar to what [ESPHome supports on the SCD4X](https://esphome.io/components/sensor/scd4x.html#configuration-variables) as well as manual calibration.

Also, it looks like the SEN5x supports periodic fan clean and manual fan clean in measurement mode, but the SEN6X does not support periodic fan clean and only supports manual fan clean in idle mode.

JirikP, 2025-06-10:
Hi,
is this resolved or not? I got lost in all the issues mentioning this feature. I do have product based on SEN66 and waiting for official support.

NuGundam, 2025-06-18:
I ordered one today I hope support comes in soon.

Scarpetta3, 2025-06-18:
Hi,
where do you order it? Because from the official website I can't see it available right now (I mean not the evaluation kit ones).

Thank You
Regards
Matteo

Scarpetta3, 2025-08-27:
Hi,
right now it can be purchased on the official site. ([Link](https://sensirion.com/products/catalog/SEN66))

Hope full support for esphome as soon as possible.

regards
Matteo

closed 2026-02-27 by swoboda1337