esphome/feature-requests #2996: HX711 reading both channels A and B

xsysel, 2024-12-22:
**Describe the problem you have/What new integration you would like**

I have two Weighing scales with HX711 for my reverse osmosis system, so I need to get data continously from both of them. 

**Please describe your use case for this integration and alternatives you've tried:**

There is already 2 channels in HX711, but only one is possible to use. Workaround is to have two HX711 connected to one ESP

**Additional context**

JCNuttall, 2024-12-30:
Agreed, being able to access both channels would make sense. I'm trying to weigh two gas bottles and as it sits ill have to use two HX711's where one should have sufficed

DjordjeMandic, 2025-01-07:
I've ordered few hx711 modules. When they arrive I'm gonna create pull request and test it with small voltages as classic adc. Since I do not have load cells, I'm gonna need you to test it with load cells

DjordjeMandic, 2025-01-25:
Waiting for https://github.com/esphome/esphome/pull/8138 to get merged, when it gets merged I can continue to work on this.

bagemihldotpro, 2025-04-16:
Hi @DjordjeMandic, 
I'm also interested in using both channels of the hx711 simultaneusly. Do you have a schedule for this?
Or can you give me a quick hint on what part of the code has to be changed? Maybe I can help.

Thanks and best regards!

DjordjeMandic, 2025-04-17:
@bagemihldotpro I have plans to rewrite most of HX711 component to fully implement all features but I've been really busy for last 2 months due to unexpected work. To properly implement this, settling time must be respected. Channel is changed by changing the gain **after reading**, take a look at datasheet.

As quick fix it would be quickest to implement action to change the value of protected member `HX711Gain gain_`. Keep in mind that after `gain_` value change, new channel will actually be set in chip AFTER the NEXT measurement. So you must do a dummy measurement. Also you must wait some time after that dummy merasurement for adc to settle. (400ms) for example.

I might continue to work on this this or next month

DjordjeMandic, 2025-05-25:
If anyone is able to test this PR: https://github.com/esphome/esphome/pull/8899, feedback would be greatly appreciated.

jemarr, 2025-06-15:
Hi Djordje,

Is it possible to read both channels A and B simultaneously of two HX711 boards connected on diferent CLK an Dout pins to ESP32 with your software ???

My chalenge is to read 4 TZT sensors with just one ESP32.

Thanks

DjordjeMandic, 2025-06-16:
> Hi Djordje,
> 
> Is it possible to read both channels A and B simultaneously of two HX711 boards connected on diferent CLK an Dout pins to ESP32 with your software ???

Hi, it should be possible if it was possible with current implementation of HX711

DjordjeMandic, 2025-06-24:
> Hi Djordje again.
> 
> I've just received HX711 boards and sensors. Reading channel A works fine. Then I tested your PR and I've got compiling errors. Attached are the yaml file and errors.
> 
> ESPHOME Version: 2025.3.3
> Board ESP32-C3
> 
> [edu2.txt](https://github.com/user-attachments/files/20834681/edu2.txt)
> 
> [image]

Hi, sorry for late reply, I'm on vacation currently. 

Can you try the same with latest release? There are some changes in esphome core that are used but do not exist in 2025.3.3.

DjordjeMandic, 2025-07-04:
@jemarr 

Pushed few commits to esphome/esphome#8899, using 2025.7.0-dev it works on my end. Compiles without warnings.

jemarr, 2025-07-05:
Hi Djordje

Finally I got my Esp32-C3 running with 4 HX711 boards on diferent Gpio pins, so don't need to use Channel B of these boards.

Nevertheless, if I have a moment I'll try your software.

Thanks again for your job.

xsysel, 2025-08-07:
@DjordjeMandic 

I spend almost 6 hours to find what is wrong... I am testing it on two boards red and green as on this link https://www.aliexpress.com/item/1005008022290648.html. I bought red maybe 15 years ago, green one last year. 

I started with **green** board
I have found it does/doesn't based on different power adapter, cable etc, so I was suspicious about power itself. But even with one cable and adapter it sometimes work and sometimes not. So I decide to connect scope to the DO (red on scope) pin and E+ (yellow on scope) to see if chip is active. As you can see on below photos, DO sometimes stays on HIGH and therefore HX711 is not active. 

[image]

[image]

I was not able to reproduce it with current HX711 library - it works all the time.

Then I was testing **red** board

It works well with current library, with all adapters. When I compile your library and send it thru OTA then it goes to HIGH on DO and stays there forever. If I compile and send current library DO goes to the LOW and works perfectly.

When I looked closer to the boards green on has DO directly to HX711 chip, whilst red one has some resistor in the way.

I hope I described it understandable - there is something what blocks proper boot of HX711 - on green board sometimes when power is connected, on red boards always even new program is sent.

BTW: I have found green board doesn't have E- connected to GND as it is described here: https://www.eevblog.com/forum/projects/how-can-a-chip-(hx711)-work-without-a-ground-pin-connected/

DjordjeMandic, 2025-10-06:
@xsysel I'm currently busy so looking at source will have to wait few days. Can you in meantime try the `measurement_ready_timeout` option? Maybe increase it to `10000ms`?

DjordjeMandic, 2025-10-23:
@xsysel I'm gonna go thru my code again, from start to finish. Thank you for pointing out issues in my code. I will do tests with real load cell and one of my boards

xsysel, 2025-10-23:
@DjordjeMandic I will check it later today. I looked into changes and I am not sure if you saw my comment here https://github.com/esphome/feature-requests/issues/2996#issuecomment-3393613610 There is mistake in pin config. I am currious how it is possible it works on your side.

xsysel, 2025-12-05:
@DjordjeMandic 
good job, it works with ESP8266 now!

I am facing another issue. I am getting same reading on both channels after some time of running. As you can see in log below - when there is not ready flag, powered down and up then it use same gain for both channels. It looks "last_gain" is 32 (due to channel B reading) before crash and it is loaded for channel A after crash.. Can you please take a look to it?

> [22:30:51.542][V][hx711:166]: 'HX711 Value': Updating
> [22:30:51.598][V][hx711:305]: 'HX711 Value': settled
> [22:30:51.610][V][hx711:389]: 'HX711 Value': last_gain=x64; force=NO
> [22:30:51.610][V][hx711:415]: 'HX711 Value': gain (x64) changed to x32
> [22:30:51.610][V][hx711:301]: 'HX711 Value': Settling
> [22:30:51.610][V][hx711:130]: 'HX711 Value': Got value 164068 (gain x64)
> [22:30:51.619][V][sensor:081]: 'HX711 Value': Received new state 164068.000000
> [22:30:51.619][D][sensor:133]: 'HX711 Value': Sending state 164068.00000  with 0 decimals of accuracy
> [22:30:54.612][V][hx711:305]: 'HX711 Value': settled
> [22:30:54.670][V][hx711:389]: 'HX711 Value': last_gain=x32; force=NO
> [22:30:54.670][V][hx711:415]: 'HX711 Value': gain (x32) changed to x64
> [22:30:54.670][V][hx711:301]: 'HX711 Value': Settling
> [22:30:54.670][D][hx711:359]: 'HX711 Channel B': Got Channel B value 35806 (gain x32)
> [22:30:54.670][V][sensor:081]: 'HX711 Channel B': Received new state 35806.000000
> [22:30:54.670][D][sensor:133]: 'HX711 Channel B': Sending state 35806.00000  with 0 decimals of accuracy
> [22:30:55.543][V][hx711:166]: 'HX711 Value': Updating
> [22:30:57.630][V][hx711:305]: 'HX711 Value': settled
> [22:30:57.647][W][hx711:375]: not ready
> [22:30:57.647][W][component:298]: hx711.sensor set Warning flag: not ready
> [22:30:57.647][V][sensor:081]: 'HX711 Value': Received new state nan
> [22:30:57.647][D][sensor:133]: 'HX711 Value': Sending state nan  with 0 decimals of accuracy
> [22:30:57.647][D][hx711:223]: Restarting
> [22:30:57.647][W][hx711:283]: 'HX711 Value': Stopping poller
> [22:30:57.647][D][hx711:289]: 'HX711 Value': powered down
> [22:30:57.656][D][hx711:252]: 'HX711 Value': powered up
> [22:30:57.656][V][hx711:258]: 'HX711 Value': setting gain to x32
> [22:30:57.732][V][hx711:389]: 'HX711 Value': last_gain=x128; force=YES
> [22:30:57.742][V][hx711:415]: 'HX711 Value': gain (x128) changed to x32
> [22:30:57.742][V][hx711:301]: 'HX711 Value': Settling
> [22:31:00.733][W][component:337]: hx711.sensor cleared Warning flag
> [22:31:00.739][V][hx711:305]: 'HX711 Value': settled
> [22:31:00.739][D][hx711:309]: 'HX711 Value': Starting poller
> [22:31:00.739][V][scheduler:144]: Scheduler interval for update is 4000ms, offset 617ms
> [22:31:01.355][V][hx711:166]: 'HX711 Value': Updating
> [22:31:01.368][V][hx711:389]: 'HX711 Value': last_gain=x32; force=NO
> [22:31:01.368][V][hx711:301]: 'HX711 Value': Settling
> [22:31:01.368][D][hx711:359]: 'HX711 Channel B': Got Channel B value 35597 (gain x32)
> [22:31:01.368][V][sensor:081]: 'HX711 Channel B': Received new state 35597.000000
> [22:31:01.375][D][sensor:133]: 'HX711 Channel B': Sending state 35597.00000  with 0 decimals of accuracy
> [22:31:04.354][V][hx711:305]: 'HX711 Value': settled
> [22:31:05.354][V][hx711:166]: 'HX711 Value': Updating
> [22:31:05.368][V][hx711:389]: 'HX711 Value': last_gain=x32; force=NO
> [22:31:05.368][V][hx711:130]: 'HX711 Value': Got value 35673 (gain x32)
> [22:31:05.368][V][sensor:081]: 'HX711 Value': Received new state 35673.000000
> [22:31:05.378][D][sensor:133]: 'HX711 Value': Sending state 35673.00000  with 0 decimals of accuracy
> [22:31:05.379][D][hx711:359]: 'HX711 Channel B': Got Channel B value 35673 (gain x32)
> [22:31:05.394][V][sensor:081]: 'HX711 Channel B': Received new state 35673.000000
> [22:31:05.394][D][sensor:133]: 'HX711 Channel B': Sending state 35673.00000  with 0 decimals of accuracy

xsysel, 2025-12-10:
I just recompile and upload. It looks better after restarting:

> [20:09:19.112][V][hx711:168]: 'HX711 Value': Updating
[20:09:19.448][V][hx711:311]: 'HX711 Value': settled
[20:09:19.460][V][hx711:395]: 'HX711 Value': last_gain=x64; force=NO
[20:09:19.460][V][hx711:421]: 'HX711 Value': gain (x64) changed to x32
[20:09:19.460][V][hx711:307]: 'HX711 Value': Settling
[20:09:19.460][V][hx711:130]: 'HX711 Value': Got value 164042 (gain x64)
[20:09:19.472][V][sensor:081]: 'HX711 Value': Received new state 164042.000000
[20:09:19.472][D][sensor:133]: 'HX711 Value': Sending state 164042.00000  with 0 decimals of accuracy
[20:09:22.459][V][hx711:311]: 'HX711 Value': settled
[20:09:22.470][W][hx711:381]: not ready
[20:09:22.470][W][component:298]: hx711.sensor set Warning flag: not ready
[20:09:22.470][V][hx711:307]: 'HX711 Value': Settling
[20:09:22.470][D][hx711:365]: 'HX711 Channel B': Got Channel B value nan (gain x32)
[20:09:22.483][V][sensor:081]: 'HX711 Channel B': Received new state nan
[20:09:22.483][D][sensor:133]: 'HX711 Channel B': Sending state nan  with 0 decimals of accuracy
[20:09:22.495][D][hx711:225]: Restarting
[20:09:22.495][W][hx711:289]: 'HX711 Value': Stopping poller
[20:09:22.495][D][hx711:295]: 'HX711 Value': powered down
[20:09:22.504][D][hx711:258]: 'HX711 Value': powered up
[20:09:22.504][V][hx711:264]: 'HX711 Value': setting gain to x64
[20:09:22.577][V][hx711:395]: 'HX711 Value': last_gain=x128; force=YES
[20:09:22.581][V][hx711:421]: 'HX711 Value': gain (x128) changed to x64
[20:09:22.581][V][hx711:307]: 'HX711 Value': Settling
[20:09:25.587][W][component:337]: hx711.sensor cleared Warning flag
[20:09:25.603][V][hx711:311]: 'HX711 Value': settled
[20:09:25.603][D][hx711:315]: 'HX711 Value': Starting poller
[20:09:25.603][V][scheduler:144]: Scheduler interval for update is 4000ms, offset 717ms
[20:09:26.316][V][hx711:168]: 'HX711 Value': Updating
[20:09:26.324][V][hx711:395]: 'HX711 Value': last_gain=x64; force=NO
[20:09:26.324][V][hx711:421]: 'HX711 Value': gain (x64) changed to x32
[20:09:26.324][V][hx711:307]: 'HX711 Value': Settling
[20:09:26.324][V][hx711:130]: 'HX711 Value': Got value 163936 (gain x64)
[20:09:26.335][V][sensor:081]: 'HX711 Value': Received new state 163936.000000
[20:09:26.335][D][sensor:133]: 'HX711 Value': Sending state 163936.00000  with 0 decimals of accuracy
[20:09:29.322][V][hx711:311]: 'HX711 Value': settled
[20:09:29.335][V][hx711:395]: 'HX711 Value': last_gain=x32; force=NO
[20:09:29.335][V][hx711:421]: 'HX711 Value': gain (x32) changed to x64
[20:09:29.335][V][hx711:307]: 'HX711 Value': Settling
[20:09:29.335][D][hx711:365]: 'HX711 Channel B': Got Channel B value 35891 (gain x32)
[20:09:29.347][V][sensor:081]: 'HX711 Channel B': Received new state 35891.000000
[20:09:29.350][D][sensor:133]: 'HX711 Channel B': Sending state 35891.00000  with 0 decimals of accuracy

It gets nan and next reading is OK.

But I have found some other strange behavior. It gets strange minus value for channel A and then channel B with A value. After that all readings are OK. Maybe change settling and update interval?

> [22:38:13.374][V][hx711:168]: 'HX711 Value': Updating
[22:38:13.419][V][hx711:311]: 'HX711 Value': settled
[22:38:13.434][V][hx711:395]: 'HX711 Value': last_gain=x64; force=NO
[22:38:13.434][V][hx711:421]: 'HX711 Value': gain (x64) changed to x32
[22:38:13.434][V][hx711:307]: 'HX711 Value': Settling
[22:38:13.434][V][hx711:130]: 'HX711 Value': Got value 163475 (gain x64)
[22:38:13.443][V][sensor:081]: 'HX711 Value': Received new state 163475.000000
[22:38:13.443][D][sensor:133]: 'HX711 Value': Sending state 163475.00000  with 0 decimals of accuracy
[22:38:16.439][V][hx711:311]: 'HX711 Value': settled
[22:38:16.455][V][hx711:395]: 'HX711 Value': last_gain=x32; force=NO
[22:38:16.455][V][hx711:421]: 'HX711 Value': gain (x32) changed to x64
[22:38:16.455][V][hx711:307]: 'HX711 Value': Settling
[22:38:16.455][D][hx711:365]: 'HX711 Channel B': Got Channel B value 36094 (gain x32)
[22:38:16.455][V][sensor:081]: 'HX711 Channel B': Received new state 36094.000000
[22:38:16.460][D][sensor:133]: 'HX711 Channel B': Sending state 36094.00000  with 0 decimals of accuracy
[22:38:17.375][V][hx711:168]: 'HX711 Value': Updating
[22:38:19.443][V][hx711:311]: 'HX711 Value': settled
[22:38:19.455][V][hx711:395]: 'HX711 Value': last_gain=x64; force=NO
[22:38:19.455][V][hx711:421]: 'HX711 Value': gain (x64) changed to x32
[22:38:19.455][V][hx711:307]: 'HX711 Value': Settling
**[22:38:19.455][V][hx711:130]: 'HX711 Value': Got value -129788 (gain x64)**
[22:38:19.468][V][sensor:081]: 'HX711 Value': Received new state -129788.000000
[22:38:19.468][D][sensor:133]: 'HX711 Value': Sending state -129788.00000  with 0 decimals of accuracy
[22:38:21.376][W][hx711:164]: 'HX711 Value': Previous update in progress
[22:38:22.459][V][hx711:311]: 'HX711 Value': settled
[22:38:22.484][V][hx711:395]: 'HX711 Value': last_gain=x32; force=NO
[22:38:22.484][V][hx711:421]: 'HX711 Value': gain (x32) changed to x64
[22:38:22.484][V][hx711:307]: 'HX711 Value': Settling
**[22:38:22.484][D][hx711:365]: 'HX711 Channel B': Got Channel B value 163512 (gain x32)**
[22:38:22.490][V][sensor:081]: 'HX711 Channel B': Received new state 163512.000000
[22:38:22.490][D][sensor:133]: 'HX711 Channel B': Sending state 163512.00000  with 0 decimals of accuracy

As you can see on this image, there are multiple very strange values during 3 hours run..

[image]

There is log file of this run: [logs_vaha_run (4).txt](https://github.com/user-attachments/files/24088571/logs_vaha_run.4.txt)

DjordjeMandic, 2025-12-10:
Are there any of these negative values when using default hx711 component?

xsysel, 2025-12-11:
I tried to test your library with one channel only and I am getting this failure during compilation:

> src/esphome/components/hx711/hx711.cpp: In member function 'void esphome::hx711::HX711Sensor::power_cycle_restart_(bool)':
src/esphome/components/hx711/hx711.cpp:227:28: error: 'struct esphome::hx711::HX711Sensor::HX711StateFlags' has no member named 'channel_b_sensor_read_pending'
  227 |   this->hx711_state_flags_.channel_b_sensor_read_pending = false;
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/vaha/src/esphome/components/hx711/hx711.cpp.o] Error 1

xsysel, 2025-12-11:
> > I compiled it today morning and let it go for whole day. As you can see there were multiple wrong values.
> > Values are around 320k ( 2x readed values, so like GAIN128), around 35k (like GAIN32 - channel B is still connected to load cells) and then -259k, -128k, -64k and -32k
> > <img alt="Image" width="1407" height="405" src="https://private-user-images.githubusercontent.com/122181130/525501015-8f80abad-6fc7-47f5-9774-0a28df543909.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjU0ODM5ODAsIm5iZiI6MTc2NTQ4MzY4MCwicGF0aCI6Ii8xMjIxODExMzAvNTI1NTAxMDE1LThmODBhYmFkLTZmYzctNDdmNS05Nzc0LTBhMjhkZjU0MzkwOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMjExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTIxMVQyMDA4MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMTg2MzEyMTJkMTFiOGQzZDJmOTQ3NDU3MDc1ODFmMmE0ZTYzZjY1MjQzOWRhMmJmZTcxMTE1MzI3Y2MyNzBlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cPYip5LPir99Z_E8RnODd8ZhFWnlMFPPn-ydUTGrU2U">
> 
> This is with default hx711 component from latest esphome release? So youre getting weird values even with default component? If so, might be that you have signal integrity issues

Yes, with latest esphome release. I have around 3cm of wire (soldered) between HX711 and Wemos D1 mini, so I would say minimum of possible noise. It is interesting, that latest release shows values to +-320k, but with your code it is in milions and always below zero. So I guess it is not due to noise, but maybe different style of reading?

xsysel, 2025-12-11:
I got this error second time with #8899 It is marked as failed and all stops. It needs to be manually restarted to start whole reading again.

> [20:54:57.761][V][hx711:168]: 'HX711 Value': Updating
[20:54:57.774][W][hx711:381]: not ready
[20:54:57.774][W][component:298]: hx711.sensor set Warning flag: not ready
[20:54:57.774][V][sensor:081]: 'HX711 Value': Received new state nan
[20:54:57.774][D][sensor:133]: 'HX711 Value': Sending state nan  with 0 decimals of accuracy
[20:54:57.784][D][hx711:225]: Restarting
[20:54:57.784][W][hx711:289]: 'HX711 Value': Stopping poller
[20:54:57.784][D][hx711:295]: 'HX711 Value': powered down
[20:54:57.795][D][hx711:258]: 'HX711 Value': powered up
[20:54:57.795][V][hx711:264]: 'HX711 Value': setting gain to x64
[20:54:57.867][V][hx711:395]: 'HX711 Value': last_gain=x128; force=YES
[20:54:57.881][V][hx711:421]: 'HX711 Value': gain (x128) changed to x64
[20:54:57.881][W][hx711:429]: 'HX711 Value': Final dout err; 0x00e04ffb
[20:54:57.881][E][component:315]: hx711.sensor set Error flag: power-up can't set gain
[20:54:57.890][E][component:211]: hx711.sensor was marked as failed
[20:54:57.890][E][hx711:340]: 'HX711 Value' failed, automatically powered down

DjordjeMandic, 2025-12-11:
> Yes, with latest esphome release. I have around 3cm of wire (soldered) between HX711 and Wemos D1 mini, so I would say minimum of possible noise. It is interesting, that latest release shows values [...]

Can you add 1k-10k resistor between DOUT/DT and VCC on HX711 module? I don't see any more bugs in code, these +-320k values are mostly signal integrity issues during read (dt pin state).

Before continuing with testing of my pr @esphome/esphome#8899, lets first fix issue with +-320k on original hx711 component...

xsysel, 2025-12-16:
@DjordjeMandic 
I was testing it whole weekend with not so much success. I added 10k resistor to DOUT - nothing changed, I added second 10k to CLK - no change. I still receive +-320k errors on original hx711 component. 
I decided to hook it on scope to see what is going on. I am not able to catch exactly moment when issue appears unfortunately. But I have spotted interesting thing. Delay between 24th and 25th pulse on PD_SCK pin does vary. If I am connected to logging then it has bigger delay. I know it is in spec per datasheet, but maybe during some lag of WiFi it is longer and then wrong data are received?? Maybe?

[image]

[image]

You can see difference between original and PR. There is difference in 25th pulse length as well.

I looked into code and there is procedure to make 24 pulses and then rest. Can this be an issue? I can see it is same in original and PR, but what about to rewrite it to cycle all pulses in on loop?

DjordjeMandic, 2025-12-16:
> @DjordjeMandic
> I was testing it whole weekend with not so much success. I added 10k resistor to DOUT - nothing changed, I added second 10k to CLK - no change. I still receive +-320k errors on [...]

Last pulses are for gain selection, i guess that there is delay because of calculation of count of pulses or initialisation of for loop that does it. Delay is just because of cpu being busy. That is my guess.

Shielding digital lines would be overkill. Maybe issue is external like emi induced noise. I honestly have no idea why are those values happening

xsysel, 2025-12-17:
@DjordjeMandic I did test it little bit more and I am pretty sure it is something to do with timing.
- with no change to hardware there is significant difference between errors - ESP release shows +-320k, whilst PR shows errors in milions. 
- I rewrote it to make all pulses in one loop - change is visible on scope (nice delays and pulses 2nd-27th pulse), but no change to errors
- There is significant length variation on first pulse (I saw up to 25us) - it is strange as it is generated within same loop
- I commented out `InterruptLock lock;` line and there is no difference in errors(not less, not more) - I would say InterruptLock doesn't work on my ESP8266

I should receive ESP32 soon, so I will try to test it instead of my Wemos D1 mini...

xsysel, 2025-12-18:
Hi @DjordjeMandic 
I have found this topic https://github.com/bogde/HX711/issues/75#issuecomment-387313449 about reading HX711 and interrupts. There is note about GND to E- shortage on the PCB. I have remind myself that I have two types of boards with and withou this shortage.
All tests up to now were with red board with shortage and  resistor in the way of DOUT as I mentioned here: https://github.com/esphome/feature-requests/issues/2996#issuecomment-3163866441 I connected this board to ESP32 and I can see almost same issue as with Wemos.
I hooked up green board of HX711 to another Wemos and it runs from yesterday evening with one error only.

So it looks red board (shortage or resistor) has influence to errors. I will keep testing and I will inform you once I will have some result.

DjordjeMandic, 2025-12-25:
Nice catch, I'm waiting for results. Thank you for testing, it has been really helpful

xsysel, 2026-01-04:
Hello in 2026!
Here are results of my testing. Both devices were sitting next to each other in my workshop with minimum movement around. No movement with devices during test, just remote reflash.
Váha 2: (Wemos D1 mini + green board + 2 sets of load cells) 
Váha ESP32: (ESP32 + red board + 2 sets of load cells)

Both devices running released HX711 with Channel A only:
[image]

Both devices running PR#8899 with Channel A only:
[image]

Both devices running PR#8899 with Channel A and B:
[image]
[image]

So my conclusion is that some errors are still there with original library and this PR, it looks like PR is introducing little bit more of them. I did some analysis in Excel and you can see number of values from each 24h test for each entity. PR is introducing more unknown values - probably when "not ready" status appears. outofrange count are values more than 1000 out of nominal.

[image]

What do you think about that?

PS: I have found this warning during compilation today:

```
src/esphome/components/hx711/hx711.cpp:327:24: warning: 'void esphome::Component::mark_failed(const char*)' is deprecated: Use mark_failed(LOG_STR("static string literal")) instead. Do NOT use .c_str() from temporary strings. Will stop working in 2026.6.0 [-Wdeprecated-declarations]
  327 |       this->mark_failed("ready timeout");
      |       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from src/esphome/components/hx711/hx711.h:3,
                 from src/esphome/components/hx711/hx711.cpp:1:
src/esphome/core/component.h:165:8: note: declared here
  165 |   void mark_failed(const char *message) {
      |        ^~~~~~~~~~~
src/esphome/components/hx711/hx711.cpp: In member function 'void esphome::hx711::HX711Sensor::mark_failed_internal_(const char*)':
src/esphome/components/hx711/hx711.cpp:339:20: warning: 'void esphome::Component::mark_failed(const char*)' is deprecated: Use mark_failed(LOG_STR("static string literal")) instead. Do NOT use .c_str() from temporary strings. Will stop working in 2026.6.0 [-Wdeprecated-declarations]
  339 |   this->mark_failed(message);
      |   ~~~~~~~~~~~~~~~~~^~~~~~~~~
src/esphome/core/component.h:165:8: note: declared here
  165 |   void mark_failed(const char *message) {
      |        ^~~~~~~~~~~
```