esphome/feature-requests #3000: ESPHome Updates - need more granular options

PanMat, 2024-12-31:
ESPHome has a convenient feature "Update All" that is really handy to update firmware on all devices with single click. But as the number of devices are increasing on the home network (I've about ~60 devices) this feature needs some refining, And here is the crux of the problem:
1. With more devices, it takes much longer time to compile and update.
2. And this creates few problems as the devices may not update for many reasons such as - a) offline, b) configuration needs fixing due to ESPHome updates or c) browser window closed etc. 
3. After each major updates it creates few segments of devices that are updated, need fixing and just need to run update again.

It will be more practical to add few options for updates to ESPHome and here are few sugegstions:

1. Update All (current feature)
2. Check Configuration
3. Compile latest but do not upload
4. Re-upload all devices with the last compiled version

This will make it practical for each user to monitor and update depending on the size/time spent.

**Please describe your use case for this integration and alternatives you've tried:**

Currently the only recourse is to let the update process run for over an hour and then revisit the ESPHome page to see which devices did not update and look at the reasons.
**Additional context**

Tom advised me to create an issue here when I posted [this request](https://community.home-assistant.io/t/esphome-validate-all-option/798105/3) in the community forum.

lightingman117, 2024-12-31:
Literally came here to write out a similar request :)

I'd like to see 3 things:

1. [Modify Behavior] Update All
    - only updates devices not running current (latest installed version)
    - currently, update all will force update all devices whether they are up-to-date or not

2. [New button] Force update
    - updates all devices regardless of current firmware level [current functionality]

3. [Modify Behavior] multi-device update
    - compiles & uploads multiple devices at the same time
    - the compiler is multi-threaded, but it runs sequentially thus significant CPU downtime between uploads and compiles
    - Better to compile & upload in separate tasks
    - I imagine a compile dialog pops up & a upload dialog pops up
        - "compiling device abc" | "waiting for abc to compile"
        - "compiling device xyz" | "uploading abc"
        - "done compiling" | "uploading xyz"

f-fritz, 2025-01-04:
Maybe a configuration parameter to exclude specific sensors from automatic / batch updating? There might be "critical" devices that should never be auto-updated. Depending on the overall number of devices it might be more comfortable to exclude a few devices than to start all other updates manually / individually.

However, this functionality might be partially realised with the proposed compile but do not upload option as it could save a lot of time because I'd then only had to watch the uploads + installations but not the compilation.

BigG1234, 2025-01-17:
In addition,   It would also be nice to have a roll back function.     eg the previous compiled version is stored so that it could be flashed back onto the device,    as I've had once or twice where an update needs some work to resolve, but the device is offline until the issue is resolved, which I might not have time to do right away.