esphome/feature-requests #3005: Speed up compilation by sharing .o files for identical dev boards / devices?

f-fritz, 2025-01-04:
**Describe the problem you have/What new integration you would like**
I use a rather underpowered machine to host ESPHome. Compiling sensor firmware takes about 5min per device. I wonder if there is a way to share .o files across individual devices and only recompile distinct .c files to save time.

**Please describe your use case for this integration and alternatives you've tried:**
I have multiple sensors that are basically identical, i.e. use the same board and even have the same sensors attached. Even though they are identical except name, keys and ip address, it appears as if every single c file is recompiled for each sensor. I wonder if there is a way to maybe store .o files at a central location, e.g. for each individual type of development board and to only individually recompile distinct files.

**Additional context**

n/a

Thanks!

numo68, 2025-01-07:
Check https://ccache.dev . You will probably need to wrestle with the symlinks in `.platformio/packages/toolchain-xtensa-esp32/bin` or tweak around `~/.platformio/packages/framework-arduinoespressif32/platform.txt`, but it should be possible. I never tried it with esphome though, just in my day job with regular gcc and plain makefiles.

randybb, 2025-01-07:
this is a duplicate of https://github.com/esphome/feature-requests/issues/975
people tried that already, it is not possible as every device is in a separate platformio project and you can run various versions, so quite a complicated.

closed 2025-01-07 by randybb

numo68, 2025-01-07:
> it is not possible as every device is in a separate platformio project and you can run various versions, so quite a complicated.

ccache explicitly states in the docs that "Another reason to use ccache is that the same cache is used for builds in different directories. If you have several versions or branches of a software stored in different directories, many of the object files in a build directory can probably be taken from the cache even if they were compiled for another version or branch.".

AFAIK they are using hashes of at least the source with dependencies plus all the flags passed to the compilation so unless that differs - which in my `esphome -v compile` case does not seem so - all the `.pioenvs` should indeed be identifiable as same.

I can well understand that this is not something the ESPHome developers are willing to maintain and support (that would be more the job of pio anyway), I would not want to either :) , but the OP might still be interested in trying...

randybb, 2025-01-07:
There is a PR https://github.com/esphome/esphome/pull/6965 that is closed, but it looks like it was working. Maybe you can give it a try...