esphome/feature-requests #3009: Cumulative Pulse Width Sensor Request

patrick-kiwi, 2025-01-04:
**Describe the problem you have/What new integration you would like**
I’m working on a solar-diverter energy monitor, where I need to add the cumulative daily on-time of a pulsing signal read by a GPIO pin.  The pulse-widths to be summed are random and can be as short as 20 ms.

**Please describe your use case for this integration and alternatives you've tried:**
I’ve tried the “Pulse Meter Sensor” but that only counts the total number of pulses. I’ve also tried the “Pulse Counter Sensor” combined with the “Integration Sensor” but again, this sensor is integrating something with respect to time, which is different to simply adding the pulse-widths.  I thought I had the solution when I found the "Pulse Width Sensor" but that sensor only measures single discrete pulses, ie it doesn't accumulate.

**Additional context**
The source code of the “Pulse Meter Sensor” looks like it’s measuring pulse-widths, but there’s no way of currently getting the daily, hourly, etc. summed up pulse-width’s.  Alternatively, it looks like the "Pulse Width Sensor" functionality could be easily extended

ssieb, 2025-01-05:
Duplicate of https://github.com/esphome/feature-requests/issues/3008

patrick-kiwi, 2025-01-05:
Hi ssieb why did this get closed so quickly?  A cumulative pulse width sensor would be a useful general tool.

ssieb, 2025-01-05:
Sorry, you created several duplicate issues and with a quick look, this appeared to be exactly the same as 3008 which you closed.

lk-ek, 2025-04-19:
Interested in this (and works for me, thank you!); any chance in reviving the linked PR?

patrick-kiwi, 2025-04-21:
Hi there.  I'm glad it worked for you.  Just make sure you use the latest accumulate_8 branch as the earlier ones had a few bugs.  I did submit a pull request, but I'm not sure how to progress it.  I was using windows which seemed to create a weird error that I couldn't fix on the PR.

lk-ek, 2025-04-21:
Thanks pat. I've seen that PR and found your repository and even sent you a small pr for my pet feature. I'd be interested in helping to get this merged. Would you be open to create another PR, hopefully including mine :)?

patrick-kiwi, 2025-04-21:
Hi Lukas

I merged your request, and invited you collaborate.  Github's a bit of a struggle for me, as I'm not a professional programmer.  Perhaps you could handle to pull request into esphome?  It's worth noting that there is an existing esphome component called "duty_cycle" that I later discovered has similar functionality to pulse_width_accumulate.  THe duty_cycle component lacks frequency monitoring on the same gpio.  Your PR essentially outputs the duty cycle which is indeed useful.  In my opinion the duty_cycle component code is way more elegant than my clunky thing, but on the other hand the mutex approach does seem to give slightly better accuracy at high frequencies of ~10kHz according to my testing.

Cheers,
Patrick

________________________________
From: Lukas
Sent: Monday, April 21, 2025 11:17
To: esphome/feature-requests
Cc: pat; Author
Subject: Re: [esphome/feature-requests] Cumulative Pulse Width Sensor Request (Issue #3009)

[https://avatars.githubusercontent.com/u/198809900?s=20&v=4]lk-ek left a comment (esphome/feature-requests#3009)<https://github.com/esphome/feature-requests/issues/3009#issuecomment-2818009597>

Thanks pat. I've seen that PR and found your repository and even sent you a small pr for my pet feature. I'd be interested in helping to get this merged. Would you be open to create another PR, hopefully including mine :)?

lk-ek, 2025-04-21:
Thanks for the hint to duty_cycle. I'm not so sure whether it copes well with pulses of varying widths and hence a non-steady frequency, though. Also, the way you designed the ISR and used mutexes seem pretty well thought through. Both do seem to work for me.

I'm not a very proficient github user either, but can start another pull request and we'll see how it goes. What I did notice is that your esphome fork does not seem to be easily mergeable back into the original esphome though. I might need to rebase/redo the accumulation_9 branch to be able to create another PR. Hopefully I'll grasp what needs to be done in the next days.

all the best,
Lukas

patrick-kiwi, 2025-04-23:
Hi Lukas

Thanks, yes it was a bit of a nightmare finding the right place for those mutex critical sections.  If you change any of the mutex locations, then it messes up the accuracy at high frequency.   Just in case you need it for testing, this pulse generator might help.  Very precise timing from esp32 using the RMT peripheral  https://github.com/patrick-kiwi/esp32_custom_pulse/tree/main

Best of luck,

Cheers,
Patrick

________________________________
From: Lukas
Sent: Monday, April 21, 2025 14:36
To: esphome/feature-requests
Cc: pat; Author
Subject: Re: [esphome/feature-requests] Cumulative Pulse Width Sensor Request (Issue #3009)

[https://avatars.githubusercontent.com/u/198809900?s=20&v=4]lk-ek left a comment (esphome/feature-requests#3009)<https://github.com/esphome/feature-requests/issues/3009#issuecomment-2818329236>

Thanks for the hint to duty_cycle. I'm not so sure whether it copes well with pulses of varying widths and hence a non-steady frequency, though. Also, the way you designed the ISR and used mutexes seem pretty well thought through. Both do seem to work for me.

I'm not a very proficient github user either, but can start another pull request and we'll see how it goes. What I did notice is that your esphome fork does not seem to be easily mergeable back into the original esphome though. I might need to rebase/redo the accumulation_9 branch to be able to create another PR. Hopefully I'll grasp what needs to be done in the next days.

all the best,
Lukas