esphome/feature-requests #3014: Api disconnect action

fran6120, 2025-01-06:
**Describe the problem you have/What new integration you would like**

I have a solar battery powered esp32-c3 device that collects data from various sensors. Due to the variability of solar power availability, I need to manage the battery usage state.

When the battery voltage is very low, I use the deep_sleep component to preserve it. It works great, when the device goes into deep sleep mode, the state of the entities in Home Assistant retains the values.

Other times, when the battery voltage is low, but not nearly enough to shut down the device, to preserve the battery, I disable wifi with disable wifi action and then I can continue collecting data from sensors without losing filters and other time-based data, to then after a certain time, enable wifi and connect to home assistant.
That works fine, but Home Assistant shows the entities as unavailable (obvious thing while the device connection is not present).

So, this is the reason to submit a feature request, to implement an api.disconnet action that manually disconnects the device from the api just like the deep_sleep component does and that keeps the values ​​of the entity states in home assistant until they are updated on the next connection.

**Please describe your use case for this integration and alternatives you've tried:**

The use I want to give it is to manually manage the connection to keep the battery charge status at values ​​that allow collecting information from the sensors, from a mode of continuous updates, to a mode of very basic and periodic updates, passing through intermediate modes of variations in the periodicity of the updates.

**Additional context**

An example of the use I want to give it would be something similar to this:

``` yaml

on_some_:
  - if:
      condition: 
        - lambda: 'return id(battery).state <= ${some_threshold};'
      then:
        - deep_sleep.enter:
            id: control_deep_sleep
            sleep_duration: ${some_duration}
      else:
        - if:
            condition:
              - lambda: 'return id(battery).state >= ${another_threshold};'
            then:
              - deep_sleep.prevent: control_deep_sleep
            else:
              - if:
                  condition:
                    - sun.is_above_horizon:
                  then:
                    - delay: 2s
                    - api.disconnect: # expected disconnection maintaining state values ​​as deep_sleep does
                    - wifi.disable:
                    - delay: ${t_wifi_off_day}
                    - wifi.enable:
                  else:
                    - delay: 2s
                    - api.disconnect: # expected disconnection maintaining state values ​​as deep_sleep does
                    - wifi.disable:
                    - delay: ${t_wifi_off_night}
                    - wifi.enable:

```

donburch888, 2025-08-02:
Do you require the ESP32 to be running all the time ?  If you are just collecting data from various sensors on a regular schedule, you would probably be better off using deep_sleep as your starting point - eg wake for 2 minutes every half hour.  Deep_sleep does a full shutdown/power off, and then a power on/on_boot - including turning the wi-fi off and on ... which should give the effect you're after, if you're not doing anything on the ESP between sensor readings.  
Note also that increasing the sleep duration and/or reducing the awake time is the easiest way to reduce power consumption. 

I am particularly curious about your statement "I disable wifi with disable wifi action and then I can continue collecting data from sensors without losing filters and other time-based data, to then after a certain time, enable wifi and connect to home assistant. That works fine".  
Is ESPHome storing several readings for the same sensor taken over a period of time and then forwarding them all when wi-fi is re-established ?  Or is ESPHome only sending the current value when wi-fi is re established ?

You note that sometimes the state of the entities in Home Assistant retains the values; but sometimes Home Assistant shows the entities as unavailable.  I read that you should delete the device in Home Assistant, and re-add. 

Since you posted this FR I put together some [notes on deep_sleep](https://community.home-assistant.io/t/notes-on-esphome-deep-sleep/860987), which you have probably already read.

ssieb, 2025-08-02:
esphome is sending the current value, but if you're using something like an averaging filter, it would lose its state over deep sleep.  This way, you still send the filtered value right away when connecting again.

If it deep sleeps, HA will retain the value.  If it just turns off the wifi, then HA marks it as unavailable because of the unexpected disconnect.  This request doesn't really have anything to do with deep sleep other than trying to avoid it.