esphome/feature-requests #3015: YAML versioning or backup integration

bqzero, 2025-01-07:
**Describe the problem you have/What new integration you would like**
When working on a new device/YAML configuration, it is often necessary to do iterative testing to get it working finally.
This also comes with some try and error some times and leads to faulty YAML configurations.
Therefore to make better progress it would be nice to have an option to version/snapshot the current config and maybe even add some title and description to capture information. Something like sensor xyz working or lambda adjusted to cover ....
Basically similar to the backup/restore function of HA, but for single YAMLs.
Of course this can be achieved by manually copy the configuration somewhere and keeping track this way, but a simple solution like the HA backup/restore would be much more convenient. 

**Please describe your use case for this integration and alternatives you've tried:**
When creating new configuration sometimes step by step a solution needs to be tested. Recording the steps and being able to retrieve a former one helps in achieving a final working solution. Keeping the configuration steps manually is a time consuming task and not very well sorted, which leads to sometimes forgetting/skipping to keep a backup for time reasons that is regretted shortly after.

**Additional context**
Furthermore this can also be a handy feature when ESPHOME gets an update that for example requires an component to be adjusted as it has changed.

swoboda1337, 2025-01-13:
HA does back up the esphome yamls they are under data/esphome/ in the HA backup file. Why just use the HA backup.

oraum, 2025-01-22:
I'm simply keeping my YAML files inside a Git repository. This allows me to track revisions, tag versions or do some experiments on branches without polluting the history of other files. 
Additionally you can also create shared configs or templates in their own repositories and include them as [packages](https://esphome.io/components/packages.html)