esphome/feature-requests #3019: Multiple devices on 1 UART

Prozpect, 2025-01-09:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ssieb, 2025-01-09:
That's not an esphome limitation.  It's the external component.

closed 2025-01-09 by ssieb

Prozpect, 2025-01-09:
What is the external component in this case?

So what they say in the first link isn't true:

" Important

Only a single tmc2209 instance (device) per UART config is currently supported by ESPHome. Multiple drivers require multiple UART connections."

ssieb, 2025-01-09:
yes

Prozpect, 2025-01-10:
@slimcdk thanks for making time to look into this.
I've tried both options, but both don't seem to validate.

Let's wait for @ssieb for his advice.

Prozpect, 2025-01-18:
Or let's not wait for his advice....

So what do we have.
On the site of the external component is states: "Only a single tmc2209 instance (device) per UART config is currently supported by espHome. Multiple drivers require multiple UART connections."

But @ssieb states "That's not an esphome limitation. It's the external component."

I that's the case, is there is a flaw in the setup of the external component?
It's not a technical issue as I am communicating with both steppers on 1 UART using an Arduino sketch
and @ssieb indicates esphome is capable of doing this.

@slimcdk does the external component need to be changed to be able to do this?
Are there other options I can try in the script?

Thanks in advance

ssieb, 2025-01-18:
There are many examples of doing this.  modbus is the big example.
The second example above with the hub is probably the best option.  But it's not about a mutex.  The hub should be handling all the uart communication.

Prozpect, 2025-01-18:
thanks for replying, much appreciated.

I'll try both options again (hopefully tomorrow) and let you know about the outcome.
Maybe I can share my codes of the options and you can point out what things I can try/change to make it work?

Really hope I can get this to work using esphome, because although I have it working using an Arduino sketch, i like the updating through HA and I prefer the ephome code.

Prozpect, 2025-01-21:
Below I show the three options i've tried with the error messages following the code.
The first is the initial code I tried.
2 and 3 are the suggestions @slimcdk provided. 

@ssieb @slimcdk 
Please let me know if you have any suggestions to solve these error messages.

Test 1

```
external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209, stepper ]

uart:
  id: comms
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 500000

stepper:
  - platform: tmc2209
    id: driver1
    uart_id: comms
    address: 0
    step_pin: GPIO14
    dir_pin: GPIO12
    diag_pin: GPIO36
    max_speed: 2500 steps/s
  - platform: tmc2209
    id: driver2
    uart_id: comms
    address: 1
    step_pin: GPIO25
    dir_pin: GPIO26
    diag_pin: GPIO39
    max_speed: 2500 steps/s
```

Restult:

> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> Failed config
> 
> uart: [source /config/esphome/esphome-web-e7b4f4.yaml:34]
>   - id: comms
>     
>     The uart tx_pin is used both by tmc2209 and tmc2209, but can only be used by one. Please create a new uart bus for tmc2209.
>     tx_pin: 
>       number: 17
>       mode: 
>         output: True
>         input: False
>         open_drain: False
>         pullup: False
>         pulldown: False
>       inverted: False
>       ignore_pin_validation_error: False
>       ignore_strapping_warning: False
>       drive_strength: 20.0
>     
>     The uart rx_pin is used both by tmc2209 and tmc2209, but can only be used by one. Please create a new uart bus for tmc2209.
>     rx_pin: 
>       number: 16
>       mode: 

Test 2

```
external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209, stepper ]

uart:
  id: comms
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 500000

stepper:
  - platform: tmc2209
    uart_id: comms
    max_speed: 2500 steps/s
    drivers:
      - id: driver1
        address: 0
        step_pin: GPIO14
        dir_pin: GPIO12
        diag_pin: GPIO36
      - id: driver2
        address: 1
        step_pin: GPIO25
        dir_pin: GPIO26
        diag_pin: GPIO39
```

Result:

> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> Failed config
> 
> stepper.tmc2209: [source /config/esphome/esphome-web-e7b4f4.yaml:41]
>   platform: tmc2209
>   uart_id: comms
>   max_speed: 2500 steps/s
>   
>   [drivers] is an invalid option for [stepper.tmc2209]. Please check the indentation.
>   drivers: 
>     - id: driver1
>       address: 0
>       step_pin: GPIO14
>       dir_pin: GPIO12
>       diag_pin: GPIO36
>     - id: driver2
>       address: 1
>       step_pin: GPIO25

Test 3

```
external_components:
  - source: github://slimcdk/esphome-custom-components
    components: [ tmc2209, stepper ]

uart:
  id: comms
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 500000

tmc2209_hub:
  id: main_hub
  uart_id: comms

stepper:
  - platform: tmc2209
    id: driver1
    tmc2209_hub: main_hub
    address: 0
    step_pin: GPIO14
    dir_pin: GPIO12
    diag_pin: GPIO36
    max_speed: 2500 steps/s
  - platform: tmc2209
    id: driver2
    tmc2209_hub: main_hub
    address: 1
    step_pin: GPIO25
    dir_pin: GPIO26
    diag_pin: GPIO39
    max_speed: 2500 steps/s
```

Result

> INFO ESPHome 2024.12.4
> INFO Reading configuration /config/esphome/esphome-web-e7b4f4.yaml...
> INFO Unable to import component tmc2209_hub: No module named 'esphome.components.tmc2209_hub'
> Failed config
> 
> tmc2209_hub: [source /config/esphome/esphome-web-e7b4f4.yaml:41]
>   
>   Component not found: tmc2209_hub.
>   id: main_hub
>   uart_id: comms

Prozpect, 2025-01-21:
Ah, thanks for the clarification, I misundertood your earlier post. If it was that easy the solution was probably given earlier.
Awesome to hear you are working on this. Take your time and let me know when I can give it a try.