esphome/feature-requests #3020: Add support for delays on (gpio) switches

oliv3r, 2025-01-10:
**Describe the problem you have/What new integration you would like**
Certain devices (heat pumps specifically, but also gas/fire boilers etc) do not like to be turned off and immediately on again. Most of these devices traditionally have a 'delay' safety feature, that after a turn off event, it takes a while before you are 'allowed' to turn it on again.

While we can script these things in home-assistant, this neglects in part the safety part of this. While a good argument to be made is that such devices should protect themselves, when driving them over simple relays, often bypassing certain interfaces and thus safety elements.

**Please describe your use case for this integration and alternatives you've tried:**
As a concrete example, I (will used to have) a thermostat that controls my gas powered boiler. The thermostat switches a relay on or off, shortening the control lines to the boiler. The thermostat has a safety feature, that when you turn it on, the UI says its on, but it takes a while before the switch gets actually turned on.

A work 'around' we can hack this in esphome as such

```yaml
light:
  - platform: status_led
    name: "Status LED"
    id: board_status_led
    pin:
      number: D4
      inverted: true
    entity_category: diagnostic

output:
    platform: gpio
    id: "boilercontrol"
    pin: D2

switch:
    platform: template
    name: "Boiler fire"
    turn_on_action:
      - light.turn_on: "board_status_led"
      - delay: "30s"
      - output.turn_on: "boilercontrol"
    turn_off_action:
      - light.turn_off: "board_status_led"
      - delay: "30s"
      - output.turn_off: "boilercontrol"
```

> Note that the status light is not needed at all and just serves to help this demonstrator along to show that when Esphome turns on the relay, and the relay is actually turned on.

The request/proposal is thus also to do something in the lines of

```yaml
switch
  platform: gpio
  name: "Boiler fire"
  pin: D2
  delay: "30s"
```

**Additional context**
The above example makes it so that if multiple on-off events are sent, the relay isn't toggled rapidly, the last received state is the action that is being performed. This is also the reason why it's important to do this in esphome directly. If multiple actors (switches, HA, web-interface etc) all send a change request to esphome, the relay doesn't trigger events breaking the devices.

nagyrobi, 2025-01-10:
Your solution is not a workaround, it's the intended correct way to implement these use cases.

oliv3r, 2025-01-10:
@nagyrobi of course it will, but it should 'act' by `delaying` for x seconds after the frontend tells it to be activated.

Which is why I suggested:
```yaml
switch
  platform: gpio
  name: "Boiler fire"
  pin: D2
  delayed_on_off: "30s"
```

from a 'UX' point of view.