esphome/feature-requests #3032: OW-ENV 1-wire sensor with esphome?

tormodnt, 2025-01-21:
Just found this 1-wire Temperature / Humidity / Barometric Pressure & Light Sensor that would fit my need perfectly, but as I now have understood each sensors need to be coded in to esphome (yes I know, that's probably not the correct terminology, I can absolutely nothing about any of this, only know how easy DS18B20 is to use I esphome). 

Link:

 https://www.embeddeddatasystems.com/OW-ENV-THPL--Temperature-Humidity-Barometric-Pressure-Light-Sensor_p_175.html?srsltid=AfmBOopBQh_fGn-ac6Nzh-ioikLoqGh1_97QD2xG7e11Vahqqr5N1Jyf

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

B48D81EFCC, 2025-01-30:
Hi,

do you have any information about the specific sensor which is used in this device?
We know that the communication protocol is 1-wire, but this says nothing about the sensor(s).

B48D81EFCC, 2025-01-30:
Okay, I found the documentation for this device.
The documentation describes the data format of the different measurements: https://www.embeddeddatasystems.com/assets/images/supportFiles/manuals/EN-UserMan%20%20OW-ENV%20Sensor%20v14.pdf

tormodnt, 2025-01-31:
Unfortunately I have little to no knowledge or experience to help with anything.
But glad you found something that may help!

I'm totally confused, I've had several people telling me that to get this to work I need something added to ESPhome, at the same time I have several people saying that there will be no problem using it as it is just hook it up as any 1-wire..

> Okay, I found the documentation for this device. The documentation describes the data format of the different measurements: https://www.embeddeddatasystems.com/assets/images/supportFiles/manuals/EN-UserMan%20%20OW-ENV%20Sensor%20v14.pdf

ssieb, 2025-01-31:
Who told you that it would be no problem?  It has to have a component in esphome that knows how to communicate with it.

tormodnt, 2025-01-31:
I can't tell you who, because I've asked so many about this both in different ESPhome and Esp22 groups and places.

Your answer is like the other side.

It could be that esp32 people say that it's not needed, and the ESPhome group says that it is.

A option is to use normal ESP stuff and mqtt instead of ESPhome if that's a workaround. I know that getting stuff added to things like ESPhome when you self have no input or nothing to contribute with isn't neither likely or easy.

But had to try like I do here at least because I really think this os a unite that could be very useful for people.

> Who told you that it would be no problem?  It has to have a component in esphome that knows how to communicate with it.

tormodnt, 2025-01-31:
> Who told you that it would be no problem?  It has to have a component in esphome that knows how to communicate with it.

... I'm basically so useless that I don't even understand GitHub...

bytenik, 2025-09-27:
These will need to be added to the C++ code by reading the documentation of the sensor protocol and then implementing it.

I have installed 16 of these throughout my home. I'm planning on writing exactly that code. Now I just need to have any free time in my life to do the coding work.

Given that I've already installed them, I can promise it will happen. Will it happen before 2030? We will see!

tormodnt, 2025-09-27:
> These will need to be added to the C++ code by reading the documentation of the sensor protocol and then implementing it.
> 
> I have installed 16 of these throughout my home. I'm planning on writing exactly that code. Now I just need to have any free time in my life to do the coding work. I have a 2 year old and a 4 year old so I'm not exactly flush with free time.
> 
> Given that I've already installed them, I can promise it will happen. Will it happen before 2030? We will see!

Amazing! There is hope!!

Do you find them as good as they look to be?
With 16 of then installed it looks like you are doing the same as I want that made me add the request. I want one in each room.

Fingers crossed for you to have some time, I understand that you can't promise anything, but now I at least have some hope!!

Best wishes from Norway to you and your family!