esphome/feature-requests #3040: Use ESPHome device as audio RTSP streamer

edurenye, 2025-01-28:
**Describe the problem you have/What new integration you would like**

Currently, the ESPHome devices like M5Stack ATOM Echo can be used as media players outputting digital streams, what I want is for them to be used as microphones for other systems producing an AAC or MP3 RTSP audio stream that could be played as output by another ESPHome device or to be processed by [StreamAssist](https://github.com/AlexxIT/StreamAssist) in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

I can't understand anything that the M5Stack ATOM Echo answers because its speaker is useless, so I want to use another media player to output the answer of my LLM. The idea is to use the ATOM Echo as a microphone, send the RTSP stream to StreamAssist have the LLM process the answer and play the answer in a better media player. But also I read that some people want to directly play the sound from a ATOM Echo in another media player in a different room, like a communication device (see: https://community.home-assistant.io/t/m5-atom-echo-audio-stream/746536/2)

**Additional context**

There is a repo for outputting RTSP video streams for the ESP32 Cam hardware (https://github.com/rzeldent/esp32cam-rtsp), but not only for audio streams, and it does not support other devices like the ATOM Echo.

zerg960, 2025-03-25:
This guy figured out how to get the LLM reply to play on a different speaker: https://youtu.be/o3yZWD_sFIE

edurenye, 2025-03-25:
Thanks, I managed to set it up last week, I forgot I had this issue open, I guess I can close it.

But it's quite complex to set up, would be great to be able to do it from the UI.

closed 2025-03-25 by edurenye

edurenye, 2025-03-25:
Not sure if in the video they use the same solution that I used, I used this solution: https://community.home-assistant.io/t/esphome-voice-assistant-speech-output-to-home-assistant-media-player/588337/37