esphome/feature-requests #3042: Logging level in INTERVAL COMPONENT

LibichJ, 2025-01-29:
**Describe the problem you have/What new integration you would like**
It would be nice to have possibility to change logging level in INTERVAL COMPONENT. 

**Please describe your use case for this integration and alternatives you've tried:**
I'm using interval component for user timers. It is easy to use, hovewer in case of interval time 1sec it fill up the logger very fast and making very problematic to see other messages in DEBUG level.

**Additional context**
Would be nice to have parameter "log_level" or similar where I can choose from usable logging levels, or deactivate logging for intervals component completely.

ssieb, 2025-01-29:
What logs?  The interval component has no logs of itself.

randybb, 2025-01-29:
and you can already change log level of any component https://esphome.io/components/logger#manual-tag-specific-log-levels

ssieb, 2025-01-29:
There's no logging from the interval component itself, so that won't help here.  Need more information.

LibichJ, 2025-01-29:
Interval component has the logging message, see attached picture.
[image]

In case of using 1sec interval it's sending 3 message lines into the debug logger and this is making difficult to find any other debug messages in logger. I would like to turn messages from interval component off, or move these messages into different logging level. 

Is it possible?

Thank you.

ssieb, 2025-01-29:
That's the `number`, not the `interval`.  I guess you're decrementing that number from an interval.  You can only filter by the tag, which is "number" in this case.  But that would remove the logging for all numbers, not just that one.
It looks like you're using the number as a timer, which is not a great method.  I suggest you come ask for help on the esphome discord server.

closed 2025-01-29 by ssieb