esphome/feature-requests #3046: Synchronise Sensor Updates

unaiur, 2025-02-09:
**Describe the problem you have/What new integration you would like**

I have a dozen of sensors that all update every 60 seconds and a few more that updates every 2 & 5 minutes. It works fine just after reboot but after some hours, each sensor updates at a different second and that makes impossible to sleep between updates. 

Those updates should be also synchronized, to avoid awakening the device more than once per minute.

**Please describe your use case for this integration and alternatives you've tried:**

In my cause, failure to enter sleep mode causes a temperature increase in the device (HW is deployed in a closed case inside a wall with super-limited ventilation). Which I think leads EspHome to hang up. Same hardware running Tasmota does not experiment this temperature raise over time and is stable. Sadly, Tasmota does not have all the features of EspHome.

Of course, fixing this issue will extend life for battery powered devices.

**Additional context**

This problem could be solved allowing to specify the update field as a limited crontab. For example:
 * "17 * *" will update every minute and every hour at second 17.
 * "17 */5 *" will update every five minutes at second 17.
 * "17 25 3" will update once per day at 3:25:17.

This solution would also covers the situation with multiple sensors.

An alternative could be group together updates when the difference is less than certain percentage (coalesce_pct). For example, if I schedule an update timer of 60 seconds, coalesce_pct is 5% and there is a running timer with 57-63 seconds remaining, we adjust the timer duration to match it.

ssieb, 2025-02-10:
Just disable the update interval, then force an update when it's connected.  The update interval does not wake the device.  It only applies while the device is running.  It sounds like you might have other issues.  Come ask on the discord server if you need help with this.

closed 2025-02-10 by ssieb

UnaiUribarri-TomTom, 2025-02-10:
I am not taking about deep sleep, just modem sleep. Modem sleep is entered after 10 seconds idle. 

But my sensors are sending updates constantly (for example, right now it is sending sensor updates at second 8, 11, 18, 29, 33, 35, 44 and 56) even when all sensors have same update period: 60 seconds.

Therefore, modem only sleeps a few seconds (2 or 3) each minute. It is easy to see connecting an ampere meter. 

Instead, I expect it to send all updates together at the same second and let the modem sleep for the rest of the minute. 

I understand the solution you are proposing, but the issue is deeper; 

Current scheduler code actively spread interval timers to avoid triggering them at once. Why is doing it and why we cannot change it to coalesce them instead?

ssieb, 2025-02-10:
The point is to spread out the activity and network load.  If you want to synchronize them, then disable the update interval and use an interval to trigger them when you want.
The only sleep that esphome supports is deep sleep, so I had no idea you meant something else.