esphome/feature-requests #3048: Add lv_cpicker in LVGL component

bennydiamond, 2025-02-11:
**Describe the problem you have/What new integration you would like**

LVGL's `lv_cpicker` component is missing in LVGL.
https://docs.lvgl.io/7.11/widgets/cpicker.html

**Please describe your use case for this integration and alternatives you've tried:**

Would be very useful for HMI devices targeted at controlling RGB lighting devices.
If not present all I could think to do is add a couple of buttons with preset colors, which is a bit limiting.

**Additional context**

`lv_colorwheel` is also a thing although I thing `lv_cpicker` would be more useful.
https://docs.lvgl.io/8.3/widgets/extra/colorwheel.html

bennydiamond, 2025-02-11:
That makes sense as to why it wasn't already supported then!

Are there alternative widgets offered in ESPHome that would offer similar functionnality?
As I said, I would like to control a RGG bulb from an HMI and I would really like a widget that lets me pick any color rather than resorting to having a couple preset colors exposed as buttons on my  HMI.

Thank you.

nagyrobi, 2025-02-11:
@clydebarrow

clydebarrow, 2025-02-11:
I have a proof of concept colour picker working, but it needs some more attention.

bennydiamond, 2025-02-12:
> I have a proof of concept colour picker working, but it needs some more attention.

My curiosity is piqued.
Do you have an example I could maybe give a go at it?

clydebarrow, 2025-02-12:
```
external_components:
  - source: github://clydebarrow/esphome@lvgl-color-picker
    refresh: 30s
    components: [lvgl]

lvgl:
  widgets:
    - color_picker:
        id: picker
        width: 400px
```

bennydiamond, 2025-02-16:
> ```
> external_components:
>   - source: github://clydebarrow/esphome@lvgl-color-picker
>     refresh: 30s
>     components: [lvgl]
> 
> lvgl:
>   widgets:
>     - color_picker:
>         id: picker
>         width: 400px
> ```

Thanks alot.
I tried your branch but I get the following error

```
Compiling .pioenvs/sdl-hmi-garage/src/esphome/components/lvgl/lvgl_esphome.o
src/esphome/components/lvgl/lvgl_esphome.cpp:2:7: error: ‘esphome’ has not been declared
    2 | using esphome::esp_log_printf_;
      |       ^~~~~~~
Compiling .pioenvs/sdl-hmi-garage/src/main.o
In file included from src/esphome/components/lvgl/lvgl_esphome.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.cpp: In function ‘void* lv_custom_mem_alloc(size_t)’:
src/esphome/core/log.h:126:3: error: ‘esp_log_printf_’ was not declared in this scope; did you mean ‘esphome::esp_log_printf_’?
  126 |   esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |   ^~~~~~~~~~~~~~~
src/esphome/core/log.h:149:28: note: in expansion of macro ‘esph_log_e’
  149 | #define ESP_LOGE(tag, ...) esph_log_e(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/lvgl/lvgl_esphome.cpp:505:5: note: in expansion of macro ‘ESP_LOGE’
  505 |     ESP_LOGE(esphome::lvgl::TAG, "Failed to allocate %zu bytes", size);
      |     ^~~~~~~~
src/esphome/core/log.h:56:6: note: ‘esphome::esp_log_printf_’ declared here
   56 | void esp_log_printf_(int level, const char *tag, int line, const char *format, ...)  // NOLINT
      |      ^~~~~~~~~~~~~~~
*** [.pioenvs/sdl-hmi-garage/src/esphome/components/lvgl/lvgl_esphome.o] Error 1
```

I tried merging lvgl_color_picker with upstream's dev branch. Same result.
Could it be related to the fact I'm using ESP-IDF instead of Arduino Framework?

clydebarrow, 2025-02-16:
I'm away from my computer for a few days, but I did test that example and it compiled and ran. I would guess you are not using the current dev version core code, in spite of your comment.

closed 2025-02-18 by bennydiamond