esphome/feature-requests #3049: other domain than .local

AchimHome, 2025-02-15:
ESPHome currently relies on mDNS for device discovery, using .local hostnames. However, some networks do not support mDNS, making it impossible to resolve ESPHome nodes without manually configuring static IP addresses. I would like ESPHome to support specifying a custom domain name that can be resolved via traditional DNS.

In my network, mDNS is not available, but I have a DNS server that can resolve FQDNs (e.g., node-terrasse.mydomain.com). Currently, I have to manually enter static IP addresses within "wifi -> manual_ip", which is not ideal for dynamic or large-scale deployments. An option to configure a custom FQDN instead of .local would improve usability in environments where mDNS is not available.

Additional context

    Ideally, this could be implemented as a configuration option in ESPHome global configuration like status_use_ping, allowing users to specify a custom hostname for OTA updates and dashboard communication.
    Some users rely on DHCP reservations instead of static IPs, so allowing a hostname that DNS can resolve dynamically would be beneficial.
    This would improve ESPHome’s compatibility with corporate and enterprise networks that do not use mDNS.

ssieb, 2025-02-15:
There is a domain option in the device network config.

AchimHome, 2025-02-15:
@ssieb I suppose you refer to:

wifi:
  domain: .mydomain.com
  ssid: !secret iot_ssid
  password: !secret iot_password

Yes, that works like static ip addresses in most cases.
My feature request was to supersede the ".local" configuration globally instead:

There is no configuration in ESPhome builder to use another domain than ".local" globally for all devices.
At least i didn't found this feature.

ssieb, 2025-02-15:
I guess it's a valid feature request.  You'll have to manually add each device to HA as well.