esphome/feature-requests #3053: Make "ESPHome Device Builder" container web UI (usually on http://esphome:6052/ ) natively support TLS

ytugarev, 2025-02-18:
**Describe the problem you have/What new integration you would like**
I run fully containerize environment and widely use TLS. Almost all web UIs are under TLS (Home Assistant, Zigbee2MQTT frontend, etc).
It seems that "ESPHome Device Builder" container web UI does not support TLS natively.
**Please describe your use case for this integration and alternatives you've tried:**
I tried to search over an hour to find any other solutions but the only one is here to use some TLS proxy (Nginx and such) in front of ESPHome UI.
What would TLS support give:
1) Following "Assume breach" security concept and make installation more secure.
2) Making basic auth (which is built-in) actually valuable. Without TLS layer it makes no sense - you can sniff credentials because they are sent in open text.
3) ESPHome Device Builder container web UI could be easily integrated with Home Assistant protected portal via Dashboards/WebPage. If HA is under HTTPS it is not possible to bring HTTP webpages as a dashboards.

**Additional context**
I bet there were other feature requests about this but unfortunately I was not able to find it. Please point me to it if possible.

ytugarev, 2025-02-18:
No, it is not. Every architectural decision has its purpose. I think Occam's razor principle is the most important one.

I run slim ARM SBC + Docker and can easily manage my certificate distribution between containers - I simply use wildcard one from Let'sEncrypt and can drop it to corresponding container volumes with a script. I do not need another point of failure. I can justify Nginx proxy when you publish stuff to the Internet but locally - no, thanks.

And BTW  - what is the point of this basic built in auth if it is in plain text? Just to make people laugh? 😆

ssieb, 2025-02-18:
The decision was made that esphome would not handle https on its own.  If you want it, you need to use a proxy.