esphome/feature-requests #3057: LVGL msgboxes needs som TLC.

phellarv, 2025-02-21:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

**LVGL**:

There is a couple of things I like to address runtime:

- Title 
  - Font
  - Color (Font and Background)
  - Text
- Body
  - Font
  - Color (Font and Background)
  - Text
- Size of object
- Placement of object
- Theme
- Style definition

clydebarrow, 2025-03-02:
Yes, there is some missing functionality there.

clydebarrow, 2025-04-07:
> If it's possible to add active widgets to message box?

What's an active widget?

mateusz-jaworski, 2025-04-07:
> > If it's possible to add active widgets to message box?
> 
> What's an active widget?

Fo example slider or gauge inside message box

clydebarrow, 2025-04-07:
Sure, just add in the usual way under the `widgets` key.