esphome/feature-requests #3059: Implement Swipe/Gestures as in upstream LVGL

phellarv, 2025-02-24:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

The current implementation of swipe in ESPHome results in firing off on_press and on_release if you start or end a swipe on a button.
I would guess that the same implies for other widgets with on_press and on_release.
This is solved in upstream by using prioritization of GESTURE over CLICKED events.
Specifically by using a call lv_indev_wait_relelase(lv_indev_act()) in the gesture.
This is described in https://github.com/lvgl/lvgl/issues/3211
And I've made an issue about this too: https://github.com/esphome/issues/issues/6777

Kind regards

clydebarrow, 2025-02-28:
You can add a lambda to call `lv_indev_wait_release()` in the gesture trigger if that is the desired behaviour. You might also want to set `press_lock` to false for any widgets that might have a gesture started on them.
```
  pages:
  - id: page_1
    on_swipe_left:
      - lambda: 'lv_indev_wait_release(lv_indev_get_act());'
      - logger.log: swiped
    widgets:
    - button:
        checkable: true
        press_lock: false
        widgets:
        - label:
            text: Button
```

closed 2025-02-28 by clydebarrow

phellarv, 2025-03-02:
Excellent. Thank you.