esphome/feature-requests #3060: Change underlying for on_press

phellarv, 2025-02-25:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

When you have defined both on_press and on_long_press, both events are fired on a long press.
When I checked upstream there is mechanisms for preventing this.

According to documentation the sequence of events is:

1.     LV_EVENT_PRESSED
2.     LV_EVENT_LONG_PRESSED (if pressed long enough)
3.     LV_EVENT_LONG_PRESSED_REPEAT (if pressed long enough)
4.     LV_EVENT_RELEASED
5.     LV_EVENT_SHORT_CLICKED (if not a long press)
6.     LV_EVENT_CLICKED (both short and long)

So the better choice for the on_press event would be LV_EVENT_SHORT_CLICKED to prevent on_press event to happen when on_long_press is the desired event.

clydebarrow, 2025-03-02:
So if `on_short_clicked` works better for you than `on_pressed`, use that instead. It's not clear what you are suggesting be changed.

phellarv, 2025-03-02:
Oh - I did not know that was an option. I thought it had to be changed in the underlying C/C++ calls. Thanks for the info.

closed 2025-03-02 by ssieb