esphome/feature-requests #3067: Add Support for DFRobot SEN0460 Air Quality Sensor (PM1/PM2.5/PM10) in ESPHome

lentex43, 2025-02-28:
Describe the problem you have / What new integration you would like
I would like to request support for the DFRobot SEN0460 air quality sensor (PM1 / PM2.5 / PM10) using the I2C communication protocol in Home Assistant > ESPHome.

Please describe your use case for this integration and alternatives you've tried:
I attempted to add this sensor to ESPHome but was unsuccessful. After reviewing the documentation and community forums, I found no information about native support for this model. Currently, there is no official solution that allows easy integration of this sensor within ESPHome.

Adding support for this device would enable users to monitor air quality in their smart homes and use this data in Home Assistant automations.

I am connecting the sensor to an ESP32-WROOM-32D, and I would really appreciate its integration.

Additional context:
Official sensor documentation: [DFRobot SEN0460 – Gravity PM2.5 Air Quality Sensor](https://wiki.dfrobot.com/Gravity_PM2.5_Air_Quality_Sensor_SKU_SEN0460)

Thank you for considering this request. I hope this integration will be possible.

Best regards!

BliXem1, 2025-03-07:
BTW: did you check and tested this? https://github.com/makertut/esphome-air-sensors

lentex43, 2025-03-07:
Thank you for your message.
I tried again with the website you sent, but I still can't see the values from the sensor.
This is also a bit of an old guide. I can't manage to get it working.

BliXem1, 2025-03-07:
Yeah I don't too. For now I bought a other device: https://esphome.io/components/sensor/hm3301.html.

lentex43, 2025-03-12:
Thank you. I was also considering buying it.
However, I will keep trying to make the one from the topic work.
I hope it will be supported someday.

DanBar1971, 2025-04-04:
Please +1

DanBar1971, 2025-04-08:
I make one and it works: https://github.com/DanBar1971/esphome/tree/dev/esphome/components/dfrobot_air_quality_sensor

BliXem1, 2025-04-08:
@DanBar1971 do you have a yaml file what we need to add in the code to load it up? :)

DanBar1971, 2025-04-08:
Just copy all files to:

esphome
├── components
         ├── dfrobot_air_quality_sensor

Add : to u code

external_components:
  - source:
      type: local
      path: components

sensor:
  # #######################
  - platform: dfrobot_air_quality_sensor
    id: air_quality_sensor_level_sensor
    update_interval: 60s
    address: 0x19
    pm_1_0:
      name: "PM1.0"
      id: pm1_0_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
      state_topic: ${name}/sensors/pm1_0
      on_value:
        - lambda: |-
              if(id(pm1_0_sensor).state > id(pm1_0_max)) { id(pm1_0_max) = id(pm1_0_sensor).state; id(pm1_0_max_sensor).state = id(pm1_0_max);}
              if(id(pm1_0_sensor).state < id(pm1_0_min)) { id(pm1_0_min) = id(pm1_0_sensor).state; id(pm1_0_min_sensor).state = id(pm1_0_min);}
    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
      state_topic: ${name}/sensors/pm2_5
      on_value:
        - lambda: |-
              if(id(pm2_5_sensor).state > id(pm2_5_max)) { id(pm2_5_max) = id(pm2_5_sensor).state; id(pm2_5_max_sensor).state = id(pm2_5_max);}
              if(id(pm2_5_sensor).state < id(pm2_5_min)) { id(pm2_5_min) = id(pm2_5_sensor).state; id(pm2_5_min_sensor).state = id(pm2_5_min);}      
    pm_10:
      name: "PM10"
      id: pm10_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
      state_topic: ${name}/sensors/pm10
      on_value:
        - lambda: |-
              if(id(pm10_sensor).state > id(pm10_max)) { id(pm10_max) = id(pm10_sensor).state; id(pm10_max_sensor).state = id(pm10_max);}
              if(id(pm10_sensor).state < id(pm10_min)) { id(pm10_min) = id(pm10_sensor).state; id(pm10_min_sensor).state = id(pm10_min);}
  
    ######################
    aqi:
      name: "AQI"
      id: aqi_sensor
      calculation_type: "CAQI" # AQI: USA air quality standard CAQI: Europe air quality standard [https://airly.org/en/air-quality-index-caqi-and-aqi-methods-of-calculation/ ] https://airly.org/map/en/#38.7205689609,-9.1496113301
      state_topic: ${name}/sensors/aqi
      icon: mdi:air-filter
      on_value:
        - lambda: |-
              if(id(aqi_sensor).state > id(aqi_max)) { id(aqi_max) = id(aqi_sensor).state; id(aqi_max_sensor).state = id(aqi_max); }
              if(id(aqi_sensor).state < id(aqi_min)) { id(aqi_min) = id(aqi_sensor).state; id(aqi_min_sensor).state = id(aqi_min); }
      on_value_range:
        - below: 0
          then:
            - lambda: id(aqi_msg_text).publish_state("Desligado!");
        - above: 0
          below: 50
          then:
            - lambda: id(aqi_msg_text).publish_state("Boa.");
        - above: 51
          below: 100
          then:
            - lambda: id(message_text).publish_state("Moderado.");
        - above: 101
          below: 150
          then:
            - lambda: id(message_text).publish_state("Insalubre para membros de grupos sensíveis.");
        - above: 151
          below: 200
          then:
            - lambda: id(message_text).publish_state("Insalubre.");
        - above: 201
          below: 300
          then:
            - lambda: id(message_text).publish_state("Muito insalubre.");
        - above: 301
          then:
            - lambda: id(message_text).publish_state("Perigosa.");
  
  #######################
  - platform: template
    name: "PM1.0 Max"
    id: pm1_0_max_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm1_0_max
    update_interval: 60s
  #######################
  - platform: template
    name: "PM1.0 Min"
    id: pm1_0_min_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm1_0_min
    update_interval: 60s
  #######################
  - platform: template
    name: "PM10 Max"
    id: pm10_max_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm10_max
    update_interval: 60s
  #######################
  - platform: template
    name: "PM10 Min"
    id: pm10_min_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm10_min
    update_interval: 60s
  #######################
  - platform: template
    name: "PM2.5 Max"
    id: pm2_5_max_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm2_5_max
    update_interval: 60s
  #######################
  - platform: template
    name: "PM2.5 Min"
    id: pm2_5_min_sensor
    unit_of_measurement: "µg/m³"
    icon: mdi:chemical-weapon
    state_topic: ${name}/sensors/pm2_50_min
    update_interval: 60s
  #######################
  - platform: template
    name: "AQI Max"
    id: aqi_max_sensor
    unit_of_measurement: "index"
    icon: mdi:air-filter
    state_topic: ${name}/sensors/aqi_max
    update_interval: 60s
  #######################
  - platform: template
    name: "AQI Min"
    id: aqi_min_sensor
    unit_of_measurement: "index"
    icon: mdi:air-filter
    state_topic: ${name}/sensors/aqi_min
    update_interval: 60s

DanBar1971, 2025-04-08:
Or just this line for testing:

```
external_components:
  - source:
      type: local
      path: components

sensor:
  - platform: dfrobot_air_quality_sensor
    id: air_quality_sensor_level_sensor
    update_interval: 60s
    address: 0x19
    pm_1_0:
      name: "PM1.0"
      id: pm1_0_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon

    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
     
    pm_10:
      name: "PM10"
      id: pm10_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon

    ######################
    aqi:
      name: "AQI"
      id: aqi_sensor
      calculation_type: "CAQI" # AQI: USA air quality standard CAQI: Europe air quality standard [https://airly.org/en/air-quality-index-caqi-and-aqi-methods-of-calculation/ ] https://airly.org/map/en/#38.7205689609,-9.1496113301
      state_topic: ${name}/sensors/aqi
      icon: mdi:air-filter
```

DanBar1971, 2025-04-08:
```
esphome:
  name: test-dfrobot-air-quality-sensor
  friendly_name: Test_dfrobot_air_quality_sensor

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "< USE YOURS >"

ota:
  - platform: esphome
    password: "< USE YOURS >"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Test-Dfrobot-Air-Quality-Sensor"
    password: "< USE YOURS >"

captive_portal:

external_components:
  - source:
      type: local
      path: components

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

sensor:
  - platform: dfrobot_air_quality_sensor
    id: air_quality_sensor_level_sensor
    update_interval: 60s
    address: 0x19

	pm_1_0:
      name: "PM1.0"
      id: pm1_0_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon

    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
    
	pm_10:
      name: "PM10"
      id: pm10_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon

    aqi:
      name: "AQI"
      id: aqi_sensor
      calculation_type: "CAQI"
      icon: mdi:air-filter  
    
```

BliXem1, 2025-04-08:
Tested and seems to be working!

BliXem1, 2025-04-09:
Maybe @nagyrobi can add it on the esphome website? :)

DanBar1971, 2025-04-09:
Update: Add number of particles in 0.1L of air: 
pm_0_3um:
pm_0_5um:
pm_5_0um:
pm_10_5um:

https://github.com/DanBar1971/esphome/tree/dev/esphome/components/dfrobot_air_quality_sensor

DanBar1971, 2025-04-09:
`esphome:
  name: test-dfrobot-air-quality-sensor
  friendly_name: Test_dfrobot_air_quality_sensor

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "< USE YOURS >"

ota:
  - platform: esphome
    password: "< USE YOURS >"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Test-Dfrobot-Air-Quality-Sensor"
    password: "< USE YOURS >"

captive_portal:

external_components:
  - source:
      type: local
      path: components

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

sensor:
  - platform: dfrobot_air_quality_sensor
    id: air_quality_sensor_level_sensor
    update_interval: 60s
    address: 0x19
	pm_1_0:
      name: "PM1.0"
      id: pm1_0_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
    pm_2_5:
      name: "PM2.5"
      id: pm2_5_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
	pm_10:
      name: "PM10"
      id: pm10_sensor
      unit_of_measurement: "µg/m³"
      icon: mdi:chemical-weapon
    aqi:
      name: "AQI"
      id: aqi_sensor
      calculation_type: "CAQI"
      icon: mdi:air-filter  
      
    #  Get the number of particles in 0.1L of air
    pm_0_3um:
      name: "PM0.3"
      id: pm03_sensor
      unit_of_measurement: "um"
      icon: mdi:chemical-weapon

    pm_0_5um:
      name: "PM0.5"
      id: pm05_sensor
      unit_of_measurement: "um"
      icon: mdi:chemical-weapon

    pm_5_0um:
      name: "PM5.0"
      id: pm5_0_sensor
      unit_of_measurement: "um"
      icon: mdi:chemical-weapon

    pm_10_5um:
      name: "PM10.5"
      id: pm10_5_sensor
      unit_of_measurement: "um"
      icon: mdi:chemical-weapon  `

lentex43, 2025-05-02:
Thank you for the solution.
It works.