esphome/feature-requests #3069: Add priority configuration to i2c bus to allow users to power down during sleep

ConnorKirkpatrick, 2025-03-01:
**Describe the problem you have/What new integration you would like**

The i2c bus is currently enabled during boot with a priority of 1000 (highest). Some users wish to be able to power on and off an i2c device during sleep to save power, however this priority (1000) means that any on-boot action to power on the device happens after the i2c startup, leading to errors and failed boots.

The plan is to edit the i2c_bus_arduino/idf file to add a field to allow the user to configure this priority. By default this will initialize as 1000 to maintain default behavior. This will alter the get_setup_priority function to return this new value.

**Please describe your use case for this integration and alternatives you've tried:**

Alternatives to the proposed method can be found in this link: https://community.home-assistant.io/t/delayed-initialization-i2c-bus/659101
They consist of either creating a lambda that can use a priority of 1000 to trigger the pins to turn on before the i2c setup, or by creating a duplicate i2c component that uses a hard-coded lower value.
Both of these work but are inherently workarounds where a cleaner solution would make this behavior more accessible to the general user.

**Additional context**

I intend to pick up this request up myself.

ssieb, 2025-03-01:
What errors and why is the boot failing?  The bus isn't really anything.  As long as the device is powered on before it is used, it should be fine.

ConnorKirkpatrick, 2025-03-01:
For certain low power projects (solar, etc) I've dabbled with the idea of attaching the i2c device's (bmp-280) power pin to one of the esp32 GPIO, such that I can turn on and off the power to the sensor for sleeping periods to conserver power. This is the same practice I use for other sensor components outside of i2c. 

When using this method, as the i2c initialisation runs before the on-boot tasks, the esp32 complains that "SCL is held low", this is due to the i2c device attached to the bus being powered off. (This error goes away if I hard wire power to the BMP, vs via the GPIO with a switch_on even during on_boot)

Other members of the community (see link in first post) have gotten around this via either hard-coding a new i2c to allow the on-boot task (switch GPIO to high) to take place before the i2c is initialised, or using a lambda outside of their switch component. 
The idea of this ticket is to create a more formal way of performing this type of behavior vs to two workarounds that are not the most intuitive.

ssieb, 2025-03-01:
Does it matter if you get the i2c bus error?  Doesn't it still work anyway?

ConnorKirkpatrick, 2025-03-01:
No the i2c bus causes a "failed start" trigger, the rest of the esphome script is never run. When testing this it meant that the device never proceeded to wireless connection that meant reverting the change required flashing the device via USB again.

ssieb, 2025-03-01:
There's nothing in the i2c component that would stop the boot.  There's no "failed start" trigger, whatever that is supposed to mean.  The only problem could be if you have the scan enabled, which you definitely shouldn't in this case.  If it is, it could timeout and trigger the WDT.  But then eventually, it would be in safe mode and you could still OTA update it.

ConnorKirkpatrick, 2025-03-01:
Noted, I probably wasn't patient enough for the safe mode to trigger. Scan is not enabled on the program end, however from looking now it looks like I should be setting that to false as it is default true. 

I'll revert back to the old firmware in the morning and copy out the logs to put here. From your comments it feels like the scan setting is most likely the culprit of the i2c issue vs power.

ssieb, 2025-03-02:
What happens if you add `setup_priority: -100` to the bme config?

ConnorKirkpatrick, 2025-03-02:
Updated with the following change:
```
- platform: bme280_i2c
    setup_priority: -100
    address: 0x76
    update_interval: 10s
    temperature:
      name: "Plant Temperature"
      id: Outside_Temperature
      oversampling: 4x
    humidity:
      name: "Plant Humidity"
      id: Outside_Humidity
    pressure:
      name: "Plant Pressure"
      id: Outside_Pressure
```

Results in the same issues with i2c:

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM1 with baud rate 115200
[16:21:37][I][logger:171]: Log initialized
[16:21:37][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[16:21:37][D][esp32.preferences:114]: Saving 1 preferences to flash...
[16:21:37][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:21:37][I][app:029]: Running through setup()...
[16:21:37][D][switch:012]: 'energise' Turning ON.
[16:21:37][D][switch:055]: 'energise': Sending state ON
[16:21:37][I][i2c.arduino:218]: Performing I2C bus recovery
[16:21:37][E][i2c.arduino:234]: Recovery failed: SCL is held LOW on the I2C bus
[16:21:37][C][switch.gpio:011]: Setting up GPIO Switch 'energise'...
[16:21:37][D][switch:016]: 'energise' Turning OFF.
[16:21:37][D][switch:055]: 'energise': Sending state OFF
[16:21:37][D][switch:016]: 'energise' Turning OFF.
[16:21:37][C][adc.esp32:025]: Setting up ADC 'Soil Moisture'...
[16:21:37][C][wifi:048]: Setting up WiFi...
[16:21:37][C][wifi:061]: Starting WiFi...
[16:21:37][C][wifi:062]:   Local MAC: 34:85:18:AB:FF:24
[16:21:37][D][wifi:482]: Starting scan...
[16:21:37][D][sensor:094]: 'Soil Moisture': Sending state 100.00000 % with 1 decimals of accuracy
[16:21:37][W][component:157]: Component wifi set Warning flag: scanning for networks
[16:21:43][D][wifi:497]: Found networks:
<redacted wifi connection>
[16:21:59][C][api:026]: Setting up Home Assistant API server...
[16:21:59][C][bme280.sensor:091]: Setting up BME280...
[16:22:00][E][component:119]: Component bme280_base was marked as failed.
[16:22:00][E][component:164]: Component bme280_base set Error flag: unspecified
[16:22:00][I][app:062]: setup() finished successfully!
[16:22:00][W][component:170]: Component wifi cleared Warning flag
<redacted wifi connection>
[16:22:01][C][logger:177]: Logger:
[16:22:01][C][logger:178]:   Max Level: DEBUG
[16:22:01][C][logger:179]:   Initial Level: DEBUG
[16:22:01][C][logger:181]:   Log Baud Rate: 115200
[16:22:01][C][logger:182]:   Hardware UART: USB_CDC
[16:22:01][C][i2c.arduino:071]: I2C Bus:
[16:22:01][C][i2c.arduino:072]:   SDA Pin: GPIO5
[16:22:01][C][i2c.arduino:073]:   SCL Pin: GPIO6
[16:22:01][C][i2c.arduino:074]:   Frequency: 50000 Hz
[16:22:01][C][i2c.arduino:089]:   Recovery: failed, SCL is held low on the bus
[16:22:01][C][switch.gpio:068]: GPIO Switch 'energise'
[16:22:01][C][switch.gpio:091]:   Restore Mode: always OFF
[16:22:01][C][switch.gpio:031]:   Pin: GPIO4
[16:22:01][C][adc.esp32:058]: ADC Sensor 'Soil Moisture'
[16:22:01][C][adc.esp32:058]:   Device Class: 'voltage'
[16:22:01][C][adc.esp32:058]:   State Class: 'measurement'
[16:22:01][C][adc.esp32:058]:   Unit of Measurement: '%'
[16:22:01][C][adc.esp32:058]:   Accuracy Decimals: 1
[... 20 lines cut ...]
[16:22:01][C][bme280.sensor:195]:   Update Interval: 10.0s
[16:22:01][C][bme280.sensor:197]:   Temperature 'Plant Temperature'
[16:22:01][C][bme280.sensor:197]:     Device Class: 'temperature'
[16:22:01][C][bme280.sensor:197]:     State Class: 'measurement'
[16:22:01][C][bme280.sensor:197]:     Unit of Measurement: '°C'
[16:22:01][C][bme280.sensor:197]:     Accuracy Decimals: 1
[16:22:01][C][bme280.sensor:198]:     Oversampling: 4x
[16:22:01][C][bme280.sensor:199]:   Pressure 'Plant Pressure'
[16:22:01][C][bme280.sensor:199]:     Device Class: 'pressure'
[16:22:01][C][bme280.sensor:199]:     State Class: 'measurement'
[16:22:01][C][bme280.sensor:199]:     Unit of Measurement: 'hPa'
[16:22:01][C][bme280.sensor:199]:     Accuracy Decimals: 1
[16:22:01][C][bme280.sensor:200]:     Oversampling: 16x
[16:22:01][C][bme280.sensor:201]:   Humidity 'Plant Humidity'
[16:22:01][C][bme280.sensor:201]:     Device Class: 'humidity'
[16:22:01][C][bme280.sensor:201]:     State Class: 'measurement'
[16:22:01][C][bme280.sensor:201]:     Unit of Measurement: '%'
[16:22:01][C][bme280.sens[C][bme280.sensor:202]:     Oversampling: 16x
[16:22:04][E][component:082]:   Component bme280_base is marked FAILED033[0;36m[D][sensor:094]: 'Soil Moisture': Sending state 100.00000 % with 1 decimals of accuracy
[16:22:34][D][sensor:094]: 'Soil Moisture': Sending state 100.00000 % with 1 decimals of accuracy
[16:22:37][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[16:22:37][D][esp32.preferences:114]: Saving 1 preferences to flash...
[16:22:37][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:23:04][D][sensor:094]: 'Soil Moisture': Sending state 100.00000 % with 1 decimals of accuracy
```

ConnorKirkpatrick, 2025-03-02:
Having implemented the changes, I can confirm this new system works to prevent the boot issues with i2c and the switch components. By being able to delay the i2c setup to after the GPIO switch changes to high, the "SCL held low" errors disappear, meaning that i2c devices can be safely powered off during sleep phases and recovered without issue. A full PR will be opened once I have made a PR for the documentation.

Minimal yaml:
```
esphome:
  name: test
  on_boot:
    - priority: 700 #set to 700 to be before the default sensor setup (600)
      then:
        - switch.turn_on: energise
  on_shutdown: 
      then:
        - switch.turn_off: energise

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

logger:

api:
  password: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: "someWifi"
  password: "somePass"
    
i2c:
  sda: GPIO5
  scl: GPIO6
  priority: 601 #NEW, set the setup priority to 601, just before the sensors are initialised

switch:
  - platform: gpio  # switch controlling the power to the i2c device (bmp280)
    pin: GPIO4
    name: energise
    id: energise

sensor:
  - platform: bme280_i2c
    address: 0x76
    update_interval: 10s
```

Output of the esp32:
```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[18:42:55][I][logger:171]: Log initialized
[18:42:55][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[18:42:55][D][esp32.preferences:114]: Saving 1 preferences to flash...
[18:42:55][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:42:55][I][app:029]: Running through setup()...
[18:42:55][C][switch.gpio:011]: Setting up GPIO Switch 'energise'...
[18:42:55][D][switch:016]: 'energise' Turning OFF.
[18:42:55][D][switch:055]: 'energise': Sending state OFF
[18:42:55][D][switch:016]: 'energise' Turning OFF.
[18:42:55][D][switch:012]: 'energise' Turning ON.
[18:42:56][D][switch:055]: 'energise': Sending state ON
[18:42:56][I][i2c.arduino:218]: Performing I2C bus recovery
[18:42:56][C][bme280.sensor:091]: Setting up BME280...
[18:42:56][C][wifi:048]: Setting up WiFi...
[18:42:56][C][wifi:061]: Starting WiFi...
[18:42:56][C][wifi:062]:   Local MAC: 34:85:18:AB:FF:24
[18:42:56][D][wifi:482]: Starting scan...
[18:42:56][W][component:157]: Component wifi set Warning flag: scanning for networks
<REDACTED Wifi connection>
[18:43:02][C][api:026]: Setting up Home Assistant API server...
[18:43:02][I][app:062]: setup() finished successfully!
[18:43:02][W][component:170]: Component wifi cleared Warning flag
[18:43:02][W][component:157]: Component api set Warning flag: unspecified
[18:43:02][I][app:100]: ESPHome version 2025.3.0-dev compiled on Mar  2 2025, 18:42:38
[18:43:02][C][wifi:600]: WiFi:
<REDACTED Wifi information>
[18:43:02][C][logger:177]: Logger:
[18:43:02][C][logger:178]:   Max Level: DEBUG
[18:43:02][C][logger:179]:   Initial Level: DEBUG
[18:43:02][C][logger:181]:   Log Baud Rate: 115200
[18:43:02][C][logger:182]:   Hardware UART: USB_CDC
[18:43:02][C][switch.gpio:068]: GPIO Switch 'energise'
[18:43:02][C][switch.gpio:091]:   Restore Mode: always OFF
[18:43:02][C][switch.gpio:031]:   Pin: GPIO4
[18:43:02][C][i2c.arduino:071]: I2C Bus:
[18:43:02][C][i2c.arduino:072]:   SDA Pin: GPIO5
[18:43:02][C][i2c.arduino:073]:   SCL Pin: GPIO6
[18:43:02][C][i2c.arduino:074]:   Frequency: 50000 Hz
[18:43:02][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[18:43:02][I][i2c.arduino:096]: Results from i2c bus scan:
[18:43:02][I][i2c.arduino:102]: Found i2c device at address 0x76
[18:43:02][C][bme280_i2c.sensor:025]:   Address: 0x76
[18:43:02][C][bme280.sensor:182]: BME280:
[18:43:02][C][bme280.sensor:194]:   IIR Filter: OFF
[18:43:02][C][bme280.sensor:195]:   Update Interval: 10.0s
[18:43:02][C][bme280.sensor:198]:     Oversampling: 16x
[18:43:02][C][bme280.sensor:200]:     Oversampling: 16x
[18:43:02][C][bme280.sensor:202]:     Oversampling: 16x
[18:43:02][C][captive_portal:089]: Captive Portal:
[18:43:02][C][mdns:116]: mDNS:
[18:43:02][C][mdns:117]:   Hostname: test
[18:43:02][C][esphome.ota:073]: Over-The-Air updates:
[18:43:02][C][esphome.ota:074]:   Address: test.local:3232
[18:43:02][C][esphome.ota:075]:   Version: 2
[18:43:02][C][safe_mode:018]: Safe Mode:
[18:43:02][C][safe_mode:020]:   Boot considered successful after 60 seconds
[18:43:02][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:43:02][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[18:43:02][C][api:140]: API Server:
[18:43:02][C][api:141]:   Address: test.local:6053
[18:43:02][C][api:145]:   Using noise encryption: NO
```

ssieb, 2025-03-02:
You're not adding anything.  You can already use setup_priority for the i2c component.

ConnorKirkpatrick, 2025-03-02:
Well that's slightly frustrating. Just double checking, this information is not available on the normal documentation page anywhere is it? I can't see a reference in the i2c area, and there does not appear to be an generic component page.

If we agree that I'm not blind and this information does not exist inside the docs anywhere, I'll close the core PR and alter the docs PR to add this information somewhere sensible.

ssieb, 2025-03-02:
It's not documented because we don't want people generally using it as it's more likely to cause problems than fix them.

ConnorKirkpatrick, 2025-03-03:
While I can agree hiding functionality that could cause issues, this has been the third attempt to fix/get around an issue that doesn't exist simply due to missing documentation.

I could put together a sub-page/section on the docs to go over the setup behavior and priorities in general along with the setup_priority function specifically. To keep with the sentiment of this being an advanced function, I could also annotate the entire section with a warning to clarify that these functions exist for advanced use-cases only, and to only make use of them if the user is confident in their understanding. 
This would prevent others from spending time fixing an issue that is already solved and improve the documentation set at the same time.

Does this seem like an acceptable action?

ssieb, 2025-03-03:
Come start a discussion in the documentation channel on discord.

closed 2025-03-04 by ConnorKirkpatrick