esphome/feature-requests #3074: UDP: Events/Actions

Cossid, 2025-03-05:
**Describe the problem you have/What new integration you would like**

The ability to send events/actions via UDP, such as button press.

**Please describe your use case for this integration and alternatives you've tried:**

UDP is currently limited to binary_sensor and sensor, which is limited in what things can be sent and what they mean.  Events wouldn't need the ability to sync, but rather be on-time, one-way events (perhaps with an ack).  For example, I've tried to share "brightness" changes through a sensor, but it is very easy to get stuck in a feedback loop, when the sync comes back, and one device is ahead of the other in a short period of time (think holding a button down sending several relative brightness increase/decrease commands).

This would make it easier for "remote" devices to send commands to the "primary" device, which then would be in charge of syncing all devices with regular sensor syncing.

**Additional context**

I'm not sure the best way to implement this and keep it dynamic or how to format the data of the event/action.  My first though was to possibly use text sensors, but that doesn't seem right.  Maybe send an integer which defines a pre-defined action.

nagyrobi, 2025-03-05:
To control RF covers I use a single node to transmit the signal for all. The covers themseves are all on this node.

To act from various touchscreens around the house on them, I just use a float, where the integer part denotes the cover, and the decimal part is just the desired position of the cover. No need for any ack in my case.

clydebarrow, 2025-03-21:
The problem with sending events via UDP is that he transmission is unreliable, so they can get lost. Sending state from a sensor can be repeated at intervals to mitigate this problem. There is work in progress to unbind the packet transport encoding from UDP which would enable a TCP implementation, which would make it practical to send events since TCP is reliable, but even then there can be dropped data if the network connection is interrupted.

So as @nagyrobi suggests, it's best to denote one node as the source of truth. Using your example, to control brightness from multiple nodes, have them send up or down "events" implemented as binary_sensor presses to the master node, which maintains the actual brightness and sends it back as a sensor value to update the other nodes if required. If some button presses are lost it's not a big deal.

You can use long-press repeats for example to send multiple presses.

Cossid, 2025-03-21:
Yeah, I certainly look forward to when this can be done with TCP via `packet_transport`.  I understand UDP is not guaranteed, but would still like to see events none-the-less.

I already have mine set up as a "source of truth", the problem is when one of the other devices needs to be the one that updates the source, hence the events.  Syncing works fine for binary, but brightness gets stuck in a loop, where it will constantly flood even the standing value unless you add a filter.  I've had to set up hacky globals to temporarily stop listening when being the one sending new states to the source of truth to send out to other devices.

The example of brightness was just a simple one.  A more complex, but common example would be syncing a RGBCT light, and doing that with just `binary_sensor` and `sensor` is difficult and verbose, as there are a minimum of 8 sensors, and sending them individually may not always turn out right (like sending colors separately from color_mode).

Obviously expanding to more sensors is another way to achieve this goal, but there are a lot, and an event would be a more generic solution in the meantime.