esphome/feature-requests #3079: Show build details when selecting "Plug into this computer" as install method

whc2001, 2025-03-11:
**Describe the problem you have/What new integration you would like**
When selecting OTA or local download as installation method, the build and upload logs is shown nicely. However if local serial is selected, it will only show "Preparing Installation" and a spinner without any more details.

**Please describe your use case for this integration and alternatives you've tried:**
I hope it could display detailed logs in the same style as OTA. Currently I can press F12 to open the devtools and it does log things to the console, but just display on the page would be more convenient.

**Additional context**
Instead of this:

[image]

I'd prefer this:

[image]

whc2001, 2025-07-13:
> +1 for this. I always select Download first, then dismiss the dialogue when it asks for download format. Then, I click Install again and select Plug into this computer again... And if the download fails, I check with esptool.py from command line to see the issue...

You can press F12 to open the browser development console to see the verbose log for downloading. However I think it would be nice to also be able to view the detailed download log in the page.

xbojer, 2025-10-29:
I would like this too