esphome/feature-requests #3088: Support for optionally disabling hardware rmt support for remote_receiver when using ESP32 IDF

juanboro, 2025-03-18:
I'd like to request the ability to optionally just use a software based interrupt based GPIO handler (like the ESP8266 does) selectively for the remote receiver, instead of using the [Remote Control Peripheral](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#rmt ) hardware.

This could be enabled by potentially adding a configuration option of something like  use_esp32_rmt in this exmaple:
```yaml
remote_receiver:
  - id: rf_receiver
    use_esp32_rmt: false
```

The reason for this request is because the rmt hardware on older ESP-32 hardware is limited to the remote symbol memory for the size of the receive.  For IR remotes - generally this is fine.  But for signals coming from RF, often the transmission size is significantly larger, resulting in the error message: hw buffer too small, received symbols truncated.

This becomes even more critical of an issue when multiple remote receivers and transmitters need to be setup, given that rmt_symbols needs to be large enough for each remote component.  A specific example would be an esphome project to mimic the behavior of a device such as [this](https://www.ibroadlink.com/productinfo/762672.html).  In this case, remote_receiver component are needed for a RF receiver and an IR receiver, and remote_transmitter components are needed for RF and IR transmitters.  This will leave only 192symbols (on/off pulses) for the RF receiver - which can be too small for some RF codes (my use case is decoding a variety of devices directly in ESPHome via rtl_433 via [this external component](https://github.com/juanboro/esphome-rtl_433-decoder/tree/main) I've created.

By allowing the RF remote receiver to not use the rmt hardware, the rmt hardware can be better utilized where it does work well (the transmitters which do not have this limitation, and the IR receiver which should always have smaller symbol sizes).

I have a working [fork](https://github.com/esphome/esphome/compare/dev...juanboro:esphome:esp32_soft_rmt) that does this I'd like to create a draft PR for assuming I'm not misreading something in the ESP-IDF about how this all works (also would like input from @swoboda1337 )

swoboda1337, 2025-03-18:
A few comments:
- The IDF version is very similar to Arduino, the same symbol restrictions apply, so IDF isn't relevant, this is an ESP32 thing IDF has nothing to do with it.  
- That error message isnt an error and espressif merged my PR to change that to a debug print. Most of the time it is just noise that triggers it. PR is here: https://github.com/espressif/esp-idf/pull/15225
- The only two ESP32s that don't support ping pong rx are the original and S2. The original has a lot of memory and S2 only has 256 with one tx would leave 192 symbol pairs or 384 on/off pulses since each symbol is a pair. 

Is 384 on/off pulses too short? Are there many signals longer than that? Does this include repetitions? I know my weather stuff is longer but if you set the idle correctly it breaks the repetitions in to separate chunks that fit nicely.

juanboro, 2025-03-18:
> * The IDF version is very similar to Arduino, the same symbol restrictions apply, this is an ESP32 thing, IDF has nothing to do with it.

Yes - agree.  I mainly was just asking for ESP-32 because I wasn't sure of the status of whether both IDFs will continue in the future or just one.  It gets a little more difficult (as it already is) to have something like this for both IDFs.  
> * That error message isnt an error and espressif merged my PR to change that to a debug print. Most of the time it is just noise that triggers it. PR is here: [change(rmt): change rx buffer prints from error to debug (IDFGH-14449) espressif/esp-idf#15225](https://github.com/espressif/esp-idf/pull/15225)

Agree the PR merge is good and yes, most of the time it is just noise that is generating those messages.
> * The only two ESP32s that don't support ping pong rx are the original and S2. The original has a lot of memory and S2 only has 256, with one tx would leave 192 symbol pairs or 384 on/off pulses since each symbol is a pair.

That is a good fact to know --- I still have a bunch of the originals... but wasn't able to find any clear documentation on when dma and ping-pong were supported on the newer ones (of which I have some of).
> Are 384 on/off pulses too short? Are there many signals longer than that? Does this include repetitions? I know my weather stuff is longer but if you set the idle correctly it breaks the repetitions in to separate chunks that fit nicely (may be a problem with a variety of devices).

How many signals is an unknown for me still --- it's mainly more just one less thing to debug with all the RF stuff.  Generally- with my stuff so far, yes 192 pulses usually cover it, because the rest that is getting lost is repetition.  The real production concern I have is my leak detectors can get kind somewhat distant from the receiver - so the multiple repeats they send help with the chance the message might make it through uncorrupt 1/3 times say for example.
> Is it really worth adding a few 100 lines just to make S2 work with this very specific use case? Maybe just get a newer ESP32?

Do I personally think it is worth it? - yes (there are still alot of boards out there with the original esp32 that don't support the improved rmt).  But it's also valid argument that maybe I just keep the patch to the side as is for this very specific use case.  For ESPHome.  I can see arguments either way (including also just maybe making the 100 lines basically common between esp32/8266/libretiny/etc.

swoboda1337, 2025-03-18:
Some day we will be able to upgrade Arduino to a newer version with the new RMT drivers and remove the old implementation. 

What version of the ESP32 do you have? The original has 512 RMT symbols, that equals 1024 on/off pulses (again each symbol is a pair of two). 

The default value is 192 rmt symbols but feel free to increase that to to 512 (1024 on/off pulses) or 448 (896 on/off pulses) if you want to do tx. S2 is the only one that is limited it 192 (384 on/off pulses). 

Yeah it would be ideal if all three could share the same sw implementation instead of having 3 copies. Not sure how to do that cleanly though.

swoboda1337, 2025-03-18:
Oh and to check support for ping pong rx just grep the platformio folder:
```
components/soc/esp32c3/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32h2/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32s3/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32p4/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32c61/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32c6/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32c3/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32c5/mp/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32c5/beta3/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32h2/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
components/soc/esp32s3/include/soc/soc_caps.h:#define SOC_RMT_SUPPORT_RX_PINGPONG
```

juanboro, 2025-03-18:
> Some day we will be able to upgrade Arduino to a newer version with the new RMT drivers and remove the old implementation.
> 
> What version of the ESP32 do you have? The original has 512 RMT symbols, that equals 1024 on/off pulses (again each symbol is a pair of two).
> 
> The default value is 192 rmt symbols but feel free to increase that to to 512 (1024 on/off pulses) or 448 (896 on/off pulses) if you want to do tx. S2 is the only one that is limited it 192 rmt symbols if you want to use tx (384 on/off pulses).

Yeah- I have the original.  It denied me turning on dma (and as far as I could tell the ping-pong notify almost full event).  I already increased it for my test board at my home address - but the one I have at my remote site is serving multiple purposes and includes IR transmit and receive for climate control - so I'm down to 192 symbols available.  Notes from my yaml configuration there:
    ## 448-192(IR RCV)-64(IR XMIT)
    rmt_symbols: 192

Separately - I had trouble with the Arduino platform getting transmit to work nicely alongside receive on the shared gpio  (I could not with the latest releases/dev branch).  I did not spend alot of time trying to debug it though.

>Yeah it would be ideal if all three could share the same sw implementation instead of having 3 copies. Not sure how to do that >cleanly though.
Time permitting - I'll take a deeper look at it on my branch and see if it's feasible.

swoboda1337, 2025-03-18:
> Separately - I had trouble with the Arduino platform getting transmit to work nicely alongside receive on the shared gpio (I could not with the latest releases/dev branch). I did not spend alot of time trying to debug it though.

Yes it isn't easy to do on Arduino, you had to do run setup again on remote_transmitter:
```
remote_transmitter:
  id: tx_id
  pin:
    number: GPIO32
    allow_other_uses: true
  carrier_duty_percent: 100%
  on_transmit:
    then:
      - sx127x.set_mode_tx
      - lambda: id(tx_id)->setup(); // workaround
  on_complete:
    then:
      - sx127x.set_mode_rx
```
Now on IDF its very flexible, can use open drain mode and set the eot_level to false:
```
remote_transmitter:
  id: tx_id
  pin:
    number: GPIO32
    mode:
      input: true
      output: true
      pullup: true
      open_drain: true
    allow_other_uses: true
  eot_level: false
  carrier_duty_percent: 100%
  on_transmit:
    then:
      - sx127x.set_mode_standby
      - lambda: 'id(tx_id)->digital_write(false);'
      - sx127x.set_mode_tx
  on_complete:
    then:
      - sx127x.set_mode_standby
      - lambda: 'id(tx_id)->digital_write(true);'
      - sx127x.set_mode_rx
```

juanboro, 2025-03-18:
>Yes it isn't easy to do on Arduino, you had to run setup again on remote_transmitter:
Yeah - that *used* to work for me.... but last I tried with the latest dev branch and platform io/etc, adding the transmit component broke receive working completely.

swoboda1337, 2025-03-18:
> > Yes it isn't easy to do on Arduino, you had to run setup again on remote_transmitter:
> > Yeah - that _used_ to work for me.... but last I tried with the latest dev branch and platform io/etc, adding the transmit component broke receive working completely.

Maybe because the setup priority of remote transmitter changed. If you change it back in the yaml with `setup_priority` it might work.

juanboro, 2025-03-18:
> Maybe because the setup priority of remote transmitter changed. If you change it back in the yaml with `setup_priority` it might work.
Yeah -  I _think_ I may have tried that (I know I thought of trying it, not sure if I actually did :-) ) --- but will try it again when I update my patch to try to unify the software remote recv across the platforms.

juanboro, 2025-03-28:
I've updated my branch to be closer to what I think would be desirable:
- Common standard remote_receiver component that is software based interrupt handler and works on all platforms that are currently supported on the dev branch (including esp32).
- Additional support on esp32 (by default) for rmt in specific remote_receiver component for esp32 (supporting both platforms).  Can be disabled on both platforms by setting ```use_esp32_rmt: false```
- I think this makes all the code easier to understand (less #ifdefs), and hopefully alot simpler.

I still have alot more testing to do before I make it a PR (and then very little hope the PR will ever be accepted given the state of backlog of PRs including a pretty simple on I have that fixes RC5 decoding right now). But it's at least all very useful for my needs.

juanboro, 2025-05-15:
Ok - Adding a few more data-points from my notes to this request.  

1.  Example test case:
Here is an example of the type of _kindof_ problematic signal in question:
[image]

The problem is for rtl_433 (or if I were to attempt to write a custom esphome PWM decoder for it) - that 9ms reset limit is kind of problematic:
```c
r_device const govee_h5054 = {
        .name        = "Govee Water Leak Detector H5054",
        .modulation  = OOK_PULSE_PWM,
        .short_width = 440,  // Threshold between short and long pulse [us]
        .long_width  = 940,  // Maximum gap size before new row of bits [us]
        .gap_limit   = 900,  // Maximum gap size before new row of bits [us]
        .reset_limit = 9000, // Maximum gap size before End Of Message [us]
        .decode_fn   = &govee_h5054_decode,
        .fields      = output_fields,
};
```
You need at least a 9ms idle time to set end of message - but the signal also repeats with the same limit (setting idle<9ms will result in no decode of anything).  As far as the rmt rx buffer overlflow --  as @swoboda1337 points out - this signal is really just repeating - so as long as one of the earlier messages makes it uncorrupted - it will decode anyway.  So in my use case - it's really just my paranoia around loosing the redundancy since these transmitters aren't real good (leak detectors all over the house, but only one esp32 in my one house at least).  And yes I agree to those that will ask ---  :-) --- arguably I could add more receivers and more redundancy that way too.  In part looking at all this has been an effort for me to just learn the inner workings of esphome and esp-idf more.

2.  I do have a working patch [branch](https://github.com/juanboro/esphome?tab=readme-ov-file) that does add the optional software decode for older esp32 devices that works great for my use case (and maybe anyone else that would need it).  It also (I think) cleans up/simplifies slightly the remote receive on all the platforms.  But it is NOT at all PR quality (in particular I'm pretty sure I did not implement config val/code generation "correctly"... it also needs documented.  It might be it just makes sense for me to just keep this as a fork branch for my pretty specific uses.  Devs can feel free to close this request if it isn't worth it (or alternatively I'll need assistance to make a proper PR).
3.  Related to this as I mentioned above - with the CC1101 I haven't been able to get transmit and receive working with the Arduino platform one way or the other (soft remote or hard remote) - and i tried a bunch of things.  The best I can achieve on that platform is RX working, then TX works, but then RX busted after doing the 1st TX.  I don't seem to be able to get the Arduino rmt transmit to leave the pin open-drain pullup mode high at EOT.  It really doesn't matter for me anyway (does it for anyone else?) ... the esp-idf platform with the new rmt code works great on my older socs/boards  (and it looks like it does now for all the newer socs as well).

swoboda1337, 2025-05-15:
Regarding Arduino it would ideal if we could upgrade that to a newer version and that would solve the problem, would be the same as IDF. I started a PR but its still very early: https://github.com/esphome/esphome/pull/8604 and there are a few issues to work through. 

Regarding the problematic signal why cant you break that up into seven pieces? Why wouldn't it decode? Reset is 9 ms but the gap is 0.9 ms? As discussed above there is only a hard limit on esp32 and esp32 s2, newer esp32s there is no limit on the receive size it can huge even like 10k symbols.

juanboro, 2025-05-15:
> Regarding Arduino it would ideal if we could upgrade that to a newer version and that would solve the problem, would be the same as IDF. I started a PR but its still very early: [esphome/esphome#8604](https://github.com/esphome/esphome/pull/8604) and there are a few issues to work through.

Yeah - Agree - that should do it and will be good all around.  Looking forward to went that goes out 👍.

> Regarding the problematic signal why cant you break that up into seven pieces? Why wouldn't it decode? Reset is 9 ms but the gap is 0.9 ms? As discussed above there is only a hard limit on esp32 and esp32 s2, newer esps there is no limit on the receive size it can huge even like 10k symbols.

The problem for [rtl_433 ](https://github.com/juanboro/esphome-rtl_433-decoder) as best I can tell is just the at recognizing the end of message.  It wants the last gap to be at least 9ms or more (with some tolerance for it to be a little less).  Now I could hack the message going into rtl_433 with on_raw lambda  to just alter the final gap to be 9ms (versus the idle value) and it probably will work - but that is definitely uglier.  I also _could_ (and may just do for the learning process) attempt to write an actual ESPhome remote encoder/decoder for these things vs using rtl_433.  But afaik - there are no existing PWM decoders/encoders to borrow from as examples right now so it would be some effort (vs PPM signals which are super easy).  Like you say - the newer ESPs all support this fine anyway without - so it's mostly academic anyway... other than just having a little more flexibility...

One thing I noticed related to this when I was looking at some IR signals -- on all the esp remote stuff (that I think _probably_ should be noted in the documentation) --- the maximum gap/pulse size that can be encoded in the rmt symbol is 15 bits - with the size being set by rmt_ticks (default 1us) - means the maximum idle time that you can set that makes any sense is around 33ms.  If you set it higher - the longer pulse/gaps will clip at 32768us.  This shows up for example on NEC remotes - where i think the repeat indicator is like 110ms from the main signal -- it's not fully clear whether it makes sense to separate the repeat signal out (<110ms idle), or include it - but if you include it (i.e set a 150ms idle time), then the gap  gets pulled in to 33ms. So it probably makes sense to not have it that long - but then receiving and transmitting the repeat sequence is pretty hard with esphome --- right now at least.  It looks like this PR [8015](https://github.com/esphome/esphome/pull/8015) deals with that reasonably.

swoboda1337, 2025-05-15:
Writing your own decoder is pretty simple. I converted the acurite one: https://github.com/merbanan/rtl_433/blob/8e813dda1c64e2f8b8b3f0cc16e6fa74fb6fb6d0/src/devices/acurite.c#L1912

Can see my version, its only a few lines: https://github.com/swoboda1337/acurite-esphome/blob/main/components/acurite/acurite.cpp#L266

Should be pretty much the same except acurite has a sync and yours doesnt.

juanboro, 2025-05-15:
> Writing your own decoder is pretty simple. I converted the acurite one: https://github.com/merbanan/rtl_433/blob/8e813dda1c64e2f8b8b3f0cc16e6fa74fb6fb6d0/src/devices/acurite.c#L1912
>
> Can see my [...]

I already wrote at least one custom encoder/decoder for my ceiling fans... which was very straightforward... just wasn't sure about how hard to do for this case - but it looks like you example is for a pwm source as well - so should be pretty helpful.  thanks.