esphome/feature-requests #3091: Add Zigbee Firmware support for ESP32-C6 to ESPHome Device Builder

gppprimo, 2025-03-20:
**Describe the problem you have/What new integration you would like**

As far as I know, ESPHome Device Builder currently cannot flash firmware with Zigbee protocol support. 
I carefully search for open issues to find threads where the discussion has been introduced but I was unsuccessful. However, I was wondering if it might be possible to customize the firmware-building process to also support Zigbee for ESP modules that are compatible with Zigbee, of course.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I don't know if there are branches or feature proposals that have already been analyzed concerning this, but I envision a feature to introduce during the creation of a new device.

[image]

When clicking "New Device," the system suggests the type of firmware communication to create (Wi-Fi/Zigbee).

It would be a good start, considering that new ESP development boards will increasingly support the Zigbee protocol.
Obviously, I can imagine that it would lead to significant modifications in the ESPHome Device Builder core.

Hedda, 2025-04-13:
Duplicate of https://github.com/esphome/feature-requests/issues/1397

Hedda, 2025-06-16:
@jesserockz Please reopen https://github.com/esphome/feature-requests/issues/1397 since original post there and all discussion there is asking for Zigbee support more than Thread.

- https://github.com/esphome/feature-requests/issues/1397

Anyway, I would even argue original post there is more about adding Zigbee device support than it is about adding Thread support.

PS: There is an open PR for Zigbee support but it only for nrf52 platform (which looks nearly ready) however missing ESP32 support:

* https://github.com/esphome/esphome/pull/7340

  * https://github.com/esphome/esphome/pull/7049