esphome/feature-requests #3092: Develop a configuration package feature to prevent recompiling for identical devices

titchy, 2025-03-21:
**Describe the problem you have/What new integration you would like**

It would be very handy if esphome treated identical devices as one compile job and then just the differentiating configuration added to the upload.

**Please describe your use case for this integration and alternatives you've tried:**

Updating my set of room sensors does 8 compile jobs when it could compile once and send 8 different configurations with the binary. This makes updates a long process which means I sometimes put it off when that is probably not what I should be doing.
I may have missed something but I can not work out how to do this based on the documents I have read so far. I know on ESP32 it could partition and use a separate partition to store a configuration store, but that may impact other platforms.

**Additional context**

ssieb, 2025-03-21:
There is no "configuration" on the device.  The yaml file is converted to C++ code and compiled to a binary.
If you want the update process to be shorter, then run esphome on something more powerful, like a desktop computer.
There is also no need to update every release unless there's something specific you need from it.

titchy, 2025-04-17:
I understand that there is no configuration on the device, though with the partition supported it would be possible. I am however  talking about making the build system a bit more intelligent so that I can define a set of identical hardware as using a specific set of libraries and core code and then have that compiled once and the configuration data only pulls in values at linking time. For most of these duplicate devices we are talking about device name and maybe a password or two being the difference, the rest of the code is exactly the same as the devices are identical