esphome/feature-requests #3098: Image component: support for Material Design Light Icons and Memory Icons

tomaszn, 2025-03-23:
**Describe the problem you have/What new integration you would like**

The "image" component already supports Material Design Icons. To use an icon, one needs to use `mdi:` schema in the `file` configuration variable. [The documentation](https://esphome.io/components/image.html#configuration-variables) points to the [MDI library](https://pictogrammers.com/library/mdi/) where one can browse icons.

I would like to use icons from the "memory" set available [at the same website](https://pictogrammers.com/library/memory/). I can't find a way to use icons from sets other that "mdi".

**Please describe your use case for this integration and alternatives you've tried:**

Icons from the "memory" set would be perfect for my project with a monochrome LED display. Alternatively, I could download the SVG images myself and add to the project files. That is a tedious task.

**Additional context**

As importing icons from that website is already implemented, adding support for "mdil:" (Material Design Light Icons) and "memory:" (Memory Icons) should be an easy task.

tomaszn, 2025-03-25:
@nagyrobi, are you sure we need a webfont? looks like for MDI we are downloading an SVG image:
https://github.com/esphome/esphome/blob/ce2e96600505f8c9416c1b07205753b631e78719/esphome/components/image/__init__.py#L319

clydebarrow, 2025-05-03:
https://github.com/esphome/esphome/pull/8676#issuecomment-2848541949

closed 2025-05-05 by jesserockz