esphome/feature-requests #3104: ac_dimmer add support for rellay

teejayek, 2025-03-27:
**Describe the problem you have/What new integration you would like**
I would like to extend ac_dimmer component with support for rellay. It would require optional parameter for rellay pin. If set then set relevant GPIO to 1 if power is more than equal to MIN POWER and set it to 0 if it  is below MIN POWER.

**Please describe your use case for this integration and alternatives you've tried:**

When I try to use this component with Makerfabs AC_ dimmer (https://github.com/Makerfabs/ESP32-WiFi-AC-Dimmer/tree/main) I get no power on output as it requires to switch on rellay. It can be seen in Makerfabs sample code.
The use case is using this 2000W dimmer (with cooler added:-) to controll watter heater (boiler) based on electricity grid overflow (Generated by photovoltaic power plant). I like this HW component as it is easy to put on DIN rail, it is in the case,... ready to use.

**Additional context**
as an example of YAML, it would have:
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO25
    **rellay_pin: GPIO26**
    zero_cross_pin:
      number: GPIO33
      mode:
        input: true
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmerized Light

ssieb, 2025-03-27:
If you're using a `light` (you should set the gamma correct to 1), then you can add a `power_supply` control to do what you want.  Adding power supply support to outputs could be a good idea.  For now, if you aren't going to use the light, you could use a template output in between that handles the power control.

teejayek, 2025-04-01:
BAsed on suggestion from developer, I am using power_supply component within ac_dimmer linked to GPIO and it works fine. That is great solution, thank you.

closed 2025-04-01 by teejayek