esphome/feature-requests #3108: Meshtastic support

Paxy, 2025-04-01:
**Describe the problem you have/What new integration you would like**
Support for Meshtastic Lora network and sx126x chips. Meshtastic provides a great, community-driven Lora broadcast network for sensor communications over long distances. Meshtastic already supports many ESP32-based chips and SX126x modems but lacks sensor support and customization similar to ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
If we could use the Meshtastic project for Lora to MQTT (an existing solution) to Home Assistant on the gateway side and ESPHome with the Meshtastic Lora network on the client side, we can achieve both flexibility and customization of both projects. This would make remote sensors easily and cheaply available to HomeAssistant.

**Additional context**

swoboda1337, 2025-04-07:
There is a PR open for SX126x support here: https://github.com/esphome/esphome/pull/8516

You can use that and have two esp32s talk to each other over LoRa.

swoboda1337, 2025-04-07:
There is an effort to make a generic packet transport: https://github.com/esphome/esphome/pull/8187

That would allow easy creation of remote sensors.

Paxy, 2025-04-09:
That's great news.
Meshtastic would probbably be the next step as it adds Meshtastic networking capabilities over simple SX126x send and receive. With Meshtastic protocol you can use Meshtastic repeating functionality with existing Meshtastic devices, capability to use separate channels and encrypt data over those channels. Also, there are predifined network modes for different repeting options (dense of sparse areas). These is somthing that can be useful to be the part of ESPHome component due to huge support of different sensors.

dala318, 2025-04-23:
Had some look into https://github.com/meshtastic/firmware and made a semi-serious attempt to wrap the mesh-networking stuff of it in an ESPHome custom_component (include as git-submodule). But very early realized that there is a lot of Arduino.h dependencies in there.
While that no complete blocker, since ESPHome has the option to use the Arduino framework, I would prefer to have such component independent of framework. And without possibility of re-use from that project repo it seem to be a lot of work to re-implement a lot of what's in the mesh folder. Maybe there is a smaller part of it to just act as a node, not being a full-fledged mesh-proxy capable device and gateway.

Jachimo, 2026-01-19:
It appears that Meshtastic has a native "IP Tunneling" capability, which presumably (I haven't tested it) includes UDP. Could that be used with the existing UDP packet transport? You would need two ESP32 modules and they would need to be on the same WiFi (or otherwise have an IP connection between them).

A way of doing packet transport over UART would be nice though, and would allow for a compact Meshtastic/ESPHome bridge node.