esphome/feature-requests #3111: Support for LN882H

omersht, 2025-04-05:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

LibreTiny now supports the LN882H chips, that are becoming more popular in Tuya devices. I wanted to know will it be possible to support does in esphome

rbonat, 2025-06-25:
+1, have 4 relays based on LN882HK. Since it is supported by LibreTiny, shouldn't be hard to add support in esphome.

omersht, 2025-07-04:
Support is present on the dev, (though for some reason captive portal doesn’t compile)

closed 2025-07-04 by omersht

Marcin685, 2025-08-03:
i have this error 

[image]