esphome/feature-requests #3113: GPS ESP-IDF

alexsavuto, 2025-04-06:
Hi, is there any way to add a GPS component to ESP-IDF? I got a message saying 'This feature is only available with frameworks ['arduino']'

Hiekkaharju, 2025-07-19:
+1
With increasing memory footprint of other parts of esphome it's getting difficult to keep using Arduino.  
btw: did not see any notice about Arduino-only in https://esphome.io/components/gps.html#gps-component
Any possibility to have esp-idf GPS-component @coogle @ximex ?

ximex, 2025-07-24:
see https://github.com/esphome/esphome/pull/9728
should already work with IDF on dev

Hiekkaharju, 2025-07-24:
Thanks !
I'll try that when I've time.

Hiekkaharju, 2025-07-25:
That works - kind of
Just github://pr#9728 on top of esphome 2025.7.2 and gps compiles and kind-of works also with esp-idf.
But with my setting
```
gps:
  update_interval: 2s
```
,and the device moving, I get new coordinates practically every 2 sec when using Arduino.
But the same configuration, with just that PR and ESP_IDF as platform, coordinates appear to be stuck for tens of seconds.
With Verbose-logging Sensor-component shows "Received new state" messages for gps-sensors every 2 sec, but the longitude and latitude values in them are frozen for long times.
This is a slowly moving device, so the changes in the coordinates should be small, typically 5th/6th decimal changing, but I can only see that with Arduino.
Despite it working without it in Arduino, tried also accuracy_decimals: 6 but it made no difference.

Hiekkaharju, 2025-07-26:
I did some "debugging" with logs and the slowdown of gps has nothing to do with IDF vs Arduino.
It's the 
```
  if (this->tiny_gps_.encode(this->read())) {
```
to
```
    if (!this->tiny_gps_.encode(this->read())) {
      return;
    }
``` 
change in https://github.com/esphome/esphome/pull/9728
With that "return" the reading does not succeed often enough and the update frequency therefore is very varying and up to tens of seconds. 
I'm running on m5stack Atom lite i.e esp32-pico

@kbx81

kbx81, 2025-07-29:
Interesting, I did not encounter that when I was testing here. I'll give it another try as it's possible I didn't let it run long enough to notice the reduction in frequency of the updates.

kbx81, 2025-07-29:
@Hiekkaharju please try the linked PR with:

```yaml
external_components:
  - source: github://pr#9953
    components: [gps]
    refresh: 1h
```

...and let us know if this fixes the issue for you. Thanks!

Hiekkaharju, 2025-07-29:
Thanks.
Yes, pr 9953 fixes the slowdown issue caused by 9728

While it makes little difference, FYI: I now tried this only with IDF.