esphome/feature-requests #3114: Support new Firmware of Pylontech Battery

Einstein098, 2025-04-06:
**Describe the problem you have/What new integration you would like**

Pylontech has change in a Firmwareupdate the Table after the command "pwr" 

I think we need an update of this integration to use both Firmware. 

**Please describe your use case for this integration and alternatives you've tried:**

The Pylontech intergration does not work with ny Battery. 
I use this intergration to get the data over MQTT: 
https://github.com/hidaba/PylontechMonitoring
I change the parce of the Value (pLineStart) to the new values now it work 

**Additional context**

The Output of the command "pwr" of my Battery: 

`pwr

@

Power Volt   Curr   Tempr  Tlow   Tlow.Id  Thigh  Thigh.Id Vlow   Vlow.Id  Vhigh  Vhigh.Id Base.St  Volt.St  Curr.St  Temp.St  Coulomb  Time                 B.V.St   B.T.St  MosTempr M.T.St   SysAlarm.St

1     48844  -33847 17300  16300  12       17200  0        3251   13       3260   8        Dischg   Normal   Normal   Normal   89%      2025-04-07 02:01:11  Normal   Normal  17700    Normal   Normal  

2     48895  -22701 18900  17200  10       17700  5        3255   0        3262   2        Dischg   Normal   Normal   Normal   84%      2025-04-07 02:01:08  Normal   Normal  18100    Normal   Normal  

3     48889  -22950 19200  17600  10       17900  5        3254   5        3262   2        Dischg   Normal   Normal   Normal   82%      2025-04-07 02:01:08  Normal   Normal  18900    Normal   Normal  

4     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

5     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

6     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

7     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

8     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

9     -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

10    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

11    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

12    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

13    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

14    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

15    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

16    -      -      -      -      -      -      -      Absent   -        -        -        -        -                    -        -       

Command completed successfully

$$

pylon_debug>`

New is 

Tlow.Id  
Thigh.Id 
Vlow.Id  
Vhigh.Id 
SysAlarm.St

My Battery Info

`info

@

Device address      : 1

Manufacturer        : Pylon

Device name         : US5000

Board version       : V10R04

Board               : NF4.E3

Main Soft version   : B69.25.0.0

Soft  version       : V2.0

Boot  version       : V1.0

Comm version        : V2.0

Release Date        : 24-05-13

Barcode             :  --deleted--

Specification       : 48V/100AH

Cell Number         : 15

Max Dischg Curr     : -105000mA

Max Charge Curr     : 105000mA

EPONPort rate       : 1200

Console Port rate   : 115200

Customer Ver.ID     : STD   

Command completed successfully

$$

pylon_debug>`

B48D81EFCC, 2025-12-03:
Hi @functionpointer,

do you know which Firmware Version / Hardware Version does the current [Pylontech](https://esphome.io/components/pylontech/) integration supports?

For example: Is US5000 supported? (I will get one this week)

functionpointer, 2025-12-03:
No idea. Try it and report back. If it doesn't work send debug logs like OP did

B48D81EFCC, 2025-12-05:
Hi and thank you very much. I am happy to test the changes.

I get an error while compiling.

```
src/esphome/components/pylontech/pylontech.cpp: In lambda function:
src/esphome/components/pylontech/pylontech.cpp:112:30: error: 'min' was not declared in this scope
  112 |     memcpy(token_buf, start, min(cursor - start, TEXT_SENSOR_MAX_LEN - 1));
      |                              ^~~
src/esphome/components/pylontech/pylontech.cpp:112:30: note: suggested alternatives:
In file included from /home/ar/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/14.2.0/algorithm:61,
                 from src/esphome/core/optional.h:19,
                 from src/esphome/core/component.h:9,
                 from src/esphome/components/pylontech/pylontech.h:3,
                 from src/esphome/components/pylontech/pylontech.cpp:1:
/home/ar/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/14.2.0/bits/stl_algo.h:5695:5: note:   'std::min'
 5695 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
In file included from /home/ar/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/14.2.0/tuple:44,
                 from /home/ar/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/14.2.0/functional:53,
                 from src/esphome/core/component.h:5:
/home/ar/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/14.2.0/bits/ranges_util.h:754:29: note:   'std::ranges::min'
  754 |   inline constexpr __min_fn min{};
      |                             ^~~
Compiling .pioenvs/eth01-evo-poe/src/esphome/components/safe_mode/safe_mode.cpp.o
*** [.pioenvs/eth01-evo-poe/src/esphome/components/pylontech/pylontech.cpp.o] Error 1
```

Some notes:
I use a ESP32 C3 wird `esp-idf` framework (I can't use Arduino)

functionpointer, 2025-12-05:
I see, I only tested with Arduino. Should be fixed now.
Try again.

It should fetch the changes automatically. If it doesn't, run `esphome clean <your.yaml>` or delete the `.esphome` folder before trying again.