esphome/feature-requests #3134: Changing from API to MQTT, log error

superprp, 2025-04-24:
**Describe the problem you have/What new integration you would like**

I got this error:
ERROR No remote or local logging method configured (api/mqtt/logger)
When i try to upload the code from yaml in esphome to esp32 after the change from using API to MQTT because i want the data persistent in home assistant while esp32 is in deepsleep mode.

**Please describe your use case for this integration and alternatives you've tried:**

This is the code from yaml:

esphome:
  name: esp-dormitorio
  friendly_name: ESP dormitorio

esp32:
  board: esp32dev
  framework:
    type: arduino

mqtt:
  broker: 192.168.1.11
  username: ''
  password: ''

ota:
  - platform: esphome
    password: xxxxx

# Enable logging
logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: xxxxxx
    password: xxxxx

captive_portal:

# I2C para INA219
i2c:
  sda: GPIO21
  scl: GPIO22
  scan: true

# Sensor INA219: voltaje, corriente y potencia
sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "Corriente batería"
      unit_of_measurement: "A"
      retain: true
    power:
      name: "Potencia batería"
      unit_of_measurement: "W"
      retain: true
    bus_voltage:
      name: "Voltaje batería"
      unit_of_measurement: "V"
      retain: true
    update_interval: 30s

# Sensor DHT11
  - platform: dht
    pin: GPIO14
    temperature:
      name: "Temperatura ambiente"
      retain: true
    humidity:
      name: "Humedad ambiente"
      retain: true
    model: DHT11
    update_interval: 30s

# Sensor PIR
binary_sensor:
  - platform: gpio
    id: gpio_motion_sensor
    pin:
      number: GPIO27
      allow_other_uses: true
    name: "Movimiento PIR"
    retain: true
    device_class: motion

# Configuración deep sleep
deep_sleep:
  id: deep_sleep_1
  run_duration: 60s         # Tiempo encendido para medir y enviar
  sleep_duration: 1h        # Despertar cada 1 hora
  wakeup_pin:
    number: GPIO27                # PIR
    allow_other_uses: true

**Additional context**

szupi-ipuzs, 2025-07-14:
In my case it's not about the persistence, but about speed. I experience big delay between the moment wifi is connected and until HA connects to the device. I give the device 30s to run before deep sleep and it is sometimes not enough to report its state via Native API.
Part of this time is wasted on bk72xx trying to connect to my wifi, but the rest is waiting for HA to connect. As far as I know HA can take up to even 1 minute to connect.
Since I can't do much about the wifi, I decided to try using mqtt.

But this issue is not about the above problems but about some error in configuration which leads to no logger being enabled. In my case I prefer to use the API logger, not mqtt. How can I explicitlly enable it, but keep mqtt-only mode for use by HA?

ssieb, 2025-07-14:
If MDNS is working on your network, HA should connect as soon as the device is connected to wifi.

You can have both api and mqtt, just don't add the device to HA using the api.

However, logging over mqtt should work if the api is not configured.  I would need to see the config and error messages.