esphome/feature-requests #3138: Ad a Reboot Button

superfunk2000, 2025-04-30:
**Describe the problem you have/What new integration you would like**
I would like a reboot button, like in the example image below.

**Please describe your use case for this integration and alternatives you've tried:**
Twice a month, I have to restart an ESP32 because MQTT is no longer working. However, it's accessible via ESP Home, and a reboot button would be useful there.

**Additional context**

[image]

randybb, 2025-04-30:
And how it is going to work? If you API/mqtt is not working, then there is no other way. Theoretically via OTA, but I am not sure if you can just initiate reboot without flashing.

You should rather focus on fixing that problem. If not, you can still add some mechanism that will reboot your device - set timeout or adding any other more sophisticated monitoring.

jzucker2, 2025-04-30:
There's already something for this: https://esphome.io/components/button/restart.html

superfunk2000, 2025-05-01:
Very good, that's exactly what I need.
Thank you very much!

closed 2025-05-01 by superfunk2000