esphome/feature-requests #3145: Country code for WIFI (ESP32-C3)

philochardbleu, 2025-05-11:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2025-05-11:
Missing description

closed 2025-05-11 by nagyrobi

philochardbleu, 2025-05-11:
A lot of new ESP32-C3 have a limited WIFI channel scan feature. They can scan all channels but not 12 neither 13. So if your router is on channel 13, it will not reach it.
In arduino IDE, I had to implement the following code to make it reach channel 13 

wifi_country_t my_country = {
    .cc = "FR", // or "JP" for japan
    .schan = 1,
    .nchan = 13,   // or 14 for Japan
    .policy = WIFI_COUNTRY_POLICY_MANUAL
  };
  esp_wifi_set_country(&my_country);

In the ESP32-C3 expressif doc, they mention the default WIFI scan
wifi_country_t config = {
    .cc = "01",
    .schan = 1,
    .nchan = 11,
    .policy = WIFI_COUNTRY_POLICY_AUTO,
};

It would be usefull to have this possibility in ESPHOME as it would enhance ESP32-C3 use in Europe and Japan !
Thank you