esphome/feature-requests #3156: Option to use an older version of esphome in Home Assistant Addon

jkufner, 2025-05-21:
**Describe the problem you have/What new integration you would like**

When an upgrade does not go as expected, it should be somewhat easy to roll back to the previous version. Home Assistant does not support to downgrade an addon, but it should be relatively easy to use an older version of esphome within Home Assistant Addon.

Home Assistant Addon could simply create a new venv with the desired version of esphome and use that instead of what is provided by default. 

**Please describe your use case for this integration and alternatives you've tried:**

This feature is needed when the addon is updated, then a device is updated, and the new version is broken due to some bug. Then, the previous version esphome is required to repair the device.

The alternative is to install esphome locally, or install the previous version of the addon manually into `~/addons`. Installing esphome locally is not that bad, as long as a suitable device is available (can't do that on a phone). Addon downgrade is quite difficult. A simple select box with esphome versions would be really great in such a case.

**Additional context**

Bugs may happen, that is ok. The question is, how difficult is to recover from them.

randybb, 2025-05-21:
Hm. Maybe there could be "N-1" add-on..or maybe not - otherwise people would not be forced to test dev/beta releases :)

jkufner, 2025-05-21:
Keep in mind, that many of the esphome-based devices are really important in our homes.  I had trouble with a fan today (which inspired me to write this bugreport), but that fan is like a half of a life support I've got... luckily, it is not summer yet. Such a situation is not really a good place to use dev/beta releases.

On the other side, the option to select the esphome version would also allow the users to test the dev/beta release before the release and to test it on devices that are not essential to their homes.

randybb, 2025-05-21:
Hard to say... I am running dev on all devices (lights, irrigation, heating,...), everything is updated every few days and no issues. So in my "world" everything is fine. But nobody is using all components, so even with automated tests there still might be something that is not caught.
Still think that N-1 could be nice.