esphome/feature-requests #3171: Allow substitutions in secret references

kjkent, 2025-06-04:
**Describe the problem you have/What new integration you would like**

Allow substitutions in secret references

**Please describe your use case for this integration and alternatives you've tried:**

The esphome UI suggests that `api.key` should be unique to each device, however this precludes using variables over hardcoding device names into the reference, as substitutions aren't expanded in secret references.

For example, if you have multiple devices each with their own API key (or any other secret), it's cleaner to have `api_key: !secret "${name}".api.key` inherited (or even repeated) across configs, than to hardcode/retype the device name in each file.

**Additional context**

Back in 2019 (https://github.com/esphome/issues/issues/486#issuecomment-506014504) it was said that secrets were resolved prior to substitutions by design and this is how it'll stay as "it makes more sense in some cases". Perhaps it would be possible to preserve this order by prepending a third step to the order of processing:

1) substitutions in `!secret` references are expanded
2) `!secret` references themselves are expanded
3) Other substitutions are expanded

I'm unsure of the level of interest in this as a whole, but I think it may make for slightly more reusable config and easier maintenance in larger esphome deployments.

Thanks! :)

kjkent, 2025-06-09:
@fortuna thanks for your response! I'm inferring that your suggestion is to set the set that config for each device, then differ the command line for each device, e.g.:

```Shell
for device in device-{1..5}; do
  API_KEY=$(cat "$device.api.key") esphome run "$device.yaml"
done
```

So that each device gets its own API key passed-in as `!env_var API_KEY`?

jpeletier, 2025-11-28:
You could do:

```yaml
substitutions:
  platform: esp32
  devices: !secret devices  # load the `devices` key out of secrets.yaml
  this_device: kitchen

api:
  encryption:
    key: ${ devices[this_device].api.key }
```
where `secrets.yaml` is:

```yaml
devices:
  kitchen:
    api:
      key: YOUR_API_KEY
  bedroom:
    api:
      key: OTHER_API_KEY
```

Alternatively, you can just use `!include secrets.yaml` to load the entire file into a substitution variable.