esphome/feature-requests #3178: Please add compatibility with BMV080 particle matter sensor

SeveranExp, 2025-06-15:
**Describe the problem you have/What new integration you would like**

 compatibility with the Bosch sensor tech sensor BMV080 to measure the amount of particles in the air

**Please describe your use case for this integration and alternatives you've tried:**

Measure the amount of particles in the air. The bmv080 is camera based without any fans so it will last longer than the usual laser based sensors.

**Additional context**

There’s an SDK dependency described here (https://github.com/sparkfun/SparkFun_BMV080_Arduino_Library ) reminds me of the bme68x BSEC2 requirement.

SeveranExp, 2026-03-14:
Ooh this is exciting! Thank you @pdennett !

BliXem1, 2026-05-14:
I did a change in the code, if this works I will let you know what I changed. I also download alle the files for locally use.