esphome/feature-requests #3192: [Midea Climate] Support display on/off commands via UART

niquitoooo, 2025-06-29:
**Describe the problem you have/What new integration you would like**
I would like to add support for turning the Midea AC display on and off via UART in the `climate.midea` component. My AC supports this feature using the Midea USB dongle, and midea-ac-py can successfully toggle the display using UART commands. However, ESPHome does not currently expose this capability.

**Please describe your use case for this integration and alternatives you've tried:**
My use case is to turn off the display panel of the indoor unit to reduce light pollution at night. I have confirmed that the AC supports this via UART using midea-ac-py, but ESPHome does not provide this function yet. Alternatives like IR are not ideal because I am using the Midea USB dongle for full control.

**Additional context**
- Midea mini split with original Midea USB dongle
- Display on/off confirmed working over UART using midea-ac-py
- ESPHome climate.midea does not support sending display commands yet

It would be great to expose `display_on` and `display_off` in ESPHome for consistency.

Thanks for your work!

nyok1912, 2025-06-29:
I join the petition and confirm that it should work! In fact, with osk10x it works on most relatively recent machines.

BenjaminAllott, 2026-01-03:
I want to follow up on what I have found thus far,

I have a Actron Air unit (Australia) with the official USB wifi dongle, and a ESP12e USB dongle in ESPHome,

I would love to swap all the AC units to ESPHome but until the LED Display is reporting status I can not.

So, in the official app and the Midea AC LAN HACS intergration there is a LED Display ON/OFF toggle, and it updates based on the IR remote as well as app (VIA UART)

I have the below data based on pressing the display toggle button (removed repeated lines)

it appears to my low IQ monkey brain that bit 25 toggles between 70 and 00 based on the LED display,
I don't have the know how to fork and start messing with the C++ code to pull the bit 25 into a usable field (binary sensor or toggle switch)

`[16:48:05.675][D][AirConditioner:177]: Enqueuing a GET_STATUS(0x41) request...
[16:48:05.715][D][ApplianceBase:173]: Enqueuing the request...
[16:48:05.716][D][ApplianceBase:075]: Getting and sending a request from the queue...
[16:48:05.716][D][ApplianceBase:162]: TX: AA 21 AC 8D 00 00 00 00 03 03 41 81 00 FF 03 FF 00 02 00 00 00 00 00 00 00 00 00 00 00 00 03 3C 2A 72 
[16:48:05.823][D][ApplianceBase:063]: RX: AA 28 AC 00 00 00 00 00 03 03 C0 01 55 66 7F 7F 00 00 00 00 00 5D 6B 09 00 89 00 00 00 00 00 00 00 00 00 00 00 00 3C 2E E8 
[16:48:05.826][D][AirConditioner:204]: New status data received. Parsing...

[16:48:06.401][D][button:022]: 'Air Conditioner Display Toggle' Pressed.
[16:48:06.404][D][AirConditioner:186]: Enqueuing a priority TOGGLE_LIGHT(0x41) request...
[16:48:06.405][D][ApplianceBase:173]: Enqueuing the request...
[16:48:06.718][D][ApplianceBase:075]: Getting and sending a request from the queue...
[16:48:06.721][D][ApplianceBase:162]: TX: AA 21 AC 8D 00 00 00 00 03 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 98 30 33 
[16:48:06.818][D][ApplianceBase:063]: RX: AA 28 AC 00 00 00 00 00 03 03 C0 01 55 66 7F 7F 00 00 00 00 00 5D 6B 09 70 89 00 00 00 00 00 00 00 00 00 00 00 00 98 AA A0 
[16:48:06.835][D][AirConditioner:204]: New status data received. Parsing...
[16:48:06.835][D][ApplianceBase:154]: Destroying the request...
[16:48:07.683][D][AirConditioner:177]: Enqueuing a GET_STATUS(0x41) request...
[16:48:07.723][D][ApplianceBase:173]: Enqueuing the request...
[16:48:07.724][D][ApplianceBase:075]: Getting and sending a request from the queue...

[16:48:11.878][D][AirConditioner:204]: New status data received. Parsing...
[16:48:11.878][D][ApplianceBase:154]: Destroying the request...
[16:48:12.603][D][button:022]: 'Air Conditioner Display Toggle' Pressed.
[16:48:12.608][D][AirConditioner:186]: Enqueuing a priority TOGGLE_LIGHT(0x41) request...
[16:48:12.609][D][ApplianceBase:173]: Enqueuing the request...
[16:48:12.773][D][ApplianceBase:075]: Getting and sending a request from the queue...
[16:48:12.776][D][ApplianceBase:162]: TX: AA 21 AC 8D 00 00 00 00 03 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A0 4C 0F 
[16:48:12.890][D][ApplianceBase:063]: RX: AA 28 AC 00 00 00 00 00 03 03 C0 01 55 66 7F 7F 00 00 00 00 00 5D 6B 09 00 89 00 00 00 00 00 00 00 00 00 00 00 00 A0 9C 16 

[16:48:15.916][D][AirConditioner:204]: New status data received. Parsing...
[16:48:15.916][D][ApplianceBase:154]: Destroying the request...
[16:48:16.775][D][AirConditioner:177]: Enqueuing a GET_STATUS(0x41) request...
[16:48:16.841][D][ApplianceBase:173]: Enqueuing the request...
[16:48:16.842][D][ApplianceBase:075]: Getting and sending a request from the queue...
[16:48:16.842][D][ApplianceBase:162]: TX: AA 21 AC 8D 00 00 00 00 03 03 41 81 00 FF 03 FF 00 02 00 00 00 00 00 00 00 00 00 00 00 00 03 45 4E 45 
[16:48:16.875][D][button:022]: 'Air Conditioner Display Toggle' Pressed.
[16:48:16.875][D][AirConditioner:186]: Enqueuing a priority TOGGLE_LIGHT(0x41) request...
[16:48:16.876][D][ApplianceBase:173]: Enqueuing the request...
[16:48:16.930][D][ApplianceBase:063]: RX: AA 28 AC 00 00 00 00 00 03 03 C0 01 55 66 7F 7F 00 00 00 00 00 5D 6B 09 00 79 00 00 00 00 00 00 00 00 00 00 00 00 45 79 A4 
[16:48:16.933][D][AirConditioner:204]: New status data received. Parsing...
[16:48:16.934][D][ApplianceBase:154]: Destroying the request...
[16:48:17.818][D][ApplianceBase:075]: Getting and sending a request from the queue...
[16:48:17.821][D][ApplianceBase:162]: TX: AA 21 AC 8D 00 00 00 00 03 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 B8 13 30 
[16:48:17.933][D][ApplianceBase:063]: RX: AA 28 AC 00 00 00 00 00 03 03 C0 01 55 66 7F 7F 00 00 00 00 00 5D 6B 09 70 79 00 00 00 00 00 00 00 00 00 00 00 00 B8 BA 80 
`

parkghost, 2026-01-03:
https://github.com/parkghost/esphome-configs/blob/main/modules/climate/midea_display_light.yaml

I've created a proof-of-concept (POC) that can detect and control screen backlight, but I haven't sent a pull request to esphome due to device compatibility testing and library dependencies. Anyone interested is welcome to test it and take over the follow-up work.

https://github.com/dudanov/MideaUART/issues/26#issuecomment-3302767708

BenjaminAllott, 2026-01-03:
> https://github.com/parkghost/esphome-configs/blob/main/modules/climate/midea_display_light.yaml
> 
> I've created a proof-of-concept (POC) that can detect and control screen backlight, but I haven't sent a pull request to esphome due to device compatibility testing and library dependencies. Anyone interested is welcome to test it and take over the follow-up work.
> 
> [dudanov/MideaUART#26 (comment)](https://github.com/dudanov/MideaUART/issues/26#issuecomment-3302767708)

Thank you very much, this works beautifully,

now it is only humidity that is missing but I can live with that =)

rbflurry, 2026-08-19:
This works for my Mr cool unit. But the switch or toggle stays on all the time. Tapping it will toggle the lcd on or off but the UI feedback always returns to On.