esphome/feature-requests #3195: Support for Olimex i2c modules (MOD-IO and MOD-IO2)

bbm-ro, 2025-07-03:
**Describe the problem you have/What new integration you would like**

  Hi,
  I would like to request the integration of the MOD-IO and MOD-IO2 modules from Olimex. They work over i2C, usually connected with UEXT cables to ESP32 board from the same manufacturer. They offer additional GPIO pins and relays which can expand the capabilities of the main boards.
  The procedure for using the modules over i2c involves writing and reading commands to/from the i2c address of the board. I know this can be done using a custom component, but it's very complicated to get it working for a non-developer. I'm a network engineer with some minimal C++ programming experience, but I couldn't wrap my head around how to get this done. If it were a native integration, it would just be a matter of using GPIO pins or i2c devices, same as any other sensor or switch.
  Thank you.

**Please describe your use case for this integration and alternatives you've tried:**

  The Olimex boards are very versatile, and they come in various models using ESP32 chips. Using the UEXT connector, they can be expanded with additional boards (which can also be chained) for adding various sensors, relay, GPIO, etc. Some of the boards are meant for industrial environments (temperature operation -40+85C, galvanic insulation from the Ethernet powering), which make them suitable for more serious projects.
  Currently, I'm using a few of their ESP32-POE-ISO boards as BLE proxies for Home Assistant. I have an ESP32-EVB-IND board, a MOD-IO module and a MOD-IO2 module, which I was hoping to get running with ESPHome so that I can replace some Raspberry Pi's I'm using to control relay boards. but this would require the relays ang GPIO pins on the MOD-IO /IO2 boards to be easily/natively configurable in ESPHome, similar to the regular pins on the ESP32 boards.
  I can control the onboard relays on the ESP32-EVB-IND, but only because those are controlled by the native ESP32 pins, which are usable in ESPHome without issues.

**Additional context**

Manuals for MOD-IO and MOD-IO2 modules, where the commands are described, with links to the firmware archive which contains C++ examples (which probably are trivial for a skilled developer to use for creating a component in ESPHome):

https://www.olimex.com/Products/Modules/IO/MOD-IO/resources/MOD-IO.pdf
https://www.olimex.com/Products/Modules/IO/MOD-IO2/resources/MOD-IO2.pdf

Other Olimex boards and modules:
https://www.olimex.com/Products/IoT/ESP32/
https://www.olimex.com/Products/IoT/ESP32-C6/
https://www.olimex.com/Products/Modules/

g8ry, 2025-09-14:
#
# **** REMEMBER TO POWER THE I2C WITH ITS OWN 12V SUPPLY ******
# 
#
substitutions:
  device_name: test-esp32-poe-mod-io2-working
  friendly_name: test_esp32_poe_mod_io2_working
  dht_pin: GPIO16
  dht_model: AM2302
  static_ip: 192.168.0.245
  use_address: 192.168.0.245
  default_on_boot: RESTORE_DEFAULT_OFF
  
esphome:
  name: "${device_name}"
#  platform: ESP32
#  board: 
  on_boot:
    priority: 600  # after i2c init etc.
    then:
      - lambda: |-
          id(relay_1_switch).publish_state(false);
          id(relay_2_switch).publish_state(false);  

esp32:
  board: esp32-poe
  framework:
    type: esp-idf  

ethernet: !include poe_lan.inc

#sensor:
#  - <<: !include dht_sensor.inc
#  - <<: !include uptime_sensor.inc

web_server:
  port: 80

#text_sensor:
#  - <<: !include uptime_human_readable.inc  
    

# UART logs
logger:

api:
ota:
  platform: esphome

i2c:
  sda: GPIO13
  scl: GPIO16
  scan: true
  id: bus_a

i2c_device:
  id: modio2
  address: 0x21   # Use the correct MOD-IO2 I2C address

globals:
  - id: relay_state
    type: uint8_t
    restore_value: no
    initial_value: '0'

interval:
  - interval: 1s   # 1s - doesn't work with anything less
    then:
      - lambda: |-
          uint8_t cmd = 0x43;
          uint8_t state = 0;
          id(modio2).write(&cmd, 1);
          id(modio2).read(&state, 1);
          id(relay_state) = state;

sensor:
  - platform: bme280_i2c
    temperature:
      name: "${friendly_name} BME280 Temperature"
    pressure:
      name: "${friendly_name} BME280 Pressure"
    humidity:
      name: "${friendly_name} BME280 Humidity"
    address: 0x77     # Switch to 0x77 if needed
    update_interval: 60s
    i2c_id: bus_a      # This is the bus you're using with UEXT          

switch:
  - platform: template
    name: "MOD-IO2 Relay 1"
    id: relay_1_switch
    restore_mode: ALWAYS_OFF
    lambda: |-
      return (id(relay_state) & 0x01) != 0;
    turn_on_action:
      - lambda: |-
          id(relay_state) |= 0x01;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);       
          id(relay_1_switch).publish_state(true);
    turn_off_action:
      - lambda: |-
          id(relay_state) &= ~0x01;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);        
          id(relay_1_switch).publish_state(false);   

  - platform: template
    name: "MOD-IO2 Relay 2"
    id: relay_2_switch
    restore_mode: ALWAYS_OFF
    lambda: |-
      return (id(relay_state) & 0x02) != 0;
    turn_on_action:
      - lambda: |-
          id(relay_state) |= 0x02;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);    
          id(relay_2_switch).publish_state(true);    
    turn_off_action:
      - lambda: |-
          id(relay_state) &= ~0x02;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);        
          id(relay_2_switch).publish_state(false);

g8ry, 2025-09-14:
This is the code I have for getting both relays to work and the BME280 UEXT module. 
Big thanks to chatgpt for doing the i2c for me.

`#
# **** REMEMBER TO POWER THE I2C WITH ITS OWN 12V SUPPLY ******
# 
#
substitutions:
  device_name: test-esp32-poe-mod-io2-working
  friendly_name: test_esp32_poe_mod_io2_working
  dht_pin: GPIO16
  dht_model: AM2302
  static_ip: 192.168.0.245
  use_address: 192.168.0.245
  default_on_boot: RESTORE_DEFAULT_OFF
  
esphome:
  name: "${device_name}"
  on_boot:
    priority: 600  # after i2c init etc.
    then:
      - lambda: |-
          id(relay_1_switch).publish_state(false);
          id(relay_2_switch).publish_state(false);  

esp32:
  board: esp32-poe
  framework:
    type: esp-idf  

ethernet: 
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  manual_ip:
    static_ip: ${static_ip}
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  use_address: ${use_address}

web_server:
  port: 80

logger:

api:
ota:
  platform: esphome

i2c:
  sda: GPIO13
  scl: GPIO16
  scan: true
  id: bus_a

i2c_device:
  id: modio2
  address: 0x21   # Use the correct MOD-IO2 I2C address

globals:
  - id: relay_state
    type: uint8_t
    restore_value: no
    initial_value: '0'

interval:
  - interval: 1s   # 1s - doesn't work with anything less
    then:
      - lambda: |-
          uint8_t cmd = 0x43;
          uint8_t state = 0;
          id(modio2).write(&cmd, 1);
          id(modio2).read(&state, 1);
          id(relay_state) = state;

sensor:
  - platform: bme280_i2c
    temperature:
      name: "${friendly_name} BME280 Temperature"
    pressure:
      name: "${friendly_name} BME280 Pressure"
    humidity:
      name: "${friendly_name} BME280 Humidity"
    address: 0x77     # Switch to 0x77 if needed
    update_interval: 60s
    i2c_id: bus_a      # This is the bus you're using with UEXT          

switch:
  - platform: template
    name: "MOD-IO2 Relay 1"
    id: relay_1_switch
    restore_mode: ALWAYS_OFF
    lambda: |-
      return (id(relay_state) & 0x01) != 0;
    turn_on_action:
      - lambda: |-
          id(relay_state) |= 0x01;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);       
          id(relay_1_switch).publish_state(true);
    turn_off_action:
      - lambda: |-
          id(relay_state) &= ~0x01;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);        
          id(relay_1_switch).publish_state(false);   

  - platform: template
    name: "MOD-IO2 Relay 2"
    id: relay_2_switch
    restore_mode: ALWAYS_OFF
    lambda: |-
      return (id(relay_state) & 0x02) != 0;
    turn_on_action:
      - lambda: |-
          id(relay_state) |= 0x02;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);    
          id(relay_2_switch).publish_state(true);    
    turn_off_action:
      - lambda: |-
          id(relay_state) &= ~0x02;
          uint8_t data[2] = {0x40, id(relay_state)};
          id(modio2).write(data, 2);        
          id(relay_2_switch).publish_state(false);`

bbm-ro, 2025-10-08:
Thanks for the code, @g8ry !
It works for the MOD-IO2 board connected to an ESP32-EVB board. Now I just have to apply my limited programming skills to adapt it for the MOD-IO board, and hopefully get them to all work together (daisy-chain them using the UEXT connectors, each board with its own power supply and, of course, different I2C addresses for the extension boards).
Sorry for the late reply, I was on vacation and I wanted to get home and test before replying.

msohanifr, 2026-04-25:
It's very hard to get the right kernel that has i2c available to boot, once you get that, then it's all about setting the address.