esphome/feature-requests #3208: [sx126x,sx127x] FineOffset 433/868MHz RF sensor protocol/gateway

LorbusChris, 2025-07-12:
**Describe the problem you have/What new integration you would like**

I'm using the the below yaml to read data from a FineOffset WH65 weather station using an sx127x which was whipped up by @swoboda1337 in https://github.com/esphome/feature-requests/issues/2886. That already works great!

However I think it'd be great to extend this to support more sensors. FineOffset sensors are sold under many brand names including Ecowitt, Froggit, Ambient Weather and many more. I only have access to the one 868MHz WH65 weather station, but I've extended the yaml with some TODOs - they shouldn't be hard to implement with https://github.com/merbanan/rtl_433 as a reference. 

Eventually, this could live in a dedicated component so that discovered sensors on the same band could be recognized and added dynamically.

Other protocols have in the past been added to remote_receiver/remote_transmitter, maybe going that route would be applicable here too?

**Please describe your use case for this integration and alternatives you've tried:**

```
esphome:
  name: "fineoffset-rf-sensor-gateway"
  friendly_name: FineOffset RF Sensor Gateway

esp32:
  board: adafruit_feather_esp32_v2
  framework:
    type: arduino

spi:
  clk_pin: GPIO5
  mosi_pin: GPIO19
  miso_pin: GPIO21

sx127x:
  dio0_pin: GPIO27 # IRQ
  cs_pin: GPIO33
  rst_pin: GPIO15
  frequency: 868350000
  modulation: FSK
  bandwidth: 100_0kHz
  bitrate: 17240
  bitsync: true
  rx_floor: -50
  rx_start: true
  sync_value: [0x2d, 0xd4]
  preamble_size: 5
  preamble_detect: 2
  preamble_errors: 8
  preamble_polarity: 0xAA
  payload_length: 17
  packet_mode: true
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "Received message: %s", format_hex_pretty(x).c_str());
          ESP_LOGD("fineoffset", "Type: 0x%02X", x[0]);
          ESP_LOGD("fineoffset", "ID: %d", x[1]);

          // Type 0x90
          // Fine Offset Electronics WS90 Weather Station
          if (x[0] == 0x90) {
            // TODO
          }

          // Type 0x80
          // Fine Offset Electronics WS80 Weather Station
          if (x[0] == 0x80) {
            // TODO
          }

          // Type 0x68
          // Fine Offset Electronics WS68 Wind Sensor
          if (x[0] == 0x68) {
            // TODO
          }

[... 366 lines cut ...]
  - platform: template
    name: "Rainfall"
    id: sensor_rainfall
    state_class: "measurement"
    unit_of_measurement: "mm"
    icon: "mdi:weather-rainy"

  - platform: template
    name: "UV Index"
    id: sensor_uv_index
    state_class: "measurement"
    icon: "mdi:sun-wireless"

  - platform: template
    name: "Light Intensity"
    id: sensor_light_intensity
    state_class: "measurement"
    unit_of_measurement: "lx"
    device_class: "illuminance"

  - platform: internal_temperature
    name: "Internal Temperature"

```
**Additional context**

swoboda1337, 2025-07-13:
Using continuous mode and remote receiver with FSK is not really viable. The OOK receiver is gated by signal strength and doesn't toggle the GPIO when the signal isn't strong enough. The FSK receiver always sends data continuously on the gpio at a very fast rate (bitrate is faster than OOK). The receiver is never gated. You can sometimes decode FSK in continuous but it's not consistent and you miss most packets.

LorbusChris, 2025-07-13:
Could the preamble length be exposed in continuous mode somehow?

LorbusChris, 2025-07-13:
sorry, I meant to say: is there a way to get the recorded preamble in packet mode?

swoboda1337, 2025-07-13:
> sorry, I meant to say: is there a way to get the recorded preamble length in packet mode?

No. Preamble is meant to help afc and agc. It is not meant to be decoded. The receiver will miss the first bit anyways before the control loops settle.

CaptainDouche, 2025-09-02:
Hi Guys,
this reminds me on how to read TX29 thermo/hygro sensors using a RFM69:
in this module, you also enter into some registers, what frequency, etc, and especially as well the preamble and the sync word - which is also 2D,D4 btw.
For TX29+RFM69 there is no need to know any length indication hidden in the preamble:
onece it has seen the sync word, it records all that comes in, and when the Fifo is full, you get an interrupt, read out the entire Fofo, and detect then on hi level the packets end.
You dont miss any packets that way, ju just need to know when the packet ends and the rest of the fifo only contains noise.
Note that even after end of the transmission, the receiver fills the fifo (with noise), so it is guaranteed that after a packet began, fifo gets full, you get informed, and see all the packets. the not-knowing where the packet ends is as i said no problem, software can easy detect it.