esphome/feature-requests #3254: VIEWESMART 2.1inch Touch Knob Display

red4dj, 2025-07-30:
### The problem

Hi all,
I purchased  VIEWESMART 2.1inch Touch Knob Display (model `UEDX48480021-MD80ET`) and cannot seem to get the display functioning. The screen is built off of a ST7701S. An output of the test card is included below.

There is another, smaller version of this display that runs on a `GC9A01`. More details are available [here](https://community.home-assistant.io/t/1-28-inch-240-240-esp32c3-round-display-with-rotary-knob-uedx24240013-md50e-by-viewe-company/786687) 

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

st7701s

### YAML Config

```yaml
esphome:
  name: touch-knob
  friendly_name: "Touch Knob"

logger:

# BOARD
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

psram:
  mode: octal
  speed: 80MHz

# DISPLAY
spi:
  clk_pin:
    number: GPIO13
    allow_other_uses: true
  mosi_pin:
    number: GPIO12
    allow_other_uses: true

display:
  - platform: st7701s
    id: knob_display
    dimensions:
      width: 480
      height: 480
    cs_pin: GPIO18
    reset_pin:
      number: GPIO8
      # allow_other_uses: True  (Uncomment when enabling touchscreen)
    de_pin: GPIO17
    hsync_pin: GPIO46
    vsync_pin: GPIO3
    pclk_pin: GPIO9
    data_pins:
      red:
        - 40         #r1
        - 41         #r2
        - 42         #r3
        - 2          #r4
        - 1          #r5
      green:
        - 21         #g0
        - 47         #g1
        - 48         #g2
        - 45         #g3
        - 38         #g4
        - 39         #g5
      blue:
        - 10         #b1
        - 11         #b2
        - number: 12 #b3
          allow_other_uses: True
        - number: 13 #b4
          allow_other_uses: True
        - 14         #b5
    color_order: RGB
    hsync_pulse_width: 255
    hsync_front_porch: 1
    hsync_back_porch: 255
    vsync_pulse_width: 254
    vsync_front_porch: 2
    vsync_back_porch: 254
    show_test_card: True

switch:
  - name: "Backlight"
    id: backlight
    platform: gpio
    pin: GPIO7
    inverted: True
    restore_mode: ALWAYS_ON 

# TOUCHSCREEN (Included for Reference, Commented Out For Issue Replication)
# i2c:
#   sda: GPIO16
#   scl: GPIO15

# touchscreen:
#   - platform: cst816
#     id: knob_touchscreen
#     reset_pin: 
#       number: GPIO8
#       allow_other_uses: True

# ENCODER (Included for Reference, Commented Out For Issue Replication)
# sensor:
#   - name: "Rotary Encoder"
#     id: knob_encoder
#     platform: rotary_encoder
#     pin_a: GPIO6
#     pin_b: GPIO5
#     resolution: 2
#     internal: True

# binary_sensor:
#   - name: "Buttton"
#     id: encoder_button
#     platform: gpio
#     pin:
#       number: GPIO0
#       inverted: True
#       ignore_strapping_warning: True
#     internal: True
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

All the docs are available here: https://github.com/VIEWESMART/UEDX48480021-MD80ESP32-2.1inch-Touch-Knob-Display

Manufacturer Product Page: https://viewedisplay.com/product/esp32-2-1-inch-480x480-round-tft-knob-touch-display-rotary-encoder-arduino-lvgl/

I personally purchased form here: https://www.amazon.com/dp/B0F99N982Y although it is available from other retailers/sellers.

Test Card Output:
[image]

ssieb, 2025-07-30:
You would be better off asking in the esphome discord server.

tadeastehan, 2025-11-15:
> Resolved by [esphome/esphome#9892](https://github.com/esphome/esphome/pull/9892)

Hello, have you tried running the display and touchscreen at the same time? I have problem to setup the touchscreen. Does your touchscreen work? Thanks in adance.